joan2937/pigpio #7: pigpio with new Raspberry 2

ErwinForum, 2015-04-02:
when i start the pigpio daemon the raspberry 2 is hanging, how can i solve this.

joan2937, 2015-04-02:
When did you last update pigpio?

Is there any error message?

What operating system are you using?

What does uname -a report?

ErwinForum, 2015-04-02:
i am running raspberry weehzly and there is no error message the pi is no longer responding i have downloaded the pigio in december 2014.
Linux raspimessen 3.18.8+ #765 PREEMPT Thu Mar 5 15:41:59 GMT 2015 armv6l GNU/Linux
should it generally running on the new raspberry without problems, then i will do a fresh install tommorrow and try i again.

joan2937, 2015-04-02:
All the libraries had to be updated for the Pi2 as memory addresses of the gpios and other peripherals have changed.  A version from December 2014 pre-dates the Pi2 and will hang.

Download and build the latest version of pigpio.

ErwinForum, 2015-04-02:
thanks for your help will try it tomorrow and report to you.

ErwinForum, 2015-04-03:
it's working now, many thanks for your tip, have a nice eastern.

closed 2015-04-03 by ErwinForum