joan2937/pigpio #13: RPi2

ha5di, 2015-05-30:
Hi

is it compatible with RPi2?

Thanks... Béla

joan2937, 2015-06-01:
Yes.   pigpio should work on all the current Pis (including the compute module).

closed 2015-06-01 by joan2937

SlySven, 2016-02-24:
When you say:
>Yes. pigpio should work on all the current Pis (including the compute module).

I am just starting to experiment with Pigpio and I am excited by the possibilities that it offers to a 'C' coder (who has virtually no experience of Python  :snake:)!

Reading through the `man` pages first, as a responsible coder does, :angel: : How does the code work when the range of usable GPIO pins on the Compute Module is nearly all the 120-odd total that the BCM2835 has - does the current code detect the platform *is* a Compute Module and allow all 120 pins to be used but on the other RPi products reject access to pins that are used "internally"?

joan2937, 2016-02-24:
I don't know how many pins the compute module has.

It has the same number of GPIO (54) as all the other Pis - however on the compute module you have physical access to all 54 GPIO.

GPIO 0-31 are fully supported by pigpio - sampling at up to 1 MHz, callbacks, PWM generation, waveforms.  The other GPIO (32-53) are restricted to the  other functions (read/write/set mode/get mode, hardware clock, hardware PWM etc.).

SlySven, 2016-02-25:
Ah further checking on the Farnell website [here](http://www.farnell.com/layout/1830504.pdf) and [here](http://www.farnell.com/datasheets/1836579.pdf) reveals that there are definitely 45 free GPIO pins (Of two more, GPIO 46 is HDMI detect and 47 seems to be used to control booting from the on-board Flash and for the "Activity" LED) - I guess that a naked BCM2835 device core *may* have 120 pins but the remainder are tied up in the HDMI interface, 2x RPi camera interfaces (each with a I2C bus BTW!), 2x RPi display and a composite video (no audio) output.