joan2937/pigpio #14: cgi can't connect to pigpio

Kaloka, 2015-06-20:
I've tried to run a cgi example from pigpio python CGI abyz.co.uk site and it doesn't work. When i open site http://rpiAdress/pigpio_cgi.py it show error on top of the page:
"Can't connect to pigpio on soft(8889) Did you start the pigpio daemon? E.g. sudo pigpiod Did you specify the correct Pi host/port in the environment variables PIGPIO_ADDR/PIGPIO_PORT? E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888 Did you specify the correct Pi host/port in the pigpio.pi() function? E.g. pigpio.pi('soft', 8888))"
I run sudo pigpiod -p 8889 before.(port 8888 is also not working)
pigpiod is working well becouse I have an application on Windows which can connect and use pigpio over network and is work properly.
Whan I need to do? (also node.js can't connect to pigpio)

Kaloka, 2015-06-20:
I must add that I'm using logmein hamachi on rpi, but I've tried to connect on both IP adresses(local network and virtual network)

joan2937, 2015-06-20:
When I run the script without starting the daemon I get this message.

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%% Can't connect to pigpio on localhost(8888) Did you start the pigpio daemon? E.g. sudo pigpiod Did you specify the correct Pi host/port in the environment variables PIGPIO_ADDR/PIGPIO_PORT? E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888 Did you specify the correct Pi host/port in the pigpio.pi() function? E.g. pigpio.pi('soft', 8888)) %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%% 

You must have altered the script (for test purposes?) or added some environment variables for it to be saying "Can't connect to pigpio on soft(8889)" rather than "Can't connect to pigpio on localhost(8888)".

If other systems can access the daemon at port 8888 that is clearly the correct port to be using.

Do you have the loopback interface running, i.e. on the Pi can you ping localhost?  If not make sure /etc/network/interfaces contains the lines

 auto lo
 iface lo inet loopback

Kaloka, 2015-06-21:
I've tried this:
in script:    pigpio()    in shell:    sudo pigpiod
it doesn't work

in script:    pigpio('soft',8888)      in shell sudo pigpiod
it doesn't work

in script:   pigpio('soft',8889)       in shell   sudo pigpiod -p 8889
it doesn't work

and in script     pigpio('192.168.10.103',8889[and 8888] )
it doesn't work

In /etc/network/interfaces   I have:
auto wlan0      (when I change it to auto lo    then rpi can't connect to wifi network)
iface lo inet loopback

I'll try install pigpio from github and we'll see.

joan2937, 2015-06-21:
Does etc/network/interface have the line auto lo? Can you ping localhost?

I said to add the lines if they were not present.

You should not change anything in pigpio_cgi.py, leave that as it is until you get it working with the defaults.  Leave everything at the defaults.

Kaloka, 2015-06-22:
When I add line "auto lo" in /etc/network/interfaces file then ping localhost is working, but there is no automatic(boot-up time) connection to wifi / internet. I must click connect in wpa_gui and then wifi(internet) and cgi script is working :) 
What I need to do to get automatic internet connection(boot-up time) ?

My /etc/network/interfaces file:

auto lo

iface lo inet loopback
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet static
address 192.168.10.103
netmask 255.255.255.0
network 192.168.10.0
broadcast 192.168.10.255
gateway 192.168.10.100
auto wlan0

joan2937, 2015-06-22:
I have no idea.  I don't know your system or what you are trying to do.  If you have a general network configuration problem you need to ask on a general software site.

Kaloka, 2015-06-22:
I've add lines:
" sudo ifdown wlan0
  sudo ifup wlan0" to “/etc/rc.local”
and edit file /etc/default/ifplugd , 
changed line HOTPLUG_INTERFACES="all" 
  to HOTPLUG_INTERFACES="eth0 wlan0"
and....    it works! 
Thanks for everything:) God bless You:)

joan2937, 2015-06-24:
I'm going to assume that getting networking working properly also cleared up the pigpio CGI script issue.

closed 2015-06-24 by joan2937