joan2937/pigpio #15: 3-wire spi doesn't work

KevinLowe360, 2015-07-27:
I open my spi-dev0 as follow(10kbps,set the 'W' flag as document says):
    spi = pi.spi_open(0,10000,0x200)
and then i write the data with this:
    pi.spi_write(spi,[0x01,0x02,0x03,0x04])
However,no output on MISO or MOSI.Where is wrong?Is 3-wire spi supported?

joan2937, 2015-07-27:
(http://abyz.co.uk/rpi/pigpio/python.html#spi_open)

Firstly 3-wire operation only works on the main SPI device and works within the limits imposed by the Pi hardware.

The hardware supports switching the MOSI line between MOSI and MISO after a user selected number of bytes.

The code above switches between MOSI and MISO after 0 bytes, i.e. immediately.

Contrast with the following which first does the above and secondly switches after 4 bytes.
```python
#!/usr/bin/env python

import pigpio

W3=(1<<9) # select 3-wire operation
W3N4=(4<<10) # switch to miso after four bytes

pi = pigpio.pi() # Connect to local Pi.

spi1 = pi.spi_open(0,10000,W3)
spi2 = pi.spi_open(0,10000,W3|W3N4)

pi.spi_write(spi1,[0x01,0x02,0x03,0x04])

pi.spi_write(spi2,[0x01,0x02,0x03,0x04])

pi.spi_close(spi1)
pi.spi_close(spi2)

pi.stop()
```
[image]

I presume the write is used for testing.  Normally xfer would be used as you expect to write some bytes then read some back.

KevinLowe360, 2015-07-28:
It works!Seems something wrong with my comprehend to 3-wire spi. I have got the perfect output with your help,many thanks.

closed 2015-07-28 by KevinLowe360