joan2937/pigpio #18: Notification structure

ColorfullyMe, 2015-08-13:
Hi,
thanks for the nice lib!

Could you please clarify a thing or two for me!?
As given [here](http://abyz.co.uk/rpi/pigpio/cif.html#gpioNotifyBegin) I try to read `/dev/pigpioX` after subscribing for notifications but get ***strange results***.

First I subscribe...
```
$ pigs no
0
$ pigs nb 0 240 // this should be (1<<4) | (1<<5) | (1<<6) | (1<<7)
```
Then I monitor `/dev/pigpioX` with Node.js this is my code:
```js
var fs = require('fs'); // file system
var rstream = fs.createReadStream('/dev/pigpio0');
var binary = require('binary');

binary.stream(rstream)
    .loop(function (end, vars) {
        this
            .word16lu('seqno')
            .word16lu('flags')
            .word32lu('tick')
            .word32lu('level')
            .tap(function (vars_) {
                console.log(vars_);
            });
    });
```

It is plain simple code. What it does is:
- Opens `/dev/pigpioX` for reading
- Starts reading present data or waits for data
- If data is present it loops in "frames" of 12 bytes as expected and assigns to the variables.
- In the loop it dumps to stdout the vars (structure)

What I get is:
```sh
...
{ seqno: 30, flags: 0, tick: 3808029467, level: 2952839535 }
{ seqno: 31, flags: 0, tick: 3812660689, level: 2952839551 }
...
```

***Questions:***
1. Why is flags 0? It looks that seqno is valid. I also compared tick with `pigs T` output. They are ~ equal.
1.1 Why is level such int? Should it not be 0/1?
2. Could you please confirm that the structure given is valid? 12 bytes 2+2+4+4?
3. What is written exactly to `/dev/pigpioX`? Is it a sequence of notification structures? Is there some bytes for delimiter?
4. I note that occasionally some events/new data occurs without me changing the gpios, why? This means something else is happening meanwhile?
 
Thanks in advance for any answers!

joan2937, 2015-08-13:
Notifications consist of 12 bytes with the following C format.

```C
typedef struct
{
   uint16_t seqno;
   uint16_t flags;
   uint32_t tick;
   uint32_t level;
} gpioReport_t;
```

seqno starts at 0 each time a notification is created and then increments by one for each report.  It can be used to detect missing notifications.  I'd expect that to be pretty rare unless you are listening to tens of thousands of events per second.

flags, if bit 5 is set then bits 0-4 of the flags indicate a gpio which has had a watchdog timeout.  No other flags are defined.

tick is the number of microseconds since system boot. It is a 32 bit quantity and will wrap around after 1h12m.

level indicates the level of each gpio. You will have to mask out the gpios you are actally interested in.  Gpio x is bit 1<<x.

There is no delimiter.  You should read multiples of 12 bytes.  A partial notification will not be in the data stream.

On your point 4.  I'm sure the intention is you only get notifications for the level changes of the gpios you have specified (or watchdog timeouts if you have requested a watchdog on one of the gpios).  I don't think I've changed anything in that area.

ColorfullyMe, 2015-08-13:
@joan2937 Thanks this was quick!
Could you please elaborate more on:
- why is flags 0 for me?
- how can 2952839615 for level be unmasked?

joan2937, 2015-08-13:
Watchdogs are not the same as notifications.  Have you set up any watchdogs on the gpios?  Flags will be zero unless a watchdog expires.  You use a watchdog to say you want to be notified of the gpio level every x milliseconds, regardless of whether the level has changed or not.  Try your test but type pigs wdog 5 15 say.  Where 5 is a gpio you asked for notifications on.  Then you will get a report with the flag set every 15 ms, provided that no other level change happened for that gpio.  Use pigs wdog 5 0 to switch off watchdogs on gpio 5.

2952839615 is 0b10110000000000001100000110111111

> 33222222222211111111110000000000
> 10987654321098765432109876543210
>
> 10110000000000001100000110111111

That means gpios 31, 29, 28, 15, 14, 8, 7, 5, 4, 3, 2, and 1 are high, the other gpios are low.

You need to use bit manipulation to extract the bits of interest.  I'm not a Javascript user so can't help with that.

ColorfullyMe, 2015-08-13:
@Nice! Thanks again! I think I get it slowly...

It looks I read the pipe data correctly, pffffff :)
I understand now that level reports all pins and I should expect the ones I am interested. Found the JS lib needed to do that. I misleading had this thinking that flags was the gpios bitmask.

The test with the wdog triggers many events as expected.
I did setup gpio only 4 for watch and with my script got:
```
...
{ seqno: 1042, flags: 36, tick: 968758001, level: 2952839583 }
{ seqno: 1043, flags: 36, tick: 968772566, level: 2952839583 }
{ seqno: 1044, flags: 36, tick: 968788046, level: 2952839583 }
{ seqno: 1045, flags: 36, tick: 968802646, level: 2952839583 }
{ seqno: 1046, flags: 36, tick: 968818101, level: 2952839583 }
...
```
which again should say gpio 5 has a timer on.

Lastly for the occasionally occurring event. Flags is 64. This confirms that a wdog/s runs without my knowledge:)

Many thanks!!!

closed 2015-08-13 by ColorfullyMe

ColorfullyMe, 2015-08-13:
@joan2937 

interesting that `pigs wdog 6 0` does not stop the notification with `{ seqno: 1097, flags: 64, tick: 3563337928, level: 2952839599 }` what could be still running?

joan2937, 2015-08-13:
Mea culpa.  I had added a new flag on bit 6 which fires once a minute for notifications if nothing else has happened for that notification in the last minute.

It's PI_NTFY_FLAGS_ALIVE and was created to simplify timekeeping within the piscope application.

I forgot about the flag.  I'll add a description to the documentation.