joan2937/pigpio #19: Feasible Interrupt support?

sgsdxzy, 2015-08-27:
Currently as far as I understand gpioSetAlertFunc() and gpioSetAlertFuncEx() use busy-waiting to implement: gpios are sampled every 5 us by default.
But with newer kernel, in RPi.GPIO and wiringPi there are interrupt support now.  They also use threads, but the thread is waken up by the kernel when kernel detects an interrupt.
The problem is kernel<->userland content switch is costy on Linux.
So it is feasible to implement interrupt support in pigpio?

joan2937, 2015-08-27:
It is incorrect to say the callbacks created by gpioSetAlertFunc() and gpioSetAlertFuncEx() use busy waiting.  DMA is used to sample the GPIOs at zero CPU cost.  Once a millisecond the latest samples are searched for changes, i.e. 200 samples are checked (at the default 200 000 samples per second rate).  Any changes result in the registered callbacks being called.

There is absolutely nothing to stop you using interrupts within pigpio.  I have done so myself.

Interrupts will give less latency (circa 50µs rather than circa 500 µs for callbacks).  However that latency gain comes at a cost, 1) you don't know when the interrupt occurred, it is not time-stamped, 2) you will miss short events in quick succession, and 3) the overhead of processing multiple interrupts per second soon uses more CPU than the sampling method.

I would need to be persuaded that the potential latency gain has a significant use.  I've played quite a lot with the the GPIOs and sensors on the Pi - I can't remember any problems where the answer was use interrupts because they have less latency.

sgsdxzy, 2015-08-27:
I want to implement quadcopter flight control on raspberry pi 2.  However I think 500 microsecond is good enough even for a quadcopter(not a satellite!). As a experience, it's even safe to update the whole system at 50Hz.
If Interrupts will create a lot of overhead, then I suppose it's not necessary.

closed 2015-08-28 by sgsdxzy

joan2937, 2015-10-02:
I have added interrupt support to the C library.  It is in a form modelled on the alert functions.

[gpioSetISRFunc](http://abyz.co.uk/rpi/pigpio/cif.html#gpioSetISRFunc) and [gpioSetISRFuncEx](http://abyz.co.uk/rpi/pigpio/cif.html#gpioSetISRFuncEx) are provided.

The latency is around 50 µs as opposed to 500 µs for alerts.  For relative event timings ISRs are several orders of magnitude worse than alerts as would be expected.  As the kernel doesn't time-stamp interrupts the tick (system microsecond clock) is used as the time-stamp (the same tick as used for alerts).  If a time-out value, x,  greater than 0 is used the callback will be called every x milliseconds after the last interrupt.

sgsdxzy, 2015-10-03:
Good job joan, I will give the interrupt a try.
The DMA method working at an 200Hz interrupt generated by motion sensors is really good enough for my case.