joan2937/pigpio #21: Cant initialise

fanatictentacle, 2015-08-31:
Calling gpioInitialise from c results in following error:

initAllocDMAMem: mbox open failed(No such device or address)

I am using the latest version from 30-08-2015

joan2937, 2015-08-31:
Could you start the pigpio daemon (sudo pigpiod) and report its version (pigs pigpv).  If the daemon refuses to start could you use the -a1 option (sudo pigpiod -a1).

Do the dates of the libraries in the build directory (ls -l lib*.so) match those in /usr/local/lib?

It's probably best to delete old style versions of the library from /usr/local/lib (sudo rm /usr/local/lib/libpig*.a).

Could you report the output of uname -a, cat /etc/os-release, and grep vision /proc/cpuinfo?

fanatictentacle, 2015-09-01:
I will look into it on Thursday, thanks for the quick response

Mit freundlichen Grüßen,

Bruno Rawohl

Der Inhalt dieses elektronischen Schreibens ist vertraulich und
ausschließlich
für den bezeichneten Adressaten bestimmt. Wenn Sie nicht der vorgesehene
Adressat dieses elektronischen Schreibens oder dessen Vertreter sein
sollten,
so beachten Sie bitte, dass jede Form der mündlichen oder schriftlichen
Weitergabe des Inhalts dieses elektronischen Schreibens unzulässig und
rechtswidrig ist. Falls Sie der falsche Adressat sind, dann bitten wir Sie,
sich in einem solchen Fall mit dem Absender des elektronischen Schreibens
unverzüglich in Verbindung zu setzen.

fanatictentacle, 2015-09-02:
pi@raspberrypi ~ $ pigs pigpv
37

Dates and sizes of the libraries are a match (build directory vs. usr/local/lib)

pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.1.6+ #810 PREEMPT Tue Aug 18 15:19:58 BST 2015 armv6l GNU/Linux

pi@raspberrypi ~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@raspberrypi ~ $ grep vision /proc/cpuinfo
CPU revision    : 7
Revision        : 0010

joan2937, 2015-09-02:
Did you start the pigpiod with or without the -a1 option.  Without -a1 it uses mailbox to allocate DMA memory (unless you have requested a large buffer), with -a1 it uses pagemap to allocate DMA memory.

Does sudo ./x_pipgpio run?  That is a C program which uses mailbox allocated DMA memory.

fanatictentacle, 2015-09-02:
I tried with both option (with and without -a1)

the result is the same:

2015-09-02 10:07:11 initInitialise: Can't lock /var/run/pigpio.pid
pigpio initialisation failed.

joan2937, 2015-09-02:
>Can't lock /var/run/pigpio.pid

That is a separate issue.  That indicates the pigpio daemon is already running.  Only one instance of the library can be running at any one time.

To get to a known state make sure the daemon isn't running with the command sudo killall pigpiod.  The lock file should have been deleted.  If not manually delete the lock file (sudo rm /var/run/pigpio.pid).

dwaynez, 2015-09-04:
I am also having a similar problem, since I ran:

sudo apt-get update
sudo apt-get upgrade

Ordinarily, I do not run the daemon. I just call PIGPIO from my program (C program called from Python). When I run under IDLE, no problem. when I run pigpiod, I get the error. When I run with -a1, no error.

I am running version 37.
I put gpioCfgMemAlloc(PI_MEM_ALLOC_PAGEMAP);

before initialize and this seems to have resolved the problem. So can I just leave the code like this or is there something I should be doing so I can go back to mailbox?

joan2937, 2015-09-04:
You can leave the code like that.  Obviously I'd like to know why it doesn't work for some people.

I wonder if there is simply no GPU memory available to be assigned.  It is taken from the portion assigned to the GPU.

I run my PIs headless and generally assign the minimum memory to the GPU.

dwaynez, 2015-09-04:
I also run headless but have not changed the gpu allocation.  I will try that too. I guess something changed with gpu memory usage/allocation with the latest software update.

joan2937, 2015-09-04:
Does /dev/vcio exist?  If it doesn't the code falls back to assuming an earlier kernel and generates its own mailbox, perhaps that assumption is invalid.

Could you detail your uname -a, etc/os-release, Pi version, pigpio version (pigs pigpv) and the exact error message you get.

I am none the wiser as to what could be causing this problem.

dwaynez, 2015-09-04:
I changed the GPU allocation from 64M to 16M and removed the gpioCfgMemAlloc statement and it seems to be working okay.

 

To gather the info you asked for below, I changed it back to 64M and rebooted. Now I cannot recreate the problem. I did not change anything else  In answer to your questions:

 

/dev/vcio exists.

 

uname -a response:

 

Linux raspberrypi 4.1.6+ #810 PREEMPT Tue Aug 18 15:19:58 BST 2015 armv6l GNU/Linux

 

etc/os-release:

 

PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"

NAME="Raspbian GNU/Linux"

VERSION_ID="7"

VERSION="7 (wheezy)"

ID=raspbian

ID_LIKE=debian

ANSI_COLOR="1;31"

HOME_URL="http://www.raspbian.org/"

SUPPORT_URL="http://www.raspbian.org/RaspbianForums"

BUG_REPORT_URL=http://www.raspbian.org/RaspbianBugs

 

Pi Version:

Model B+

 

pigs pigpv:

37

 

Exact error message:

joan2937, 2015-09-04:
Well that's good I suppose.  How did you change the GPU memory allocation?  I wonder if there is some interaction with the format of /boot/config.txt going on.  The Pi doesn't have any other state to affect what you did.

dwaynez, 2015-09-04:
I used raspi-config.

 

When I looked at /boot/config.txt I did not see any allocation statements, so I assume that 64M was the default. After I ran raspi-config it deleted the statement. I have tried it with the statement in and I went in and manually deleted it and I still can't recreate the problem (I did restart each time).

 

Right now there is no GPU statement in config.txt and everything  is working normally. Don't know why.

joan2937, 2015-09-04:
I've used an empty /boot/config.txt and just entries for gpu_mem from 0 to 256.  I can't force the problem.

I see no way of progressing this so I'll ignore the problem until we get more error cases.

garyjohnson, 2015-09-08:
Also started getting this after doing a `sudo apt-get upgrade`. Wondering if this is related to the kernel upgrade that also broke the PiTFT (http://forums.adafruit.com/viewtopic.php?f=50&t=79362) but haven't done any further research. Launching with `-a1` appears to workaround the issue. If there's anything you want me to try out from a diagnostics standpoint let me know.

dwaynez, 2015-09-08:
I started to think more about what I did and I think that while I upgraded to the latest version of PIGPIO, I did not recompile my program. I think after doing a kernel upgrade PIGPIO must be upgraded to the latest version and all programs need to be recompiled.

joan2937, 2015-09-08:
@garyjohnson Have you upgraded pigpio to the latest version?  The initialisation problem was introduced by kernel 4.1.6 and fixed in pigpio version 36 on the 7th of August.

garyjohnson, 2015-09-08:
@joan2937 Yup that was the issue. I recompiled after the upgrade but was working off older code by accident. Thanks.

fanatictentacle, 2015-09-09:
After reading through the other comments, I also checked if recompiling my program would work. The result is: Yes

Joan: Maybe you should put this little advice in the readme: 
After updating the kernel to a version greater than 4.1.6, update pigpio and recompile your programs... since there are more people forgetting this little detail. From my side this issue can be closed

closed 2015-09-09 by fanatictentacle

joan2937, 2015-09-09:
Whenever you link to a library you need to relink your code if the library is changed.  E.g. if your program is linked with an option like -lpigpio you must run the link stage again.  Most times it is simplest to do a complete build.  So each time you download a new version of pigpio you need to recompile your C programs.  There is nothing special about 4.1.6 in this regard.

This doesn't affect programs which use the services of the pigpio daemon.  The pigpio daemon interface remains the same regardless of any library changes.  So Python scripts (and C programs which use the pigpiod_if library) do not need to be rebuilt (unless you want to make use of features added to the daemon).