joan2937/pigpio #23: Hardware revision on raspberry pi 2

PierreRust, 2015-09-23:
On the raspberry pi 2, the hwver command returns 10620993

    pi@raspberrypi ~/tmp $ pigs hwver
    10620993

I get the same result when using the socket interface (directly or with the python client lib).

When I look at `/proc/cpuinfo` I find 1041, which match the information found [here](http://elinux.org/RPi_HardwareHistory)

    pi@raspberrypi ~/tmp $ cat /proc/cpuinfo 
    [...]
    Hardware        : BCM2709
    Revision        : a21041
    Serial          : 00000000030b1d4f

Really not a big deal, I just wanted to let you know.

joan2937, 2015-09-23:
That's because pigs hwver displays the hexadecimal revision number as decimal.

Your revision is actually hexadecimal a21041 (the same as shown on the elinux page).

So pigs hwver shows (decimal) 10620993.

My Pi2 displays (decimal) 10489921 which is hexadecimal a01041.

PierreRust, 2015-09-24:
Damn,  I feel stupid, never realized it was the was thing in hexa...

BTW, I'm porting the gpio python lib to use asyncio instead of threading and I may have a few questions for you ; is here the right place to post these (they are questions, not bugs !)  ?

joan2937, 2015-09-24:
I'll close this as an issue.

Issues don't have to be bugs so no reason you can't ask about asyncio.  I should warn you I probably won't be able to help as it's likely I know less Python than you.

closed 2015-09-24 by joan2937