joan2937/pigpio #25: Interrupt - Edge Falling - Problem

rhowitt, 2015-09-25:
We are using  the PIGPIO python library on a raspberry pi 2 using python 2.7.

Our project has a Raspberry Pi hooked into another PCB that has a micro controller and gpio expander (SX1506).  Both the micro controller and SX1506 send interrupts to the raspberry pi via the Pis GPIO (BCM 12 and 26 ).  The interrupt is to be sensed by the Pi on a falling edge ( 1->0 ).

We found that if those pins are high (1) when the Pi is turned on, PIGPIO will not create an interrupt the first time the pin goes from 1 -> 0.  Once the pin on the  GPIO has transitioned from 0 back to 1, PIGPIO will create an interrupt the next time the pin goes from 1 -> 0.

In the code the GPIO pins are set as an Input, and to start a subroutine procedure when the pin senses a falling edged.

Is there some intialization of the GPIO I am missing?

joan2937, 2015-09-25:
This certainly sounds like an error.

The pigpio daemon should notice the state change and send an update to the Python module.  I wonder if it's because the Python module has no knowledge of the GPIO state until it's informed of the change so sees the first report as the initial state.

I'll have a look at how it works and get back to you.

joan2937, 2015-09-25:
Okay.  In the Python module the GPIO levels are defaulted to all LOW as far as callbacks are concerned.  So the first transition to LOW will be missed.

I'll change the Python to default the levels to the current levels for the next release.

If you want to incorporate the change now edit pigpio.py and search for "Runs the notification thread" (without the quotes).

Change line

lastLevel = 0

to

lastLevel = _pigpio_command(self.control,  _PI_CMD_BR1, 0, 0)

rhowitt, 2015-09-25:
Thank you for the quick response.  I will try the patch.

 

Rick Howitt

joan2937, 2015-10-02:
Added to V38.

closed 2015-10-02 by joan2937

rhowitt, 2015-10-20:
New Problem. It is not with pigpio, but hope you can shed some light.  This one has to do with I2C.  Same project as above.  The system has the following devices on the I2C bus 2 LCD (20 char x 4 lines ) , Humidity, pressure , SX1506 ( gpio expansion ) and the pcb micro.  The I2C bus is at baudrate=100000. There have been strange characters on the LCD , the humidity probe has random errors, and there are random errors with the sx1506 and the pcb micro.  We know the Adafruit's LCD which is using the I2C backback is a heavy user of the I2C. When it is writting to the LCD, it looks like there are about 2000 messages per second.( we put an I2C analyzer on the bus ).  

We are wondering if the python drivers (SMBUS ) for the I2C bus has a timing problem and if so is there a baudrate value that would minimize or eliminate the errors?  Or should we be using a different driver?

joan2937, 2015-10-20:
Are you making all the I2C calls from the same thread?  It shouldn't make a difference, I'm just wondering.

I only use I2C when I'm playing with a device and have no real-world experience of expected error rates.

I'd be tempted to remove one device in turn to see if the errors stop.

You can set the I2C bit rate with a device tree setting in /boot/config.txt.  There should be details in /boot/overlays/README.