joan2937/pigpio #26: Get the current pull up or pull down state for a pin

fivdi, 2015-10-14:
The gpioSetPullUpDown function can be used to set or clear resistor pull up or down on a gpio. Is there a function which can be used to get the current pull up or pull down state for a pin?

joan2937, 2015-10-14:
There is no way of reading the settings of the GPIO internal pulls.

They are set at power-on (GPIO 0-9 pull-up, the rest pull-down) but there is no way of knowing what software may affect the state before your program runs.  You will have to explicitly set the state.

The pull-ups/downs are quite weak, about 50k, so if you need a fixed state use an external resistor of an appropriate value.

fivdi, 2015-10-14:
I should have read the BCM2835 ARM Peripherals datasheet  before opening the issue :) All the information is there:

> Note that it is not possible to read back the current Pull-up/down settings and so it is the
users’ responsibility to ‘remember’ which pull-up/downs are active. The reason for this is
that GPIO pull-ups are maintained even in power-down mode when the core is off, when
all register contents is lost.

Thanks.

closed 2015-10-14 by fivdi