joan2937/pigpio #27: bad ISR initialisation

fivdi, 2015-10-19:
The goal of the program below is the following:

* detect interrupts on GPIO 4 for one seconds with isr1
* disable interrupts on GPIO 4 for one second
* detect interrupts on GPIO 4 for one seconds with isr2

However, `gpioSetISRFunc(GPIO, EITHER_EDGE, 0, NULL)`, which I think should disable interrupts is returning -123 which means `bad ISR initialisation`.

The output of the program is:
```
version: 38
isr1 - gpio: 4, level: 2
isr1 - gpio: 4, level: 2
isr1 - gpio: 4, level: 2
isr1 - gpio: 4, level: 2
rc: -123
```
Am I doing something incorrectly?

Additional info:
```
pi@raspberrypi ~/dev/pigpioc $ cat /etc/debian_version
7.8
pi@raspberrypi ~/dev/pigpioc $ uname -a
Linux raspberrypi 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux
pi@raspberrypi ~/dev/pigpioc $ 
```
```c
#include <unistd.h>
#include <stdio.h>
#include <pigpio.h>

#define GPIO 4

void isr1(int gpio, int level, uint32_t tick) {
  printf("isr1 - gpio: %d, level: %d\n", gpio, level);
}

void isr2(int gpio, int level, uint32_t tick) {
  printf("isr2\n");
}

int main(int argc, char *argv[]) {
  int version;
  int rc;

  if ((version = gpioInitialise()) < 0) {
    return -1;
  }

  printf("version: %d\n", version);

  if (gpioSetISRFunc(GPIO, EITHER_EDGE, 200, isr1) != 0) {
    return -1;
  }

  sleep(1);

  if ((rc = gpioSetISRFunc(GPIO, EITHER_EDGE, 0, NULL)) != 0) {
    printf("rc: %d\n", rc); // program prints -123 here which means
    return -1;              // bad ISR initialisation 
  }

  sleep(1);

  if (gpioSetISRFunc(GPIO, EITHER_EDGE, 200, isr2) != 0) {
    return -1;
  }

  sleep(1);

  gpioTerminate();
}
```

joan2937, 2015-10-20:
That is just a plain error on my part.

The erroneous error is because of an incorrect check when the GPIO is unexported.  The unexport itself will be fine.  However a consequential error means the GPIO is still marked exported internally and the software will not set up a new ISR.

The line

 if (err != sizeof(buf)) return PI_BAD_ISR_INIT;
 
should be

 if (err != strlen(buf)) return PI_BAD_ISR_INIT;
 
in the intGpioSetISRFunc function.

I'll correct the software in the next release (probably before 27th Oct 15).

fivdi, 2015-10-20:
Great, thanks.

joan2937, 2015-10-28:
Implemented in V39.

closed 2015-10-28 by joan2937

fivdi, 2015-10-31:
Thanks for fixing this. Everything looks a lot better now and the issue has been resolved. However, there is a new issue.

Here's the output of the first run of the test program from the original post:

```
version: 39
isr1 - gpio: 4, level: 2
isr1 - gpio: 4, level: 2
isr1 - gpio: 4, level: 2
isr1 - gpio: 4, level: 2
isr2
isr2
isr2
isr2
2015-10-31 07:08:41 sigHandler: Unhandled signal 11, terminating
```

Here's the output of the second run:

```
2015-10-31 07:09:23 initMboxBlock: init mbox zaps failed
```

As can be seen, everything isn't quite right yet. However, this is a different problem so I'll open a new issue for it.

This problem can be solved by calling `gpioSetISRFunc(GPIO, EITHER_EDGE, 0, NULL)` directly before calling `gpioTerminate()`.