joan2937/pigpio #28: timeout when setting a callback

dan-cristian, 2015-10-20:
Hi Joan,

I hope you can help with this issue.

On raspberry pi 2 I have installed pigpio and python binding and I am trying to setup an input callback function. In few seconds after setting the callback I get the following error:

Oct 21 01:02:30 pi-face userspaceServices[6329]: 2015-10-21 01:02:30,846 haiot INFO __init__:setup_in_ports Set pincode=11 type=pi-stdgpio index=11 as input
Oct 21 01:02:40 pi-face userspaceServices[6329]: Traceback (most recent call last):
Oct 21 01:02:40 pi-face userspaceServices[6329]: File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
Oct 21 01:02:40 pi-face userspaceServices[6329]: self.run()
Oct 21 01:02:40 pi-face userspaceServices[6329]: File "/home/haiot/PYC/venv/local/lib/python2.7/site-packages/pigpio.py", line 908, in run
Oct 21 01:02:40 pi-face userspaceServices[6329]: buf = self.sl.s.recv(MSG_SIZ)
Oct 21 01:02:40 pi-face userspaceServices[6329]: timeout: timed out

My init code is as follows:
--------------
import pigpio
__pi = pigpio.pi()
__pi.set_mode(11, pigpio.INPUT)
 __callback.append(__pi.callback(user_gpio=11,
                                                edge=pigpio.EITHER_EDGE, func=input_event))
...

def input_event(gpio, level, tick):
    Log.logger.info("Received pigpio input gpio={} level={} tick={}".format(gpio, level, tick))
------------------
Command pigs MG 11 works fine:
root@pi-face:~# pigs MG 10
4

joan2937, 2015-10-21:
I changed the snippet as follows so I had a complete program to test.

It appears to work fine.

You can test using pigs

pigs pfs 11 0 # Set PWM frequency to minimum
pigs p 11 128 # Send pulses

Does it work for you?  If so there must be something else in your environment causing the problem.

```python
import pigpio
import time

def input_event(gpio, level, tick):
   print("Received pigpio input gpio={} level={} tick={}".format(gpio, level, tick))

__callback = []
__pi = pigpio.pi()
__pi.set_mode(11, pigpio.INPUT)
__callback.append(__pi.callback(user_gpio=11, edge=pigpio.EITHER_EDGE, func=input_event))

while True:
   time.sleep(0.1)
```

dan-cristian, 2015-10-21:
I have tried your example and it works fine. 
The only difference I have on my code is that I do not have a while True: time.sleep(0.1) loop.
I use instead a worker thread pool that is running other tasks in parallel. Is this the problem? Does pigpio expect the thread that set up the callbacks to be idling in a time.sleep manner?

joan2937, 2015-10-21:
The pigpio Python module consists of two threads - the main thread and a callback thread.  The callback thread receives all GPIO updates and calls the registered callbacks.

I don't know enough about the workings of Python to know what might be going wrong.

Could you generate a small, complete, script which demonstrates the fault?  I could try running that.

dan-cristian, 2015-10-21:
Yes, I'm trying to build that script now, thanks for the fast response.

dan-cristian, 2015-10-21:
I found the issue. My app had a timeout set for sockets: socket.setdefaulttimeout(10).
The callback python code on pigpio only works is socket timeout is None, which is true by default:

import socket
socket.setdefaulttimeout(None)

Might be worth changing the pipgio code and implement a socket timeout check when callbacks are defined and issue a warning to the user, or force timeout set to none on socket receive in your code.

Thanks for your support, this is a great piece of code.

joan2937, 2015-10-21:
Thanks for the update.

I have taken the latter option and explicitly set a timeout of None for the two sockets used by pigpio.

If you want to test the change add the following line to pigpio.py just after the two lines containing  socket.socket.

```python
      self.sl.s.settimeout(None)
```
I'll probably release the change in a few days.

dan-cristian, 2015-10-21:
Thanks, this is perfect.
Please also check the FAQ section where I added a question regarding a debounce issue.

PierreRust, 2015-10-21:
I wrote a decorator for debouncing, you can find it [here](https://github.com/PierreRust/apigpio/blob/master/apigpio/utils.py). Everything else in this repo is an in-progess asyncio pigdiod client, not ready for anything serious, but the Debouncer should be usable directly with pigpio official client.

However, given the amount of events you have in your logs, I suspect you have something else going one ; I often have some 'bouncing' too, but never that much.

dan-cristian, 2015-10-22:
Thanks Pierre. I used your decorator and I get the same large amount of debounce. See the code below and the output, same result with both Pull Up and Pull Down inputs. It seems the decorator does not work also because I increased the threshold to a large value and still get noise. What else can be wrong?

.................

import functools
import pigpio
import time
import socket

def Debounce(threshold=150000):
    """
    Simple debouncing decorator for apigpio callbacks.
    Example:
    `@Debouncer()
     def my_cb(gpio, level, tick)
         print('gpio cb: {} {} {}'.format(gpio, level, tick))
    `
    The threshold can be given to the decorator as an argument (in millisec).
    This decorator can be used both on function and object's methods.
    """
    threshold = threshold/100

    class _decorated(object):

        def __init__(self, pigpio_cb):
            self._fn = pigpio_cb
            self.last = 0
            self.is_method = False

        def __call__(self, *args, **kwargs):
            if self.is_method:
                tick = args[3]
            else:
                tick = args[2]
            if tick - self.last > threshold:
                self._fn(*args, **kwargs)
                self.last = tick

        def __get__(self, instance, type=None):
            # with is called when an of `_decorated` is used as a class
            # attribute, which is the case when decorating a method in a class
            self.is_method = True
            return functools.partial(self, instance)

    return _decorated

@Debounce()
def input_event(gpio, level, tick):
   print("Received pigpio input gpio={} level={} tick={}".format(gpio, level, tick))

print socket.getdefaulttimeout()
__callback = []
__pi = pigpio.pi()
__pi.set_mode(18, pigpio.INPUT)
__pi.set_mode(21, pigpio.INPUT)
__pi.set_pull_up_down(18, pigpio.PUD_UP)
__pi.set_pull_up_down(21, pigpio.PUD_DOWN)

__callback.append(__pi.callback(user_gpio=18, edge=pigpio.EITHER_EDGE, func=input_event))
__callback.append(__pi.callback(user_gpio=21, edge=pigpio.EITHER_EDGE, func=input_event))

while True:
   time.sleep(0.1)

......................................................

Received pigpio input gpio=18 level=0 tick=70745424
Received pigpio input gpio=18 level=1 tick=70860599
Received pigpio input gpio=18 level=0 tick=73027570
Received pigpio input gpio=18 level=1 tick=73300196
Received pigpio input gpio=18 level=0 tick=74736634
Received pigpio input gpio=18 level=1 tick=74738584
Received pigpio input gpio=18 level=1 tick=74740764
Received pigpio input gpio=18 level=1 tick=75071585
Received pigpio input gpio=18 level=0 tick=75340141
Received pigpio input gpio=18 level=1 tick=75476276
Received pigpio input gpio=18 level=0 tick=75477901
Received pigpio input gpio=18 level=0 tick=75690042
Received pigpio input gpio=18 level=1 tick=75738132
Received pigpio input gpio=18 level=0 tick=75912407
Received pigpio input gpio=18 level=1 tick=75981697
Received pigpio input gpio=18 level=0 tick=76093448
Received pigpio input gpio=18 level=1 tick=76153358
Received pigpio input gpio=18 level=0 tick=76488299
Received pigpio input gpio=18 level=1 tick=76505889
Received pigpio input gpio=21 level=1 tick=93135937
Received pigpio input gpio=21 level=0 tick=93137746
Received pigpio input gpio=21 level=0 tick=93139736
Received pigpio input gpio=21 level=0 tick=93141636
Received pigpio input gpio=21 level=0 tick=93153336
Received pigpio input gpio=21 level=0 tick=93190502
Received pigpio input gpio=21 level=0 tick=93198677
Received pigpio input gpio=21 level=0 tick=93200852
Received pigpio input gpio=21 level=0 tick=93202492
Received pigpio input gpio=21 level=1 tick=93204267
Received pigpio input gpio=21 level=1 tick=96038584
Received pigpio input gpio=21 level=0 tick=96044889
Received pigpio input gpio=21 level=0 tick=96055539
Received pigpio input gpio=21 level=0 tick=96061779
Received pigpio input gpio=21 level=0 tick=96063954
Received pigpio input gpio=21 level=0 tick=96066079
Received pigpio input gpio=21 level=0 tick=96067859
Received pigpio input gpio=21 level=0 tick=96069979
Received pigpio input gpio=21 level=0 tick=96074749
Received pigpio input gpio=21 level=0 tick=96076529
Received pigpio input gpio=21 level=0 tick=96080759
Received pigpio input gpio=21 level=0 tick=96098959
Received pigpio input gpio=21 level=0 tick=96104624
Received pigpio input gpio=21 level=0 tick=96106834
Received pigpio input gpio=21 level=0 tick=96108349
Received pigpio input gpio=21 level=0 tick=96111149
Received pigpio input gpio=21 level=0 tick=96116759
Received pigpio input gpio=21 level=0 tick=96121029
Received pigpio input gpio=21 level=0 tick=96122799
Received pigpio input gpio=21 level=0 tick=96124424
Received pigpio input gpio=21 level=0 tick=96125984
Received pigpio input gpio=21 level=0 tick=96128649
Received pigpio input gpio=21 level=0 tick=96131984
Received pigpio input gpio=21 level=0 tick=96134409
Received pigpio input gpio=21 level=0 tick=96135954
Received pigpio input gpio=21 level=1 tick=98772141
Received pigpio input gpio=21 level=0 tick=98788886
Received pigpio input gpio=21 level=0 tick=98790472
Received pigpio input gpio=21 level=0 tick=98793891
Received pigpio input gpio=21 level=0 tick=98796031
Received pigpio input gpio=21 level=0 tick=98800846

PierreRust, 2015-10-22:
My mistake, I had an error in the threshold setting. Could you try with the updated code ?

As a side note for @joan2937, when I remove the debouncer, I often see several callbacks with exactly the same tick, like this: 

    on_input 18 1 2592497675
    on_input 18 1 2592497675

I'm wondering if that is a problem on pigpio side (maybe not-unregistered subscription that are still alive ?), I'll try to investigate this.

joan2937, 2015-10-22:
@dan-cristian 

That is an awful lot of bounce.  The microseconds between edges on your first comment.

> 5, 5, 5, 20, 15, 15, 154141, 5, 5, 9721380, 35, 15, 230, 5, 15, 5, 30, 35, 5, 30, 5, 10, 5, 5, 30, 15, 5, 10, 5, 5, 10, 10, 5, 5, 5, 20, 5, 5.

I take it this is a mechanical button.  Have you tried simple hardware debounce? A small capacitor between ground and the GPIO perhaps?

@PierreRust 

I have no memory of seeing repeated events.  The callback thread (should) only calls a registered callback if a level change has happened.

The relevant code in pigpio.py
```python
      while self.go:

         buf = self.sl.s.recv(MSG_SIZ)

         while self.go and len(buf) < MSG_SIZ:
            buf += self.sl.s.recv(MSG_SIZ-len(buf))

         if self.go:
            seq, flags, tick, level = (struct.unpack('HHII', buf))

            if flags == 0:
               changed = level ^ lastLevel
               lastLevel = level
               for cb in self.callbacks:
                  if cb.bit & changed:
                     newLevel = 0
                     if cb.bit & level:
                        newLevel = 1
                     if (cb.edge ^ newLevel):
                         cb.func(cb.gpio, newLevel, tick)
            else:
               if flags & NTFY_FLAGS_WDOG:
                  gpio = flags & NTFY_FLAGS_GPIO
                  for cb in self.callbacks:
                     if cb.gpio == gpio:
                        cb.func(cb.gpio, TIMEOUT, tick)
```

PierreRust, 2015-10-22:
Sorry, it seems to be something wrong with my asyncio-based client, not yours. In my case I often receive several notifications on the socket, with seqno being the only value changing among these notifications. I'll want might be going wrong but it's on my side.

dan-cristian, 2015-10-23:
Yes, it is a push down button. I will try to add a capacitor this weekend to see if that helps.

 

Thanks,

Dan

dan-cristian, 2015-10-23:
I have built an async debouncer that works apparently well after some quick tests.
The code launches a thread on first IO state change and announce the change, then waits a certain period for input to settle and announce the 2nd change. Should be thread safe and relatively small CPU usage.

```python
import time
from threading import Thread, Lock

__import_ok = False
initialised = False
__callback = []
__pi = None
__callback_thread = None
__pin_tick_dict = {}  # {InputEvent}
__lock_dict = {}  # lock list for each gpio

class Log():

    class Logger():
        def info(self, text):
            print(text)

    logger = Logger()

try:
    import pigpio
    __import_ok = True
except Exception, ex:
    __import_ok = False

class InputEvent:
    __seed_id = 0

    def __init__(self, gpio, level, tick):
        self.id = InputEvent.__seed_id
        InputEvent.__seed_id += 1
        self.tick = tick
        self.level = level
        #self.first_level = None
        self.gpio = gpio
        self.event_count = 0
        self.processed = False

    def __repr__(self):
        return 'gpio={} level={} prev= tick={} count={} processed={}'.format(self.gpio, self.level, self.tick,
                                                                               self.event_count, self.processed)

def announce_event(event):
    Log.logger.info("DISPATCH IN io={} lvl={} count={} tick={}".format(event.gpio, event.level, event.event_count,
                                                                       event.tick))
    #dispatcher.send(Constant.SIGNAL_GPIO, gpio_pin_code=event.gpio,
    #                direction=Constant.GPIO_PIN_DIRECTION_IN,
    #                pin_value=event.level, pin_connected=(event.level == 0))

# executed by a haiot thread (not by gpiopd thread)
def check_notify_event(event):
    print("Debounce thread started for event {}".format(event))
    global __pin_tick_dict, __pi, __callback_thread, __lock_dict
    all_events_processed = False
    while not all_events_processed:
        time.sleep(0.1)  # latency in detecting changes
        current_tick = __pi.get_current_tick()
        all_events_processed = True
        for event in __pin_tick_dict.values():
            lock = __lock_dict[event.gpio]
            lock.acquire()
            try:
                # debounce period
                if not event.processed and (current_tick - event.tick > 100000):
                    # event old enough to not be a noise event
                    announce_event(event)
[... 61 lines cut ...]
   time.sleep(0.1)

```

output:
DISPATCH IN io=18 lvl=0 count=1 tick=1781472616
Debounce thread started for event gpio=18 level=1 prev= tick=1781554791 count=2 processed=False
DISPATCH IN io=18 lvl=1 count=2 tick=1781554791
Debounce thread exit

DISPATCH IN io=18 lvl=0 count=1 tick=1784129467
Debounce thread started for event gpio=18 level=1 prev= tick=1784202728 count=2 processed=False
DISPATCH IN io=18 lvl=1 count=6 tick=1784202753
Debounce thread exit

DISPATCH IN io=18 lvl=0 count=1 tick=1785564236
Debounce thread started for event gpio=18 level=1 prev= tick=1785616576 count=2 processed=False
DISPATCH IN io=18 lvl=1 count=2 tick=1785616576
Debounce thread exit

DISPATCH IN io=18 lvl=0 count=1 tick=1787771322
Debounce thread started for event gpio=18 level=1 prev= tick=1787771407 count=2 processed=False
DISPATCH IN io=18 lvl=0 count=5 tick=1787794357
Debounce thread exit
DISPATCH IN io=18 lvl=1 count=1 tick=1788981770

PierreRust, 2015-10-24:
I might have missed something but I don't understand why you would need a thread for debouncing (sounds overkill and expensive). Doesn't my debouncer decorator work in your case ?

dan-cristian, 2015-10-25:
Maybe there is a better way? 

I need the thread to ensure I do not miss the last input event received from the board.

Assuming we get these input events:
tick 0, level 1 (processed)
tick 3, level 0 (ignored)
tick 5, level 0 (processed)
tick 10, level 1 (pocessed)
tick 11, level 0 (ignored, MISSED)

If we only take the level changes with a 5 step tick, the last event will be missed. By using a thread that is started once an event occur, I can ask the thread to wait a bit more and process also the last event. There will maximum 1 thread running that will parse all IO input events, and will be shutted down if there is no activity, so seems like a small overhead comparing with the high CPU utilisation induced by pipgiod daemon .

Here is a god whitepaper on debouncing:
http://www.eng.utah.edu/~cs5780/debouncing.pdf

joan2937, 2015-10-28:
The root problem (socket timeout) has been fixed in V39.

There is now a function which (as a side effect) may be used to debounce.

Python [set_glitch_filter](http://abyz.co.uk/rpi/pigpio/python.html#set_glitch_filter), C [gpioGlitchFilter](http://abyz.co.uk/rpi/pigpio/cif.html#gpioGlitchFilter) may be used to ignore edges shorter than x microseconds.  Be aware that as a consequence events on that GPIO will be timestamped x microseconds after they were first detected.

closed 2015-10-28 by joan2937