joan2937/pigpio #29: High CPU utilisation on idle

dan-cristian, 2015-10-23:
I notice that, while doing nothing (no I/O events) and no client connected,  pigpiod daemon jumps from time to time up to 15% CPU on Raspberry PI 2, and very often stays between 5.7% - 7.6%.

The usage seems too high, do you know how can be reduced?

Thanks,
Dan

HeatfanJohn, 2015-10-24:
I see 10% most of the time on a gen 1 Pi.  I've been told that the daemon is constantly polling the status of all the gpio lines.  That said I wonder if there's a way to have an option to only poll if some request to read the gpio has been performed?

dan-cristian, 2015-10-25:
Exactly, this will be a very good optimisation.

dan-cristian, 2015-10-25:
I'm not very familiar with Broadcom BCM2835 chip low level programming but I assumed pigpio uses interrupts and will not need to perform pooling, isn't that the case?

https://blog.mayflower.de/2073-Using-GPIO-on-Raspberry-Pi.html

joan2937, 2015-10-25:
pigpio is continually sampling GPIO 0-31 at 200 thousand times a second by default.

You simply can not capture the data captured by pigpio by using any other method (known to me).

Using interrupts would fail on a number of counts, 1) the GPIO edges would not be accurately timestamped, 2) short (circa 100 µs or less) GPIO edges in succession would be lost, 3) the CPU would be spending 100% of its time handling interrupts.

The only area where interrupts score is in better average latency, 50 µs versus 500 µs for the batched sample approach of pigpio.

The sampling itself is done at 0 CPU cost.  The CPU time is spent going through the samples and collating GPIO changes.

At the default 5 µs sampling rate 7.5% CPU is a good average.  Remember that the Pi2 has four CPUs so pigpio is using about 2% of the available power.

dan-cristian, 2015-10-25:
Thanks for clarifications. 
Is there a way to reduce the number of GPIO sampling loops per second, less than 200.000? My app will not need this level of sensitivity, I only look for button presses.

joan2937, 2015-10-26:
You could start the daemon with 10 µs sampling (sudo pigpiod -s 10).  It'll make a difference but not a great deal.

If all you want to do with the GPIO is detect button presses you would be better off with one of the other GPIO libraries which don't have pigpio's sampling overhead.

dan-cristian, 2015-10-26:
Thanks, I will try.

I really like pgpiod because it does not need root access right for the
client app, seems to be the only one, compared with other libraries out
there, so I'll stick with it :-)

dan-cristian, 2015-10-26:
I tried with parameter -s 10 and CPU usage dropped to 2.9 - 3.9% on my PI 2.
pigpiod is still very sensitive to button state changes so I think my app will work well enough with an even lower sensitivity, thus decreasing CPU usage even more.

Any chance to add more sensitivity levels above 10µs?

joan2937, 2015-10-28:
I'll think about adding a 20 µs sampling rate.  It will not be a priority though and I'll only add it if it causes pretty much no disruption to the existing code.

dan-cristian, 2015-10-28:
understood, thanks.

paviro, 2015-12-25:
Is it possible to set the sampling rate to something close to 0 if you only write pins but don't read any pin?

joan2937, 2015-12-26:
@paviro You could probably find a more appropriate module, e.g. perhaps servoblaster, if all you are interested in is PWM.

joan2937, 2016-02-29:
I shall not be adding a lower sampling rate.  If the sampling overhead is too much for the application then pigpio probably isn't the right library to use.

closed 2016-02-29 by joan2937

joan2937, 2018-04-15:
You could try the -m option to the pigpio daemon as that will disable alert sampling.   If you are not using the daemon you could use the gpioCfgInterfaces() option and or in PI_DISABLE_ALERT.

fab1an, 2018-04-20:
Thanks for the tipp. Which version has the `-m` switch? pigpio V64 doesn't seem to have it.