joan2937/pigpio #30: pigpiod + wiegand = incorrect read

VecH, 2015-10-24:
I have the keys wiegand with number (read and verify in other devices) and these numbers are written on the keys themselves:
0009030510
0009487621
0009236067
0009328295
I am git clone pigpiod library, maked and installed
I use modified example: /pigpio/EXAMPLES/Python/WIEGAND_CODE/wiegand.py
```   def callback(bits, value):
      print("bits={} value={}".format(bits, value))
```to```   def callback(bits, value):
      print("bits={} value={}".format(bits, value))
      print(int(bin(value)[3:27],2))
```for remove CRC bits and convert to integer and print

Only first key 0009030510 correct read and print
```bits=26 value=51615453
9030510```
[u]other keys returns an incorrect value[/u]
0009487621 return 2198027
0009236067 return 1694918
0009328295 return 1879374

I tried to understand what the problem is, not work
translation into binary values, is evident that they are shifted
```0009030510=9030510
100010011100101101101110
100010011100101101101110
---------
0009487621=2198027
100100001100010100000101
   1000011000101000001011
---------
0009236067=1694918
100011001110111001100011
    110011101110011000110
---------
0009328295=1879374
100011100101011010100111
    111001010110101001110```

joan2937, 2015-10-25:
I don't have any Wiegand devices so can't really help out on this.  When I wrote the software I generated Wiegand data from a TI Launchpad.  I could quite easily have made an off by one mistake.

To be able to help I'd need to have the full bits sent and received as seen by the original pigpio script, i.e. no pre or post processing.

VecH, 2015-10-26:
how to do it, to make it clear to you?

joan2937, 2015-10-26:
Run the Wiegand script from http://abyz.co.uk/rpi/pigpio/examples.html#Python_code.  Don't make any changes other than the GPIO used for the wires.

VecH, 2015-11-02:
http://abyz.co.uk/rpi/pigpio/code/wiegand_py.zip
Just return the wrong values

joan2937, 2015-11-02:
I want to know the full code on the card and the output of the script.

Saying it returns the wrong values adds nothing useful.

If you actually want to find out what (if anything) is going wrong could you also save a piscope trace (in piscope format) of the read?  Could you post that file or a link?

VecH, 2015-11-16:
I solved his problem in a class of wiegand to give the value as string "01010101010101010101" and has more compare how many bits are produced (it should be 26), and only then verifies the CRC and convert to INT
everything is working fine

wiegand conversion inside the class are not always successful, because they get is not always 26 bits

closed 2015-11-16 by VecH

umesh1770, 2016-05-16:
Can u explain where how u did the string output in the weigand class

7aman, 2019-01-24:
My RP is fixed this printing bug.
(https://github.com/joan2937/pigpio/pull/260)

7aman, 2019-01-24:
If you use this script to decode RFID Card and Tags, use this callback function instead

```python
   def callback(bits, value):
      print("bits={} value={:026b}".format(bits, value))
      card_id = int("{:026b}".format(value)[1:25],2)
      print("Card ID: {:010d}".format(card_id))
```

DAThosting, 2020-08-28:
> Apparently "format" function used in this "callback" function will convert "value" (currently in binary format) to its "integer representation". Also It will omit leading zeros.
> If you use this script to decode RFID Card and Tags, use this callback function instead
> 
> ```python
>    def callback(bits, value):
>       print("bits={} value={:026b}".format(bits, value))
>       card_id = int("{:026b}".format(value)[1:25],2)
>       print("Card ID: {:010d}".format(card_id))
> ```

This helped us, but why you using [1:25] this should be [0:26] or am I wrong?
This is a substring? Why trimming a 26 char string to 24 char?

Another problem which I have, the pigpio give me the value '00000100100110011010100111' => '0001205927' which is wrong, the chips value is '0034157395'.
In this case the last digit from wiegand value have set to the first, then this is working. ('10000010010011001101010011')

I can't figure out why this problem occurs.

7aman, 2020-08-28:
@DAThosting
It is an old post an I am not sure about details. I never used it again. But according to [wiegand wikipedia page](https://en.wikipedia.org/wiki/Wiegand_interface):  
```
The original Wiegand format had 

 - one parity bit
  - 8 bits of facility code
 - 16 bits of ID code
 - and a trailing parity bit

for a total of 26 bits.  
The first parity bit is calculated from the first 12 bits of the code and the trailing parity bit from the last 12 bits.  
However, many inconsistent implementations and extensions to the basic format exist. 
```

So first and last bits are parity and must be ignored.

pierol, 2021-07-15:
I'm trying to use the example code in python to read a tag rfid with a wiegand reader setup with WG26bit output. The strange fact is that sometimes read 30, 29, 27, 28 bits. This is the output of the python script (not modified) with repeated readings of the same tag rfid:
```bash
bits=29 value=130019798
bits=27 value=33028330
bits=28 value=66056426
bits=27 value=32501994
bits=27 value=32502218
bits=27 value=32504042
bits=26 value=16251114
bits=27 value=32502234
bits=26 value=16251114
bits=26 value=16251114
bits=30 value=260032434

```

pierol, 2021-07-15:
> I'm trying to use the example code in python to read a tag rfid with a wiegand reader setup with WG26bit output. The strange fact is that sometimes read 30, 29, 27, 28 bits. This is the output of the python script (not modified) with repeated readings of the same tag rfid:
> 
> ```shell
> bits=29 value=130019798
> bits=27 value=33028330
> bits=28 value=66056426
> bits=27 value=32501994
> bits=27 value=32502218
> bits=27 value=32504042
> bits=26 value=16251114
> bits=27 value=32502234
> bits=26 value=16251114
> bits=26 value=16251114
> bits=30 value=260032434
> ```

After some more invastigation i found that the problem was a wired wrong connection. I didn't connected the ground of the power supplier of the reader to the raspberrypi ground pin GPIO, now is working fine no more stange bits length.

hutch120, 2023-01-10:
In regard to adding parity bits, the `send_wiegand_code` function could be altered as follows to support writing 26 bit wiegand per the specfication.

```
void send_wiegand_code(
    int pi, long long code, int bits)
{
    int i, pos = 0;
    char chain[128];

    chain[pos++] = 255;
    chain[pos++] = 0; /*Start of wave block. */

    int parityBitStart = pos++;
    int startParityOnesCount = 0;
    int endParityOnesCount = 0;

    for (i = 0; i < bits - 2; i++)
    {
        // fprintf(stdout, "Bit: %i ", i);

        if (code & (1 << (bits - 3 - i)))
        {
            if (i + 1 <= (bits - 2) / 2)
            {
                startParityOnesCount++;
            }
            else
            {
                endParityOnesCount++;
            }

            // fprintf(stdout, "1\n");
            chain[pos++] = wid1;
        }
        else
        {
            // fprintf(stdout, "0\n");
            chain[pos++] = wid0;
        }
    }

    // Add First Parity bit (first half of bits)
    if (startParityOnesCount % 2 == 0)
    {
        chain[parityBitStart] = wid0; 
    }
    else
    {
        chain[parityBitStart] = wid1;
    }

    int parityBitEnd = pos++;

    // Add End Parity bit (second half of bits)
    if (endParityOnesCount % 2 == 0)
    {
        chain[parityBitEnd] = wid0;
    }
    else
    {
        chain[parityBitEnd] = wid1;
    }

    chain[pos++] = 255;
    chain[pos++] = 2; /* Delay optMsgGap. */
    chain[pos++] = optMsgGap & 255;
    chain[pos++] = (optMsgGap >> 8) & 255;

    chain[pos++] = 255;
    chain[pos++] = 1; /* Repeat optRepeats. */
    chain[pos++] = optRepeats & 255;
    chain[pos++] = (optRepeats >> 8) & 255;

    wave_chain(pi, chain, pos);

    while (wave_tx_busy(pi))
        time_sleep(0.05);
}
```