joan2937/pigpio #32: pigpiod shutdown problem (kernel 4.1.12+)

VecH, 2015-11-02:
```root@raspberrypi ~ # uname -a```
```Linux raspberrypi 4.1.12+ #824 PREEMPT Wed Oct 28 16:39:49 GMT 2015 armv6l GNU/Linux```

pigpiod current version, recompiled and reinstall
```make uninstall```
```make clean```
```make install```

```shutdown -h now```
on screen this error ***and computer restarted (not shutdown)***
[image]

joan2937, 2015-11-02:
I don't know what is happening.

I've just upgraded a Pi 2 to 4.1.12-v7+ #824 and don't appear to get the same result.

I've tried four or five times.  Twice it shutdown straight away, the other times it shutdown after 80 seconds or so.  It has not restarted.

If pigpiod receives SIGTERM it will shutdown unless the signal handling has been overriden by the user.

Check with

sudo killall -s TERM pigpiod

I'm not convinced that it's pigpiod holding up the shutdown process.

VecH, 2015-11-16:
I learned what it was to blame for the watchdog if the kernel module is loaded with the parameter (nowayout = 1)
Excuse me

closed 2015-11-16 by VecH