joan2937/pigpio #33: Notifications with incorrect sequence numbers

fivdi, 2015-11-18:
The effect seen here may be an edge case, or I may be doing something incorrectly, or I may be expecting too much.

The following program starts hardware PWM on GPIO18 at a frequency of 100KHz with a 50% duty cycle. It then activates notifications on GPIO18. This means about 200,000 notifications per second. Every 1,000,000 notifications the program prints the number of notifications so far. If the seqno for a notification is not the expected value an error message is printed.

```c
#include <fcntl.h>
#include <stdio.h>
#include <pigpio.h>

#define LED 18

int main() {
  int version;
  int handle;
  char pipePath[32];
  int fd;
  int bytes;
  int notificationCount = 0;
  gpioReport_t r;
  uint16_t lastSeqno = 0xffff;
  uint32_t lastTick;

  gpioCfgClock(1, 1, 1);

  if ((version = gpioInitialise()) < 0) {
    printf("gpioInitialise error\n");
    return -1;
  }

  printf("version: %d\n", version);

  if (gpioHardwarePWM(LED, 100000, 500000) < 0) {
    printf("gpioHardwarePWM error\n");
    return -1;
  }

  if ((handle = gpioNotifyOpen()) < 0) {
    printf("gpioNotifyOpen error\n");
    return -1;
  }

  if (gpioNotifyBegin(handle, (1 << LED)) < 0) {
    printf("gpioNotifyBegin error\n");
    return -1;
  }

  sprintf(pipePath, "/dev/pigpio%d", handle);

  if ((fd = open(pipePath, O_RDONLY)) < 0) {
    printf("open error\n");
    return -1;
  }

  for (;;) {
    bytes = read(fd, &r, 12);
    if (bytes != 12) {
      printf("read error\n");
      return -1;
    }

    notificationCount++;
    if (notificationCount % 1000000 == 0) {
      printf("%d\n", notificationCount);
    }

    if (((uint16_t) (lastSeqno + 1)) != r.seqno) {
      printf("  sequence error - lastSeqno: %d, seqno: %d\n", lastSeqno, r.seqno);
      printf("    %d\n", notificationCount);
    }

    lastSeqno = r.seqno;
    lastTick = r.tick;
  }

  gpioTerminate();

  return 0;
}
```

If the program is run without any additional load everything looks fine and the output will be something like this:

```
version: 40
1000000
2000000
3000000
4000000
5000000
6000000
7000000
8000000
9000000
10000000
11000000
12000000
13000000
14000000
15000000
16000000
17000000
18000000
19000000
20000000
...
```

However, if there is additional load on the system, the output can look like this:

```
version: 40
1000000
2000000
  sequence error - lastSeqno: 28427, seqno: 28583
    2780941
  sequence error - lastSeqno: 31399, seqno: 32082
    2783758
  sequence error - lastSeqno: 32199, seqno: 33000
    2783876
  sequence error - lastSeqno: 33340, seqno: 33800
    2784217
  sequence error - lastSeqno: 34140, seqno: 34452
    2784558
  sequence error - lastSeqno: 34792, seqno: 35701
    2784899
  sequence error - lastSeqno: 38414, seqno: 39273
    2787613
  sequence error - lastSeqno: 39613, seqno: 40075
    2787954
  sequence error - lastSeqno: 40415, seqno: 40873
    2788295
  sequence error - lastSeqno: 41213, seqno: 42072
    2788636
  sequence error - lastSeqno: 42412, seqno: 42470
    2788977
  sequence error - lastSeqno: 42870, seqno: 43212
    2789378
  sequence error - lastSeqno: 44811, seqno: 44870
    2790978
  sequence error - lastSeqno: 45611, seqno: 46071
    2791720
  sequence error - lastSeqno: 46411, seqno: 46872
    2792061
  sequence error - lastSeqno: 47212, seqno: 47669
    2792402
  sequence error - lastSeqno: 48009, seqno: 48471
    2792743
  sequence error - lastSeqno: 48811, seqno: 49671
    2793084
  sequence error - lastSeqno: 50011, seqno: 50071
    2793425
  sequence error - lastSeqno: 50411, seqno: 50485
    2793766
  sequence error - lastSeqno: 51670, seqno: 52012
    2794952
  sequence error - lastSeqno: 52069, seqno: 52411
    2795010
  sequence error - lastSeqno: 52471, seqno: 52813
    2795071
  sequence error - lastSeqno: 52871, seqno: 53213
    2795130
  sequence error - lastSeqno: 53609, seqno: 54470
    2795527
  sequence error - lastSeqno: 55211, seqno: 55672
    2796269
  sequence error - lastSeqno: 56012, seqno: 56473
    2796610
  sequence error - lastSeqno: 56813, seqno: 56872
    2796951
  sequence error - lastSeqno: 57212, seqno: 57673
    2797292
  sequence error - lastSeqno: 58013, seqno: 58472
    2797633
  sequence error - lastSeqno: 58812, seqno: 58872
    2797974
  sequence error - lastSeqno: 59212, seqno: 59671
    2798315
  sequence error - lastSeqno: 60833, seqno: 61271
    2799478
  sequence error - lastSeqno: 61611, seqno: 62070
    2799819
  sequence error - lastSeqno: 62410, seqno: 62471
    2800160
  sequence error - lastSeqno: 63212, seqno: 63672
    2800902
  sequence error - lastSeqno: 64012, seqno: 64470
    2801243
  sequence error - lastSeqno: 64810, seqno: 64870
    2801584
  sequence error - lastSeqno: 65210, seqno: 134
    2801925
  sequence error - lastSeqno: 474, seqno: 602
    2802266
  sequence error - lastSeqno: 1274, seqno: 1735
    2802939
  sequence error - lastSeqno: 2875, seqno: 3337
    2804080
  sequence error - lastSeqno: 3677, seqno: 3773
    2804421
  sequence error - lastSeqno: 4113, seqno: 4536
    2804762
  sequence error - lastSeqno: 5290, seqno: 5735
    2805517
  sequence error - lastSeqno: 6075, seqno: 6135
    2805858
  sequence error - lastSeqno: 6475, seqno: 6934
    2806199
  sequence error - lastSeqno: 7274, seqno: 7336
    2806540
  sequence error - lastSeqno: 7676, seqno: 8135
    2806881
  sequence error - lastSeqno: 8475, seqno: 8536
    2807222
  sequence error - lastSeqno: 8876, seqno: 9336
    2807563
  sequence error - lastSeqno: 10875, seqno: 11381
    2809103
  sequence error - lastSeqno: 11721, seqno: 12136
    2809444
  sequence error - lastSeqno: 12902, seqno: 12935
    2810211
  sequence error - lastSeqno: 13275, seqno: 13737
    2810552
  sequence error - lastSeqno: 14077, seqno: 14133
    2810893
  sequence error - lastSeqno: 14473, seqno: 14936
    2811234
  sequence error - lastSeqno: 15276, seqno: 15737
    2811575
  sequence error - lastSeqno: 16077, seqno: 16136
    2811916
  sequence error - lastSeqno: 16476, seqno: 16934
    2812257
  sequence error - lastSeqno: 17676, seqno: 17733
    2813000
  sequence error - lastSeqno: 18073, seqno: 18136
    2813341
  sequence error - lastSeqno: 18476, seqno: 18536
    2813682
  sequence error - lastSeqno: 18876, seqno: 18935
    2814023
  sequence error - lastSeqno: 19275, seqno: 19736
    2814364
  sequence error - lastSeqno: 20519, seqno: 20937
    2815148
  sequence error - lastSeqno: 21377, seqno: 21719
    2815589
  sequence error - lastSeqno: 22074, seqno: 22556
    2815945
3000000
4000000
```

The sequence numbers are often off by a few hundred here. The additional load in this case was an `scp` command which copied a large file from another Linux machine to the Pi. 

Setting the PWM frequency to 200KHz will result in similar sequence number issues without any additional load on the system.

Is this to be expected or am I doing something incorrectly?

fivdi, 2015-11-20:
I tried what you suggested with `pigs csi 512` and the test program from the original post. There were 59 sequence number errors and the stats said that pipe would block 101 times:

```
version: 40
1000000
2000000
3000000
4000000
  sequence error - lastSeqno: 15026, seqno: 15174
    4405940
  sequence error - lastSeqno: 15846, seqno: 15905
    4406613
  sequence error - lastSeqno: 16245, seqno: 16307
    4406954
  sequence error - lastSeqno: 16647, seqno: 16708
    4407295
  sequence error - lastSeqno: 17048, seqno: 17131
    4407636
  sequence error - lastSeqno: 17471, seqno: 17509
    4407977
  sequence error - lastSeqno: 17849, seqno: 18313
    4408318
  sequence error - lastSeqno: 18653, seqno: 18706
    4408659
  sequence error - lastSeqno: 19046, seqno: 19106
    4409000
  sequence error - lastSeqno: 19446, seqno: 19506
    4409341
  sequence error - lastSeqno: 19846, seqno: 19906
    4409682
  sequence error - lastSeqno: 20246, seqno: 20306
    4410023
  sequence error - lastSeqno: 20646, seqno: 20706
    4410364
  sequence error - lastSeqno: 21106, seqno: 21448
    4410765
  sequence error - lastSeqno: 21845, seqno: 22304
    4411163
  sequence error - lastSeqno: 22644, seqno: 23106
    4411504
  sequence error - lastSeqno: 23446, seqno: 24306
    4411845
  sequence error - lastSeqno: 24646, seqno: 25107
    4412186
  sequence error - lastSeqno: 25853, seqno: 26308
    4412933
  sequence error - lastSeqno: 26648, seqno: 26708
    4413274
  sequence error - lastSeqno: 27048, seqno: 27107
    4413615
  sequence error - lastSeqno: 27447, seqno: 27506
    4413956
  sequence error - lastSeqno: 28274, seqno: 28710
    4414725
  sequence error - lastSeqno: 29446, seqno: 29506
    4415462
  sequence error - lastSeqno: 29846, seqno: 29904
    4415803
  sequence error - lastSeqno: 30244, seqno: 30707
    4416144
  sequence error - lastSeqno: 31047, seqno: 31106
    4416485
  sequence error - lastSeqno: 31446, seqno: 31504
    4416826
  sequence error - lastSeqno: 32245, seqno: 32355
    4417568
  sequence error - lastSeqno: 32695, seqno: 33105
    4417909
  sequence error - lastSeqno: 33445, seqno: 33559
    4418250
[... 61 lines cut ...]
following and e-mail to pigpio@abyz.co.uk
pigpio version=40 internals=200
micros=1 allocMode=0 dmaInitCbs=1 DMARestarts=0
samples 5353507 maxSamples 800 maxEmit 800 emitFrags 13315
cbTicks 12336, cbCalls 13373
pipe: good 26586, short 0, would block 101
alertTicks 13350, lateTicks 11, moreToDo 11
        0         0         0         0         0         0         0         0         0         0         0         1         0         0         0         0         0         0         0         1        45       218       902      2792     42388    978341     42108      2716       826       218        55        23        13         4         7         2         0         3         6        12        13        12         9         0         1         0         1         0         0         0 
#####################################################
```

However, as already mentioned, the original test program is silly as it reads the notifications one by one. Reading up to 1000 at the same time gives far better results.

The test code here is implemented in C for demonstration purposes. In reality, I'm using pigpio from a Node.js addon. Unfortunately, reading as many notifications as possible from the pipe in Node.js doesn't really improve things. The pipe is still being filled before it's emptied. If I'm not mistaken, the pipe has the default pipe size which is 65536 bytes. This is enough space for 65536/12=5416 notifications. The Node.js test program receives 80,000 notifications per second (unlike the C program above which receives 200,000 notifications per second). This means the pipe fills in 65536/12/80000=0.068 seconds, or 68 milliseconds. What I have noticed is that increasing the pipe size to 1048576 (the max possible) results in huge improvements for the Node.js code. There are no longer any sequence number errors and 80,000 notifications per second are easily handled.

The pipe size was modified by adding the following code
```
fcntl(fd, F_SETPIPE_SZ, 1048576);
```
directly after [this](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L10103) line of code in pigpio.c
```
fd = open(name, O_RDWR|O_NONBLOCK);
```

Is there a way to configure the size of the pipe used for pigpio notifications?

joan2937, 2015-11-21:
Do you do the open from C?  I've briefly considered adding a parameter to gpioNotifyBegin but I'd rather leave that interface alone.

My current preference is for a new function gpioNotifyBeginWithSize taking a single parameter bufSiz.  0 for default, otherwise byte size.  Would that suit you?

I am unlikely to export that beyond the C interface (at least until and unless I do a general tidy-up).

fivdi, 2015-11-21:
>Do you do the open from C? I've briefly considered adding a parameter to gpioNotifyBegin but I'd rather leave that interface alone.

I'm calling gpioNotifyOpen and gpioNotifyBegin from C++ which is more or less the same as calling them from C.

>My current preference is for a new function gpioNotifyBeginWithSize taking a single parameter bufSiz. 0 for default, otherwise byte size. Would that suit you?

A new function that takes a parameter specifying the pipe size would be excellent and would suit me perfectly. I would have expected the new function to be called gpioNotifyOpenWithSize rather than gpioNotifyBeginWithSize as gpioNotifyBegin starts notifications on a previously opened handle, already has two parameters, and can be called multiple times for the same handle.

joan2937, 2015-11-21:
The name was a typo on my part.

Provisionally I'll make the following change.

```C
int gpioNotifyOpenWithSize(int bufSize)
{
   int i, slot, fd;
   char name[32];

   DBG(DBG_USER, "");

   CHECK_INITED;

   slot = -1;

   for (i=0; i<PI_NOTIFY_SLOTS; i++)
   {
      if (gpioNotify[i].state == PI_NOTIFY_CLOSED)
      {
         gpioNotify[i].state = PI_NOTIFY_OPENED;
         slot = i;
         break;
      }
   }

   if (slot < 0)
      SOFT_ERROR(PI_NO_HANDLE, "no handle");

   sprintf(name, "/dev/pigpio%d", slot);

   myCreatePipe(name, 0664);

   fd = open(name, O_RDWR|O_NONBLOCK);

   if (fd < 0)
   {
      gpioNotify[slot].state = PI_NOTIFY_CLOSED;
      SOFT_ERROR(PI_BAD_PATHNAME, "open %s failed (%m)", name);
   }

   if (bufSize != 0)
   {
      i = fcntl(fd, F_SETPIPE_SZ, bufSize);
      if (i != bufSize)
      {
         gpioNotify[slot].state = PI_NOTIFY_CLOSED;
         SOFT_ERROR(PI_BAD_PATHNAME,
            "fcntl %s size %d failed (%m)", name, bufSize);
      }
   }

   gpioNotify[slot].seqno = 0;
   gpioNotify[slot].bits  = 0;
   gpioNotify[slot].fd    = fd;
   gpioNotify[slot].pipe  = 1;
   gpioNotify[slot].max_emits  = MAX_EMITS;
   gpioNotify[slot].lastReportTick = gpioTick();

   closeOrphanedNotifications(slot, fd);

   return slot;
}

int gpioNotifyOpen(void)
{
   return gpioNotifyOpenWithSize(0);
}
```

fivdi, 2015-11-21:
Excellent, thank you very much. I'll add the modification to my installation and let you know how it goes.

fivdi, 2015-11-21:
I added the modifications to my installation and things are looking good :)

Two tests were performed from Node.js with bufSize set to 1048576.

The first test was 80,000 notifications per second with additional load (scp 4GB of files from a PC to the Pi 2.) Everything went well, there were no sequence number errors, every level change appears to have been detected, and the stats said that pipe would block was 0. The test ran for about 9 minutes. 44,014,932 notifications were processed.

The second test was 200,000 notifications per second without additional load. Everything went well, there were no sequence number errors, every level change appears to have been detected, and the stats said that pipe would block was 0. The test ran for a little over 8 minutes. 100,000,307 notifications were processed.

fivdi, 2015-11-22:
As expected, the modified version of gpioNotifyOpen functions correctly too.

joan2937, 2015-11-22:
I'll probably push out another release later on today (Sunday).

joan2937, 2015-11-22:
Proposed change added to V41.

By the way they seem to be pretty impressive figures you are getting with Node.js.

One word of caution, I wouldn't recommend the use of the 1µs sampling rate in general.  It's something I would only use if there was no other way of detecting edges.  It puts an appreciable strain on the system.

fivdi, 2015-11-22:
I just gave it a try and all looks well at 200,000 notifications per second. Up to 300,000 per second should work in Node.js, but there won't be much time left for doing much with them. The 1µs sampling rate was used in the tests as that's what was being used previously. I'll do some tests with other sampling rates.

One thing that I noticed with V41 compared to V40 is that the performance of gpioRead and gpioWrite appears to have dropped by about 50%. With V40 about 1,200,000 calls per second were possible from Node.js. With V41 it's about 600,000 calls per second. I haven't checked to see why this is the case but is there a V41 change that would explain it? It's not something that bothers me and it may be related to the JavaScript/Node.js side of things.

Once again, thanks a million.

closed 2015-11-22 by fivdi

joan2937, 2015-11-22:
No changes were made to gpioRead/gpioWrite.  Unfortunately it's not the sort of thing I benchmark.  Was it just a busy loop spin?  If you give me an idea of how you were calculating the figures I'll see if I can replicate your results.

fivdi, 2015-11-23:
Yes, it was just a busy loop spin in JavaScript, for example here's the test for gpioRead: https://github.com/fivdi/pigpio/blob/master/test/digital-read-performance.js

Is V40 still available for download somewhere? I can double check to see if the figures are correct.

joan2937, 2015-11-23:
I'm sure github has a way, may be simpler to download a zip from http://abyz.co.uk/rpi/pigpio/VERSIONS/

fivdi, 2015-11-23:
Ok, thanks, I'll compare V40 and V41 to see what the figures are.

fivdi, 2015-11-26:
The performance issue isn't related to pigpio. It appears to be related to the number of fields in objects used by the V8 JavaScript engine.

joan2937, 2015-11-26:
Thanks for the update.

SlySven, 2016-02-28:
> Is V40 still available for download somewhere? I can double check to see if the figures are correct.

For anyone who wants to look at this code in the privacy of their own system they can grab a local copy of the code with:

+ `git clone https://github.com/joan2937/pigpio.git ./pigpio/` in your `~/src/` directory then change into the newly built local repository.
+ Start up a copy of `gitk` *other git management applications are available!*.
+ Select the text describing the commit at that point on the history and right click to `Create a branch` - give it a suitable name.
+ Select the branch name (may need to `Update` / `Reload` the view using commands on the `File` menu) - then  right click and `Check out this branch`.

The directory will than contain a local copy of the files as they were when that commit was made...

+ Read up on `git` - noting that `git `{any git command}` --help` should bring up the `man`page in a terminal window {that being a little quicker to type than `man git-{any git command}` if you have command line editing in place and are part way through typing a command that you need help with! :grinning: 

To **store** changes you will also want `git gui` and `git fetch` installed / available - these two and `gitk` can call and will interoperate with each other.

Hope that brief start up guide is of use.

Oh, and if you wish to publish stuff for @joan2937 to use you can use the "Fork" command here to make your own copy here on github and use the `git remote` command to add your `fork` of this repository into your local system and then push to that and invite Joan to use your contributions with a "Pull Request" here.

joan2937, 2016-02-28:
You should be able to get particular versions using the appropriate git commit hash.  Alternatively I publish the versions at http://abyz.co.uk/rpi/pigpio/VERSIONS/ as ordinary zip/tar files.