joan2937/pigpio #34: DHT11/DHT22 pigpio Anomalies with Jessie Environment

baqwas, 2015-11-20:
The code contribution at https://github.com/joan2937/pigpio/commit/523fe868626edd65fdba7d478b0414ced6dfbcbe by srounet does not work under Python 3 under Jessie on RPi for a DHT11 sensor. After the very nominal syntax change for the iter definition, the module runs, throws an exception after the first iteration and continues to report constant values for temperature and relative humidity despite changes in the ambient conditions. The numbers themselves are spurious because they vary for each separate invocation of the module. Since I have the sensor working with the wiringPi library without any modifications to the breadboard, I will use joan2937's suggestion to use the new library at http://abyz.co.uk/rpi/pigpio/examples.html#pdif2_DHTXXD for further experiments (including DHT22) but I thought I would pass along my note in the event others seeking a Python 3 solution (for whatever reason) spin futile cycles to make this particular code work. I did not try the DHT22 sensor with the original code.

Of course, I will be willing to test any forthcoming revisions to the code. I understand the caveats in using these types of sensors under RPi's OS since I have an Arduino UNO sending back correct data to its RPi host via the Python pyserial module on the RPi. The only reason I became partial to Python was owing to the ease in using SQLite and MySQL interfaces through Python for further extensions.

Thanks for your guidance. Rdgs.

joan2937, 2015-11-20:
I have removed the reference to DHT22 as there is no evidence that code does not work as expected for Python 2 and Python 3.

You should be aware that the Python module and the (C) pigpio daemon interface module (C libraries libpigpiod_if and libpigpiod_if2) are equally capable of producing the correct results.  They both get the raw timing data from the pigpio daemon.

baqwas, 2015-11-20:
I can confirm that libpigpiod_if2 yesterday worked fine without any modifications to the breadboard connections. Fully understand the need to avoid confusion references to sensors that did not report any errors.

closed 2015-11-20 by baqwas