joan2937/pigpio #36: Ease cross-compilation

Xfel, 2015-11-30:
Currently, it's quite difficult to build pigpio using a cross compiler. I actually have to change every single command. Additionally, the install target does not allow to set a prefix path for a custom install location.

Could this please be fixed?

joan2937, 2015-11-30:
How?  I have no idea which cross compiler you or anyone else will choose to use.

Personally I use a make file called MakeCross.

I use "make -f MakeCross" for soft float and "HF=1 make -f MakeCross -j4" for hard float.
```make
# MakeCross
ifdef HF
	CC      = arm-linux-gnueabihf-gcc
	AR      = arm-linux-gnueabihf-ar
	RANLIB  = arm-linux-gnueabihf-ranlib
	SIZE    = arm-linux-gnueabihf-size
	SHLIB   = arm-linux-gnueabihf-gcc -shared
	STRIPLIB = arm-linux-gnueabihf-strip --strip-unneeded
else
	CC      = arm-linux-gnueabi-gcc
	AR      = arm-linux-gnueabi-ar
	RANLIB  = arm-linux-gnueabi-ranlib
	SIZE    = arm-linux-gnueabi-size
	SHLIB   = arm-linux-gnueabi-gcc -shared
	STRIPLIB = arm-linux-gnueabi-strip --strip-unneeded
endif

CFLAGS	+= -O3 -Wall -pthread

LIB1     = libpigpio.so
OBJ1     = pigpio.o command.o

LIB2     = libpigpiod_if.so
OBJ2     = pigpiod_if.o command.o

LIB3     = libpigpiod_if2.so
OBJ3     = pigpiod_if2.o command.o

LIB      = $(LIB1) $(LIB2) $(LIB3)

ALL     = $(LIB) x_pigpio x_pigpiod_if x_pigpiod_if2 pig2vcd pigpiod pigs

LL1      = -L. -lpigpio -pthread -lrt

LL2      = -L. -lpigpiod_if -pthread -lrt

LL3      = -L. -lpigpiod_if2 -pthread -lrt

all:	$(ALL)

pigpio.o: pigpio.c pigpio.h command.h custom.cext
	$(CC) $(CFLAGS) -fpic -c -o pigpio.o pigpio.c

pigpiod_if.o: pigpiod_if.c pigpio.h command.h pigpiod_if.h
	$(CC) $(CFLAGS) -fpic -c -o pigpiod_if.o pigpiod_if.c

pigpiod_if2.o: pigpiod_if2.c pigpio.h command.h pigpiod_if2.h
	$(CC) $(CFLAGS) -fpic -c -o pigpiod_if2.o pigpiod_if2.c

command.o: command.c pigpio.h command.h
	$(CC) $(CFLAGS) -fpic -c -o command.o command.c

x_pigpio:	x_pigpio.o $(LIB1)
	$(CC) -o x_pigpio x_pigpio.c $(LL1)

x_pigpiod_if:	x_pigpiod_if.o $(LIB2)
	$(CC) -o x_pigpiod_if x_pigpiod_if.c $(LL2)

x_pigpiod_if2:	x_pigpiod_if2.o $(LIB3)
	$(CC) -o x_pigpiod_if2 x_pigpiod_if2.c $(LL3)

pigpiod:	pigpiod.o $(LIB1)
	$(CC) -o pigpiod pigpiod.c $(LL1)

pigs:	pigs.o command.o
	$(CC) -o pigs pigs.c command.c

pig2vcd:	pig2vcd.o
	$(CC) -o pig2vcd pig2vcd.c

clean:
	rm -f *.o *.i *.s *~ $(ALL)

$(LIB1):	$(OBJ1)
	$(SHLIB) -o $(LIB1) $(OBJ1)
	$(STRIPLIB) $(LIB1)
	$(SIZE)     $(LIB1)

$(LIB2):	$(OBJ2)
	$(SHLIB) -o $(LIB2) $(OBJ2)
	$(STRIPLIB) $(LIB2)
	$(SIZE)     $(LIB2)

$(LIB3):	$(OBJ3)
	$(SHLIB) -o $(LIB3) $(OBJ3)
	$(STRIPLIB) $(LIB3)
	$(SIZE)     $(LIB3)

# generated using gcc -MM *.c

command.o: command.c pigpio.h command.h
pig2vcd.o: pig2vcd.c pigpio.h
pigpio.o: pigpio.c pigpio.h command.h custom.cext
pigpiod.o: pigpiod.c pigpio.h
pigpiod_if.o: pigpiod_if.c pigpio.h command.h pigpiod_if.h
pigpiod_if2.o: pigpiod_if2.c pigpio.h command.h pigpiod_if2.h
pigs.o: pigs.c pigpio.h command.h
x_pigpio.o: x_pigpio.c pigpio.h
x_pigpiod_if.o: x_pigpiod_if.c pigpiod_if.h pigpio.h
x_pigpiod_if2.o: x_pigpiod_if2.c pigpiod_if2.h pigpio.h
```

Xfel, 2015-12-02:
Some way to change the installation base path would be really helpful, as separating the outputs manually is error-prone. Also, many make files allow setting a "compiler command prefix" for use with cross-compilation.

joan2937, 2015-12-02:
I have no objection, it's just that my knowledge of makefiles and cross-compilation is severely limited.

Could you prepare a Makefile along the lines of what you want?  If all I need to do is replace the current Makefile I would be quite happy to do so, as long as I can understand what it is doing.

joan2937, 2015-12-27:
I will close this as an issue until someone has some concrete proposals.

closed 2015-12-27 by joan2937