joan2937/pigpio #39: Allow only 0 to 255 in set_pwm

Loris1123, 2015-12-12:
Setting a value of more than 255 or less than 0 to `set_PWM_dutycycle` makes no sense of course.

But when you do something like

    i = 255
    while i > 0:
        set_PWM_dutycycle(PIN, i)
        i -= 2

you will get the error pigpio.error: 'dutycycle not 0-range (default 255)'.
Maybe it would be a good idea to say: everything smaller than 0 means off, and everything larger than 255 means 100% on.
I modified `set_PWM_dutycycle` with

    if dutycycle < 0:
        dutycycle = 0
    if dutycycle > 255:
        dutycycle = 255
to do this for me.

joan2937, 2015-12-12:
This is more to do with philosophy than anything else.

The user should not be calling a function with incorrect parameters.  If a function is called incorrectly it indicates there is a logic error in the program.  I do not believe that such logic errors should be ignored, they should be corrected.

Loris1123, 2015-12-12:
Sure, I totally agree with this.  
For functions like I posted, it is a bit annoying to check `if i < 0: i=0`.
A way to solve this would be a kwarg like this:

    def set_PWM_dutycycle(self, user_gpio, dutycycle, error_save=False):
    if error_save:
        if dutycycle < 0:
            dutycycle = 0
        if dutycycle > 255:
            dutycycle = 255
    return _u2i(_pigpio_command(
         self.sl, _PI_CMD_PWM, user_gpio, int(dutycycle)))

joan2937, 2015-12-12:
An even better way would be to correct the logic error.

I shall not be changing the way error handling works in this or similar functions.

By the way the dutycycle range defaults to 0..255, but the user can change the range.

closed 2015-12-27 by joan2937