joan2937/pigpio #41: Writing a Python module with pigpio

paviro, 2015-12-25:
Hi!
I have a function which can fade some LEDs, it works without a problem. When I put that code into another file and import it into my main file and call my function from there I get the following error:

`Dez 25 21:52:51 Spiegel python[711]: Traceback (most recent call last):
Dez 25 21:52:51 Spiegel python[711]: File "/usr/local/bin/mirrorcontroller/mirrorcontroller.py", line 151, in socket_server
Dez 25 21:52:51 Spiegel python[711]: pulse(1, 5)
Dez 25 21:52:51 Spiegel python[711]: File "/usr/local/bin/mirrorcontroller/mirrorcontroller.py", line 114, in pulse
Dez 25 21:52:51 Spiegel python[711]: brightness_up(color_selected)
Dez 25 21:52:51 Spiegel python[711]: File "/usr/local/bin/mirrorcontroller/mirrorcontroller.py", line 104, in brightness_up
Dez 25 21:52:51 Spiegel python[711]: set_color(color_selected,brightness)
Dez 25 21:52:51 Spiegel python[711]: File "/usr/local/bin/mirrorcontroller/mirrorcontroller.py", line 93, in set_color
Dez 25 21:52:51 Spiegel python[711]: pi.set_PWM_dutycycle(RED_PIN, color[color_selected][1] * brightness)
Dez 25 21:52:51 Spiegel python[711]: File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1133, in set_PWM_dutycycle
Dez 25 21:52:51 Spiegel python[711]: self.sl, _PI_CMD_PWM, user_gpio, int(dutycycle)))
Dez 25 21:52:51 Spiegel python[711]: File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 834, in _pigpio_command
Dez 25 21:52:51 Spiegel python[711]: sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
Dez 25 21:52:51 Spiegel python[711]: AttributeError: 'NoneType' object has no attribute 'send'`

joan2937, 2015-12-26:
There is not enough context to be sure what is happening.

Where do you connect to the daemon, e.g. pi = pigpio.pi(), and how do you pass "pi" to the called module?

paviro, 2015-12-26:
I am using pi-blaster now which works. I called it inside the module.

joan2937, 2015-12-27:
I'm closing this as it seems to be more of how do I write a Python program than a issue with pigpio.

closed 2015-12-27 by joan2937