joan2937/pigpio #42: PWM set issue

edharman, 2015-12-26:
Using a new install compile and load pigpiod -t 1 
from python load library and associate local pi and issue 
pi.set_PWM_range(6,500)
I assumed this would give me ~2ms pulse width increments, however measuring on a scope I get 
pi.set_PWM_dutycycle(6,0) 
actual pulse width 0
and 0 right up to set dutycycle 3 where I get a pulse width of 4.6us
this in itself is not an issue however -
Set Actual
4 4.7
5 9.7
6 9.7
7 9.7 
8 14.69
9 14.69
10 19.68
11 19.68
12 19.68
13 24.68

Ignoring absolute values as errors in my scope measurements I would have expected to the pulse width to increment consistently.
Any ideas as to what may be going wrong ?

joan2937, 2015-12-26:
sudo pigpiod -t1 starts the daemon with the PCM clock (the default).

At the default 5 µs sampling rate the default PWM frequency is 800 Hz.  So each pulse happens in a 1000000/800 or 1250 µs time slot.  There are 250 steps between off and fully-on (each step equates to 1 sampling period, or 5 µs at a 5 µs sampling rate, and 1250/5 gives 250).  The figure of 250 is what I refer to as the real range.  For convenience you can map the 250 steps into a range of your choice, but there are still only 250 discrete steps.

The C library (well the code) is the definitive statement of what happens, for instance, see  http://abyz.co.uk/rpi/pigpio/cif.html#gpioSetPWMrange

edharman, 2015-12-26:
Hi thanks for the swift response,  user error....
I would like to be able to use more than 250 steps - am dimming some commercial LED's whose current/versus lumen output is non-linear especially toward low duty cycles. Will have a play with the range option which may work out.
In any case thanks for a brilliant and very useful library,
/Ed

joan2937, 2015-12-26:
You will have to lower the PWM frequency or increase the sample rate.

I would not suggest sampling higher than every 2 µs.  Use set_PWM_frequency to lower the frequency.

Remember there are two hardware PWM channels.  pigpio programs them to have a million steps between off and fully on.  See http://abyz.co.uk/rpi/pigpio/python.html#hardware_PWM

edharman, 2015-12-27:
Hi Thanks for the pointers - have had a quick play and it works much closer to what I want - I can't use the hardware capable PWM channels as have maybe 10-13 output channels I want to control. 
Thanks for the pointers

edharman, 2015-12-27:
Hi Joan,
Have had a play with the hardware gpio and they fit the bill - I can control the illumination in almost imperceptible steps so looks like I'll be buying a PI Compute module or two...
Many thanks for your support, and also for such a useful library
/Ed

closed 2015-12-27 by edharman

joan2937, 2015-12-27:
Remember the compute modules have the same number of hardware PWM channels (2) as the other Pi's with the 40 pin expansion header.

If I remember correctly the choice of a million steps was arbitrary, it could just have easily been 10 million.  I do not know what resolution the underlying hardware/GPIO can manage.

edharman, 2015-12-27:
hmm - looking at http://abyz.co.uk/rpi/pigpio/python.html#hardware_PWM indicates to me that with a Pi 2 I only can access 4 GPIO's with hardware PWM but can get another 5 with the compute module.
Am I missing something?

joan2937, 2015-12-27:
That's correct, however, there are only two channels.  So all GPIO connected to channel 0 will have one setting and all GPIO connected to channel 1 will have another.

```
The same PWM channel is available on multiple gpios. The latest frequency and dutycycle setting will be used by all gpios which share a PWM channel. 
```

edharman, 2015-12-27:
Argh - just tested and.... you are correct!. Hmm, I wonder why this is designed this way - I would have expected and desired that I could independently PWM each channel.
..back to the drawing board..

edharman, 2015-12-27:
correction - expected to independently PWM each GPIO.....

joan2937, 2015-12-27:
You seem to have pretty tough requirements.  I suggest you post what you need from PWM in detail, e.g. number of unique channels, frequencies, and resolution on https://www.raspberrypi.org/forums/.  I reckon you will need additional hardware support. There should be people who can suggest suitable hardware.

edharman, 2015-12-27:
Hi Thanks - will post my requirements - deploying  a Pi for every two lights not only is overkill but expensive!!
I may take a look at servoblaster - however I am not at heart a programmer, and your very nice library allowed me to knock something together pretty quickly - so kudos to you!
Thankyou once again for your help and assistance

SlySven, 2016-02-12:
How about a Pi Zero for every two lights - is that more doable? :dollar: :pound: :smile:

edharman, 2016-02-12:
Hi SlySven, I had considered that, however it seemed to me a somewhat brute force approach to solving my problem! I have resolved the expansion issue by using a pair of PCA9685 I2C bus 16 channel PWM adaptors. These are stateless -so single command to each channel to set PWM duty cycle so negligible cpu load, however I then realised my next challenge was how to handle the number of required input channels-  mainly from rotary encoders which including an on-off switch would require 3 GPIO's each ...and I need ~20 channels.
So currently testing using MCP23017 -16 channel GPIO's and using an interrupt routine triggered from a hardwire to a Pi GPIO since polling 64 channels was never going to work!
Have got the bare bones working with a single adaptor, and currently refining the code to efficiently expand it to cope with the 4 adaptors I need to use.
All good fun!!

SlySven, 2016-02-12:
I was looking at the Compute module a short while ago - now that **does** have a lot more GPIO pins available (~120 I believe)! :astonished: