joan2937/pigpio #44: pigpio.py: close and atexit not the best method

DavidAustin, 2015-12-31:
The current registration of close as an atexit callback isn't the best way to clean up - resources
will only be freed at program exit (which is almost pointless).  

I'm seeing a problem with a long-running python script
with many pigpio connections - it's running out of resources to create threads.

For now, my fix is to explicitly call close.

However, the better solution would be to use a __del__ destructor to call close once
the reference to the pigpio class is out of scope.

joan2937, 2015-12-31:
atexit is only used to register the connection stop method.  That's done to ensure any open connections are  closed down at program termination.

I would expect that the stop method would be called explicitly prior to program termination whenever a connection is no longer needed.

DavidAustin, 2015-12-31:
> atexit is only used to register the connection stop method. That's done to ensure any open connections are closed down at program termination.

As I wrote, that's pretty much pointless - all of a program's resources are freed by the OS
at exit.

> I would expect that the stop method would be called explicitly prior to program termination whenever a connection is no longer needed.

The documentation doesn't clarify this.  The "Pythonic" way is to assume that an object
will be cleaned up by the garbage collector once there's no longer a reference to it.
Therefore, in this case, use of the `__del__` destructor to call the stop() method.

joan2937, 2016-02-29:
I shall not be making any changes to the pigpio.py module for this issue.  Mainly because my Python knowledge is too limited.  If anyone has a "Pythonic" solution I'll take it on board provided I understand the solution and it is not out of kilter with the rest of the code.

closed 2016-02-29 by joan2937