joan2937/pigpio #46: timestamping with CLOCK_REALTIME

nachoplus, 2016-01-11:
Hello everyone,

First of all I would like to say thank you for this great piece of soft.

In my aplication I need to get precise UTC timestamp of gpio activation.  I learnt that pigpio is capable of timestamping gpio changes with precision of microseconds but using system ticks but the traslation to UTC is not so obvious if you consider latencies, clock jitters... 

I know that normal system "wall clock" deviate to much to consider for a precise timestamping. To avoid that I run a NTP Stratum 1 using a GPS and PPS that get system clock between +/-10 microseconds of UTC.

Could be posible to use CLOCK_REALTIME in order to obtain a UTC timestamp with pigpio?

Thank in advance

joan2937, 2016-01-12:
Probably.

I use the following type f code

```C
t1=0;
t2=10;
while ((t2-t1) > 2)
{
   t1 = gpioTick();
   c = time_time();
   t2 = gpioTick();
}
```
That will show the correspondence between Linux time and the system tick to within a couple of microseconds.

nachoplus, 2016-01-12:
I'll try. Thank you Joan2937!

Also I am thinking about to take advantange of the PPS signal reading it with pigpio lib in order to get the ticks from the start of each seconds.

Is pigpio compatible with gpio_pps kernel module? I mean it is safe to read PPS signal with pigpio without disturbing it?

Nacho

joan2937, 2016-01-12:
Yes, you can use pigpio to monitor any GPIO, regardless of whether they have been set up as inputs or outputs.  Note that the PPS signal will have to be 5 µs or greater in width to be seen at the default pigpio sampling rate of 5 µs.

closed 2016-02-29 by joan2937

nachoplus, 2016-04-25:
Hi,
Finally everything work as as expected and I concluded my project. A timestamping raspi used to get the picture time with precision below 5micros form UTC! I use for astrophoto of space junk. If you feel curiosity see https://github.com/nachoplus/cronoStamper The docs are on the way..

Thanks!