joan2937/pigpio #47: Alert thread lagging behind

Xfel, 2016-01-13:
We are currently trying to use the code from the rotary encoder example (http://abyz.co.uk/rpi/pigpio/ex_rotary_encoder.html) to read a position for PID control. The control loop is running in the main thread.

The issue we came across now is that sometimes, the alert callbacks do not get called for a very long time before being processed in a burst. This gives us uncontrollable errors in control which could lead to physical damage to our setup.

We are using a custom version of raspbian with the preemt-rt kernel patch and use the `gpioCfgSetInternals(PI_CFG_RT_PRIORITY);` to enable realtime processing. One possible problem with this seems to be the fact that there is no priority configuration for the single threads, so setting the alert thread on a higher priority is a solution we considered, but unfortunately there is no config option for this. Also, as the Pi2 has a quad core processor, it should be possible for the main thread and the alert thread to run in parallel.

joan2937, 2016-01-13:
The main thread, alert thread, pipe thread, and socket thread should be running in separate cores on the Pi2.  Nothing special needs to be done.

I'll have a look at the alert thread as I periodically try to tune the way it works.  Perhaps I have broken something.

Which pigpio version are you using (pigpio.h PIGPIO_VERSION)?

Xfel, 2016-01-13:
We are using version 42. 
There are also some other threads/processes running on the system, could they have an impact? Also, do the socket and pipe threads need to be in real-time? I'd normally give IO threads a much lower priority...

joan2937, 2016-01-13:
Nothing obvious.

Before I spend any more time on this could you confirm the data rates you are talking about, i.e. how many alert function calls are made per second.  Also confirm that the processing required can keep up with that data rate.

Xfel, 2016-01-14:
The data rate depends on the motor speed, so providing exact values is difficult. In the normal operation range we got about 6500 alert calls. I also did a quick test using a free motor and spinning it up to maximum speed, there we got 22000 alert calls per second. These test runs worked flawlessly, so I would conclude that the processing can indeed keep up.

joan2937, 2016-01-14:
Okay.  There shouldn't be any problem at those rates.  I'll have a closer look.  It might take several days depending on what else I need to do.  If you have a cut down program which exhibits the problem it would be useful.  I can simulate the inputs.

Xfel, 2016-01-14:
Thank you for your help. A testcase program would be a bit difficult as the error doesn't occur deterministically, but I'll see what I can do.

Xfel, 2016-01-14:
Here is a shrunk-down program that should be able to reproduce the error:
https://www.dropbox.com/s/girq48d7mwqpan4/motorcontrol_test.zip?dl=0

joan2937, 2016-01-23:
Just a note to say I haven't forgotten this issue.  Just busier with other things at the moment.

Xfel, 2016-01-27:
We now switched to using software polling for the gpios we're interested in, and so far we didn't get any problems. However, the software polling has a big impact on the overall cpu usage, so if there was a solution for this, I'd greatly appreciate.

joan2937, 2016-02-06:
I have tried to separate out the GPIO sampling from the issuing of reports in V44.  Could you see if you still have problems (or if I've introduced new ones).

joan2937, 2016-05-18:
In the absence of further information I'm going to close this as an issue.

closed 2016-05-18 by joan2937

msandres13, 2018-05-17:
[image]
I see the same bursting behavior in my application. I read pulses from a powermeter (S0 Interface) 90ms duration one pulse.
I run a standard raspberian debian version 9.4.
I am still trying to isolate the problem. I keep you posted

msandres13, 2018-05-19:
It was my silly mistake, sorry for bothering. In one area of the code I used a tick count with a simple isGreater compare which caused the clause to missfunction when the tick counter wrapped. In all other cases I used your proposed code to compare tick counts.
Thanks for this great support and pigpio.