joan2937/pigpio #48: A question about callbacks and ticks

fivdi, 2016-02-14:
The `gpioSetAlertFunc` provided by the pigpio C interface can be used to register a callback function to detect gpio state changes. Has the percision and accuracy of the tick value passed to the callback function improved in a recent version of pigpio? The tick values I'm seeing with V45 look better than those that I was seeing with V42.

joan2937, 2016-02-14:
Better in what sense?  The tick values should relate to the time of the level change in exactly the same way.

I have been doing some work trying to find an "optimal" graceful degradation strategy when pigpio becomes overloaded.  By overloaded I mean when a "user" can't process the requested samples in a timely fashion, mean that pigpio's internal buffers overflow and samples are lost.

That work may have reduced sample latency in busy situations.  The ticks themselves shouldn't have changed.

fivdi, 2016-02-14:
I'm currently adding alert callback functionality to a JavaScript module. For test purposes, hardware PWM is being used on GPIO18. The PWM frequency is 250Hz, the duty cycle is 7 microseconds. Alert callbacks are registered for GPIO18. The sampling rate is 1 microsecond. The alert callback function measures the duty cycle width, which should be around 7 microseconds. Yesterday I was using pigpio V42 and most of the measurements were very close to 7 microseconds. However, there were occasional measurements that I didn't understand, for example, ~14000 microseconds or ~19000 microseconds. Today, after upgrading to V45 this is no longer the case and all measurements are close to 7 microseconds. It's not essential for me to know why the measurements are better today than they were yesterday, I was just wondering if something in pigpio had changed and would explain the improvement.

joan2937, 2016-02-14:
That sounds like a bug, but I have no idea what might have been causing it or how it might have been fixed.  I'll try V42 a little later and see if I see the same effect.  That may give a clue as to what is/was happening.

fivdi, 2016-05-15:
@joan2937 I'm going to close this one. Feel free to reopen if necessary.

closed 2016-05-15 by fivdi