joan2937/pigpio #49: pigpio error

harshangshah, 2016-02-24:
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio on localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

I am getting the following error when I am trying to use pigpio for i2c.
I have started the pigpio daemon by sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "ina219_02_23_2016_pigpio.py", line 23, in <module>
    handle_ina_219 = pi.i2c_open(1, address_ina_219)
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 2221, in i2c_open
    self.sl, _PI_CMD_I2CO, i2c_bus, i2c_address, 4, extents))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 857, in _pigpio_command_ext
    sl.s.sendall(ext)
AttributeError: 'NoneType' object has no attribute 'sendall'

joan2937, 2016-02-24:
What does your pi = line contain?

harshangshah, 2016-02-24:
pi=pigpio.pi()

joan2937, 2016-02-24:
After the pi=pigpio.pi() line add the following

print(pi.connected)

If it's False you have not connected to the pigpio daemon.  If you are running the script on the local Pi the most likely reason is you don't have a loopback device defined in /etc/network/interfaces.

harshangshah, 2016-02-24:
I read in other closed discussion about the loopback, could you please explain me what it should be and what it should not be? I could not understand it well.

harshangshah, 2016-02-24:
This is how my /etc/network/interfaces look like:

auto wlan0

iface lo inet loopback
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet static
address 192.168.0.10
netmask 255.255.255.0
gateway 192.168.0.1
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp

joan2937, 2016-02-24:
Try adding auto lo.

# The loopback network interface
auto lo
iface lo inet loopback

harshangshah, 2016-02-24:
wow, it works now :-) Thanks joan

joan2937, 2016-02-24:
Okay.

closed 2016-02-24 by joan2937

harshangshah, 2016-06-14:
Hi Joan,

I am getting an error: 
Nonetype object has no attribute send

Could you please help me debug it?

Thanks in advance

joan2937, 2016-06-14:
You need to give all relevant information.

erg-samowzbudnik, 2019-08-09:
Hi. I'm having similar issue on arm64 not sure it's relevant.
installed pigpio, it fails to start:
/etc/init.d/pigpiod start
/usr/bin/pigpiod: error while loading shared libraries: ligpigpio.so: cannot open shared object file: No such file or directory

in python3 interpreter "import pigpio" works but then "pi = pigpio.pi() fails with Can't connect to localhost(8888)

I have added auto lo to /etc/network/interfaces

any tips would be great,

cheers,

erg

joan2937, 2019-08-09:
I can't help with 64-bit issues as it's all a mystery to me.  However I have just merged #284 which might possibly make a difference.