joan2937/pigpio #50: Is it possible to support NEO-PIXELS now?

SlySven, 2016-02-25:
Having spotted the [rpi-ws821x](https://github.com/jgarff/rpi_ws281x) library mentioned on the Adafruit site I was wondering whether it's use of the Hardware PWM could be incorporated (Licence renegotiation permitting) into **pigpio**?

I have a couple of those sort of strips and wanted to incorporate them into my *wicked-plans* :smirk: but I was aware that in the past the timing was too tight to do on any RPi - I was contemplating designing a PIC or other micro-controller *buffer* to accept a 2-wire serial output that RPis (and pigpio) should *easily(?)* be able to do and then squirt out the tightly timed 1-wire serial equivalent that the WS8211B IC (and the WS8212B LED module incorporating it) that *cheaper* LED strips have - then I realised that I might have pixel data storage issues unless I got a bit inventive.  So if this is something that *could* be done in the light of developments in PWM coding it might be quite useful.

Failing that - if I have to reserve the PWM stuff for this other library - just how crippled will pigpio be without that facility?

joan2937, 2016-02-25:
By default pigpio uses the PCM peripheral for timing.  It only uses the PWM peripheral if waveforms are used or if hardware PWM is used.  I may have forgotten something but basically if you don't use a function with wave in the name or a function with hardware PWM in the name you should be fine.

On the general point - pigpio works in the microsecond arena, it has no solution for the sub-microsecond timing needed by WS821 LEDs.

It may be possible to drive the LEDs now with the latest Linux SPI driver.  It supports DMA and should be able to "bodge" the timing. I did some experiments with the pigpio userland SPI driver and that had limited success but I couldn't control filling the SPI buffers in a timely fashion from userland.

I say "bodge" the timing as any SPI solution must incorporate the 1.5 bit time gap between octets inserted by the Pi SPI hardware.  There should be a post from me on the raspberrypi.org forum outlining the attempt at driving those LEDs.

As a final note I won't incorporate non Public Domain source into the code base I maintain.  Of course forks can do what they want.

SlySven, 2016-02-25:
Given a chance (rather slim right now!) I'll take a closer look and see how the other library (it IS on GitHub) does the, very tight as I understand, timing - there was the observation that there were issues on the RPi 2B but that is not clear if it is still the case.

Obviously any combination I or others make of your *Unlicence* code with the other's *BSD 2-Line license* type code would have to be licensed under the latter's...

SlySven, 2016-03-12:
With the licence disparity the functionality could not be directly brought in so I'll close this issue here but may look to do something in my own fork, eventually...

closed 2016-03-12 by SlySven