joan2937/pigpio #51: make -f MakeRemote needs 3 invokations to complete?

SlySven, 2016-02-25:
Reading the `README` pointed out the ability to compile the remote capabilities (and to generate the `man` pages *locally* - which is useful when `ssh`ing into a headless RPi for creation/compiling code).

However the compilation took three attempts before completing - is this to be expected for a fresh install?

Preliminary step:
> stephen@Ripley:~/src/pigpio$ make -f MakeRemote clean
rm -f *.o *.i *.s *~ libpigpiod_if.so libpigpiod_if2.so pigs x_pigpiod_if x_pigpiod_if2

1<sup>st</sup> try:
> stephen@Ripley:~/src/pigpio$ make -f MakeRemote
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
/usr/bin/ld: cannot find -lpigpiod_if
collect2: error: ld returned 1 exit status
make: *** [x_pigpiod_if] Error 1
make: *** Waiting for unfinished jobs....

2<sup>nd</sup> try:
> stephen@Ripley:~/src/pigpio$ make -f MakeRemote
gcc -shared -o libpigpiod_if.so pigpiod_if.o command.o
gcc -shared -o libpigpiod_if2.so pigpiod_if2.o command.o
gcc -O3 -Wall -pthread -fpic -o pigs pigs.c command.c
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
/usr/bin/ld: cannot find -lpigpiod_if
./libpigpiod_if2.so: file not recognized: File truncated
collect2: error: ld returned 1 exit status
make: *** [x_pigpiod_if] Error 1
make: *** Waiting for unfinished jobs....
collect2: error: ld returned 1 exit status
make: *** [x_pigpiod_if2] Error 1
strip --strip-unneeded libpigpiod_if2.so
strip --strip-unneeded libpigpiod_if.so
size     libpigpiod_if2.so
size     libpigpiod_if.so
   text	   data	    bss	    dec	    hex	filename
  69539	   6624	  49408	 125571	  1ea83	libpigpiod_if.so
   text	   data	    bss	    dec	    hex	filename
  56875	   6592	   2592	  66059	  1020b	libpigpiod_if2.so

3<sup>rd</sup> and successfully completed try:
> stephen@Ripley:~/src/pigpio$ make -f MakeRemote
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
stephen@Ripley:~/src/pigpio$ make -f MakeRemote
make: Nothing to be done for `all'.

It isn't a major issue but it might be worth pointing out in the top-level README... 

Also for the installation step:
>stephen@Ripley:~/src/pigpio$ sudo make -f MakeRemote install
[sudo] password for stephen: 
install -m 0755 -d                /usr/local/include
install -m 0644 pigpio.h          /usr/local/include
install -m 0644 pigpiod_if.h      /usr/local/include
install -m 0644 pigpiod_if2.h     /usr/local/include
install -m 0755 -d                /usr/local/lib
install -m 0644 libpigpiod_if.so  /usr/local/lib
install -m 0644 libpigpiod_if2.so /usr/local/lib
install -m 0755 -d                /usr/local/bin
install -m 0755 pigs              /usr/local/bin
python2 setup.py install
running install
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
copying pigpio.py -> build/lib.linux-x86_64-2.7
running install_lib
copying build/lib.linux-x86_64-2.7/pigpio.py -> /usr/local/lib/python2.7/dist-packages
byte-compiling /usr/local/lib/python2.7/dist-packages/pigpio.py to pigpio.pyc
running install_egg_info
Writing /usr/local/lib/python2.7/dist-packages/pigpio-1.27.egg-info
python3 setup.py install
make: python3: Command not found
make: *** [install] Error 127

I only have `python` 2.x installed, so the installation bailing out when it can't find `python3` is not surprising - I dunno whether it is worth checking to see which version(s) are around before trying to install for them, obviously your work is focused on the RPi end of things and the effort to have support for the extra remote features on lots of different host types is going to hit the *diminishing returns* curve fairly quickly!

joan2937, 2016-02-25:
What machine were you building on?  I only use that command on a Linux laptop and it hasn't shown that behaviour.

SlySven, 2016-02-25:
A Debian "Wheezy" (+ backports) PC.

Ah ha, I have it :exclamation:

This is a AMD **Quad-core** machine and I recall now that I have a `MAKEFLAGS = -j5` set in my environment {to speed up compilation when I'm working on [Mudlet](https://github.com/Mudlet/Mudlet) which is a large CPP project} - so the core that get to run the *link* stage is not waiting for the other cores to finish the constituent files before combining them into the library (`ar`chived) files!

You can close this I guess. :blush:

closed 2016-02-29 by joan2937