joan2937/pigpio #52: State changes not being detected with notifications

fivdi, 2016-03-05:
While testing the pigpio Node.js module I noticed that state changes don't always appear to be detected by the notification mechanisms provided by the pigpio C library. Similar issues can be seen when the pigpio C library is used directly, i.e., without the Node.js module.

Here is how the test was performed with the pigpio C library.

The following program was run on an ATmega328P clocked at 8 Mhz. The program generates generates 50000 pulses or 100000 state changes on PB1 of the ATmega. The pulses are approximately 8 microseconds wide with a 50% duty cycle. After generating the pulses, the program waits 5 seconds before repeating the same thing again.

```C
/*
avr-gcc -Wall -c -g -Os -mmcu=atmega328p -DF_CPU=8000000UL pulse.c
avr-gcc -g -mmcu=atmega328p -o pulse.elf pulse.o
avr-objcopy -j .text -j .data -O ihex pulse.elf pulse.hex
sudo avrdude -c dragon_isp -P usb -p m328p -e -U flash:w:pulse.hex
*/

#include <avr/io.h>
#include <avr/power.h>
#include <util/delay.h>

#define LED PB0
#define LED_DDR DDRB
#define LED_PORT PORTB

#define PULSE PB1
#define PULSE_DDR DDRB
#define PULSE_PORT PORTB

#define DELAYTIME 4

#define setBit(sfr, bit) (_SFR_BYTE(sfr) |= (1 << bit))
#define clearBit(sfr, bit) (_SFR_BYTE(sfr) &= ~(1 << bit))
#define toggleBit(sfr, bit) (_SFR_BYTE(sfr) ^= (1 << bit))

int main(void) {
  uint32_t i;

  clock_prescale_set(clock_div_1); // 8Mhz clock

  setBit(LED_DDR, LED); // configure LED pin as an output
  setBit(PULSE_DDR, PULSE); // configure PULSE pin as an output

  while (1) {
    setBit(LED_PORT, LED); // LED on

    for (i = 0; i != 50000; ++i) {
      setBit(PULSE_PORT, PULSE); // PULSE high
      _delay_us(DELAYTIME);

      clearBit(PULSE_PORT, PULSE); // PULSE low
      _delay_us(DELAYTIME);
    }

    clearBit(LED_PORT, LED); // LED off

    _delay_ms(5000);
  }

  return (0);
}
```

PB1 on the ATmega328P is connected to GPIO18 on a Raspberry Pi 3. The following program is run on the Pi. It simply counts the number of state changes on GPIO18 and prints the number of state changes detected once per second.

```C
// gcc -o pulsein pulsein.c -lpigpio -lpthread -lrt

#include <fcntl.h>
#include <stdio.h>
#include <pigpio.h>

#define PULSE_GPIO 18
#define BUF_SIZE 1000

static uint32_t notificationCount = 0;

void timerFunc(void) {
  printf("%u\n", notificationCount);
}

int main() {
  int version;
  int handle;
  char pipePath[32];
  int fd;
  int bytes;
  gpioReport_t r[BUF_SIZE];
  uint16_t lastSeqno = 0xffff;
  int entries;
  int i;

  gpioCfgClock(1, 1, 1);

  if ((version = gpioInitialise()) < 0) {
    printf("gpioInitialise error\n");
    return -1;
  }

  printf("version: %d\n", version);

  if (gpioSetTimerFunc(0, 2000, timerFunc) < 0) {
    printf("gpioSetTimerFunc error\n");
  }

  if ((handle = gpioNotifyOpen()) < 0) {
    printf("gpioNotifyOpen error\n");
    return -1;
  }

  if (gpioNotifyBegin(handle, (1 << PULSE_GPIO)) < 0) {
    printf("gpioNotifyBegin error\n");
    return -1;
  }

  sprintf(pipePath, "/dev/pigpio%d", handle);

  if ((fd = open(pipePath, O_RDONLY)) < 0) {
    printf("open error\n");
    return -1;
  }

  for (;;) {
    bytes = read(fd, r, 12 * BUF_SIZE);

    if (bytes <= 0) {
      printf("read error\n");
      return -1;
    }

    entries = bytes / 12;

    if (entries * 12 != bytes) {
      printf("Oops partial entries are not handled");
      return -1;
    }

    for (i = 0; i != entries; ++i) {
      ++notificationCount;

      if (((uint16_t) (lastSeqno + 1)) != r[i].seqno) {
        printf("  sequence error - lastSeqno: %d, seqno: %d\n", lastSeqno, r[i].seqno);
        printf("    %d\n", notificationCount);
      }

      lastSeqno = r[i].seqno;
    }
  }

  gpioTerminate();

  return 0;
}
```

Here is an example of typical console output on the Pi. As can be seen, some state changes are not detected. In this particular test run sometimes 4 and sometimes 2 state changes were not detected for batches of 100000 state changes.

```
root@raspberrypi:/home/pi/pigpioc# ./pulsein 
version: 47
100000
100000
199996
199996
199996
299994
299994
299994
399990
399990
499986
499986
499986
599982
599982
599982
699978
699978
799974
799974
799974
899970
```

Note that it's not essential to generate batches of 50000 pulses to see the issue. It can also be seen with less pulses, but it can take longer to reproduce the issue.

The question is, is this to be expected or is there a problem that needs to be fixed?

joan2937, 2016-03-05:
Do you have disable_pvt=1 in your /boot/config.txt?

The Pi GPU stalls the DMA for a fraction of a second twice per second to temperature correct RAM refresh rates.  That might be a factor.

fivdi, 2016-03-06:
Thanks for the feedback :)

No, disable_pvt=1 wasn't in /boot/config.txt but adding it does improve things.

Below is some output from the test program with disable_pvt=1 in /boot/config.txt. Here are some notes describing the output::

 * When the test program is started, sometimes two state changes are detected. These are not authentic state changes. They are not generated by the ATmega328P. I don't know where they are coming from and am not sure if they are in anyway related to the original issue reported here.
 * The test program detected 1100002 state changes without issue (except for the first 2.) There was no concurrent activity in other process other than the activity in process running on a default installation of Raspbian Jessie 2016-02-26.
 * For the state changes detected after state change 1100002 and up to and including state change 1799974 there was addition and intentional concurrent activity in another process. This concurrent activity resulted in state changes not being detected.
 * For the state changes detected after state change 1799974 there was no additional and intentional concurrent activity and all state changes were detected.

```
root@raspberrypi:/home/pi/pigpioc# ./pulsein 
version: 47
2            <- The two unauthentic state changes
100002
100002
182688
200002
200002
300002
300002
300002
400002
400002
486283
500002
500002
600002
600002
600002
700002
700002
791612
800002
800002
900002
900002
900002
1000002
1000002
1092226
1100002 <- No concurrent activity in other processes up to and including here.
1100002 
1199998 <- Concurrent activity in other processes started while detecting state changes counted here
1199998
1199998
1299998
1299998
1393830
1399994
1399994
1499990
1499990
1499990
1599986
1599986
1698847
1699982
1699982 
1799974 <- last batch counted where there was concurrent activity in other processes
1799974
1799974
1899974
1899974
1999974
1999974
1999974
2099974
2099974
2099974
2199974
2199974
2299974
2299974
2299974
2399974
2399974
2399974
2499974
2499974
2599974
2599974
2599974
2699974
2699974
2699974
2799974
2799974
2899974
2899974
2899974
2999974
2999974
2999974
3099974
3099974
3199974
3199974
3199974
3299974
3299974
3299974
3399974
3399974
3499974
3499974
3499974
3599974
3599974
```

joan2937, 2016-03-06:
I get similar results when I test between a Pi2B generating the pulses and a Pi3B receiving.

I used your code with a slight modification to only show the final count and GPIO 13.

```C
/*
issue52.c

gcc -Wall -pthread -o 52 issue52.c -lpigpio
*/

#include <fcntl.h>
#include <stdio.h>
#include <unistd.h>

#include <pigpio.h>

#define PULSE_GPIO 13
#define BUF_SIZE 1000

static uint32_t notificationCount = 0;

void timerFunc(void) {
  static int s_same=0, s_oldcount = 0;
  if (s_oldcount == notificationCount)
  {
    if (++s_same > 2)
    {
       s_same = -1000;
       if (s_oldcount) printf("%u\n", notificationCount);
       notificationCount = 0;
    }
  }
  else
  {
    s_oldcount = notificationCount;
    s_same = 0;
  }
}

int main() {
  int version;
  int handle;
  char pipePath[32];
  int fd;
  int bytes;
  gpioReport_t r[BUF_SIZE];
  uint16_t lastSeqno = 0xffff;
  int entries;
  int i;

  gpioCfgDMAchannels(6, 14);
  gpioCfgClock(1, 1, 1);

  if ((version = gpioInitialise()) < 0) {
    printf("gpioInitialise error\n");
    return -1;
  }

  printf("version: %d\n", version);

  if (gpioSetTimerFunc(0, 2000, timerFunc) < 0) {
    printf("gpioSetTimerFunc error\n");
  }

  if ((handle = gpioNotifyOpen()) < 0) {
    printf("gpioNotifyOpen error\n");
    return -1;
  }

  if (gpioNotifyBegin(handle, (1 << PULSE_GPIO)) < 0) {
    printf("gpioNotifyBegin error\n");
    return -1;
  }

  sprintf(pipePath, "/dev/pigpio%d", handle);

  if ((fd = open(pipePath, O_RDONLY)) < 0) {
    printf("open error\n");
    return -1;
  }

  for (;;) {
    bytes = read(fd, r, 12 * BUF_SIZE);

    if (bytes <= 0) {
      printf("read error\n");
      return -1;
    }

    entries = bytes / 12;

    if (entries * 12 != bytes) {
      printf("Oops partial entries are not handled");
      return -1;
    }

    for (i = 0; i != entries; ++i) {
      ++notificationCount;

      if (((uint16_t) (lastSeqno + 1)) != r[i].seqno) {
        printf("  sequence error - lastSeqno: %d, seqno: %d\n", lastSeqno, r[i].seqno);
        printf("    %d\n", notificationCount);
      }

      lastSeqno = r[i].seqno;
    }
  }

  gpioTerminate();

  return 0;
}
```
I used wave chains to generate known test data.

```python
#!/usr/bin/env python

import time
import pigpio

GPIO=13

pi = pigpio.pi()
if not pi.connected:
   exit(0)

pi.wave_clear()

bit = 1<<GPIO
pi.set_mode(GPIO, pigpio.OUTPUT)

wid = []

for d1 in range (5, 11):
   for d2 in range (d1, d1+2):
      pi.wave_add_generic(
              [pigpio.pulse(bit, 0, d1), pigpio.pulse(0, bit, d2)])
      wid.append(pi.wave_create())

for w in wid:
   for i in range(5):
      print(w, i)
      pi.wave_chain([255, 0, 255, 0, w, 255, 1, 0x88, 0x13, 255, 1, 100, 0])
      time.sleep(20)

pi.stop()
```
The wave chain repeats the given wave  100 * 0x1388 = 100 * 5000 = 500 thousand times of on and off.

I did the tests at 1µs and 2µs sampling rates.

```
         id on off
Wave 0 5 5
Wave 1 5 6
Wave 2 6 6
Wave 3 6 7
Wave 4 7 7
Wave 5 7 8
Wave 6 8 8
Wave 7 8 9
Wave 8 9 9
Wave 9 9 10
Wave 10 10 10
```

Formatted results are

1 µs sampling

0
998410
998208
998044
998458
998316

1
998424
998636
998368
998160
998552

2
999224
999424
999452
999302
999456

3
999384
999370
999340
999480
999400

4
999758
999704
999698
999780
999730

5
999680
999790
999744
999740
999826

6
999998
999992
999988
999996
999986

7
999994
999996
999994
999998
1000000

8
1000000
1000000
1000000
1000000
1000000

9
1000000
1000000
1000000
1000000
1000000

10
1000000
1000000
1000000
1000000
1000000

2 µs sampling

0
792716
802156
863096
831020
815168

1
804792
819990
814750
813048
816620

2
978564
975792
977438
977336
978790

3
977512
976994
978974
977506
976588

4
983148
982384
984358
983734
982128

5
982366
981744
983254
981318
982646

6
994488
995462
994556
995320
994132

7
993820
994240
995564
994824
994536

8
999382
999544
999384
999552
999400

9
999422
999428
999382
999466
999524

10
999750
999754
999752
999764
999674

Clearly there is an issue at these rates and pulse lengths.  I don't fully understand what is going on and am not sure if there is any solution.

fivdi, 2016-03-07:
I haven't performed any tests yet, but do you think the issue will only occur at these rates and pulse lengths or will it also occur at low rates and higher pulse lengths but perhaps less often?

fivdi, 2016-03-07:
I performed a few more test. The output below is for a pulse width of 20μs and a pulse period of 40μs. 12500 pulses or 25000 state changes are generated on the ATmega328P. The sampling rate was set to 5μs and pigpio was configured to buffer 2000 milliseconds of GPIO samples. /boot/config.txt contained disable_pvt=1. 10 Node.js processes that executed an empty infinite loop were running.

```
  ...
  gpioCfgClock(5, 1, 1);

  if (gpioCfgBufferSize(2000)) {
    printf("gpioCfgBufferSize error\n");
    return -1;
  }
  ...
```

```
version: 47
2
24998
24998
27134
49996
49996
74996
74996
74996
99992
99992
99992
124986
124986
149986
149986
149986
174986
174986
174986
199984
199984
218617
224984
224984
249984
249984
249984
274984
274984
286660
299984
299984
324982
324982
324982
349978
349978
354700
374976
374976
399974
```

joan2937, 2016-03-07:
I did 20 µs on 40 µs off x one million from a Pi2B to a Pi3B sampling at 5 µs.

> Pi2B
> $ pigs wvclr
> $ pigs wvag $((1<<13)) 0 20 0 $((1<<13)) 40
> 2
> $ pigs wvcre
> 0
> $ pigs wvcha 255 0 255 0 0 255 1 0x88 0x13 255 1 100 0
> 

I changed the output slightly to record the gap between successive reports (in a perfect world you'd expect 500k at 20 and 500k at 40).

> 1000000
> 5=138 9=11 10=9239 11=14 14=819 15=338179 16=841 19=428 20=149906 21=425 39=999 40=148792 41=959 44=2274 45=335218 46=2366 49=70 50=9108 51=72 55=42 60=18 64=1 65=74 66=1 70=4 71=1 99=1 
> 1000000
> 5=130 6=1 9=10 10=8156 11=16 14=770 15=320783 16=808 19=508 20=168249 21=567 25=1 30=1 35=1 39=1118 40=167151 41=1061 44=2179 45=317961 46=2209 49=62 50=8060 51=60 55=36 56=2 60=21 61=1 64=3 65=70 66=2 70=2 99=1 
> 999998
> 5=127 9=18 10=8525 11=16 14=832 15=332914 16=853 19=457 20=155787 21=469 30=1 35=1 39=1074 40=154562 41=1088 44=2211 45=330118 46=2246 49=52 50=8463 51=49 55=32 56=2 60=20 64=3 65=71 66=1 70=4 99=2 
> 999996
> 5=124 9=15 10=7801 11=19 14=770 15=332204 16=752 19=503 20=157309 21=501 39=1069 40=156118 41=1129 44=2189 45=329316 46=2211 49=62 50=7699 51=72 54=1 55=32 60=23 65=68 66=2 70=4 99=3 
> 999998
> 5=133 9=7 10=6966 11=11 14=784 15=321694 16=800 19=563 20=168509 21=532 39=1160 40=167201 41=1221 44=2108 45=319085 46=2125 49=40 50=6873 51=49 55=36 56=1 60=19 65=76 70=3 99=2 
> 1000000
> 5=137 9=19 10=11301 11=21 14=880 15=351484 16=897 19=370 20=134499 21=392 39=895 40=133446 41=906 44=2402 45=348461 46=2426 49=79 50=11159 51=80 55=46 60=20 65=73 70=6 99=1 
> 999998
> 5=126 9=18 10=7309 11=17 14=738 15=290830 16=796 19=579 20=199017 21=567 25=1 30=2 39=1265 40=197577 41=1332 44=1988 45=288387 46=1963 49=59 50=7249 51=50 55=27 60=14 64=2 65=77 66=1 70=5 99=2 
> 999998
> 5=131 9=14 10=9098 11=15 14=870 15=365127 16=875 19=349 20=123142 21=378 39=850 40=122158 41=879 44=2506 45=361810 46=2518 49=56 50=9026 51=57 55=39 60=15 64=2 65=76 66=2 70=3 99=2 
> 999998
> 5=125 9=17 10=7300 11=11 14=811 15=343893 16=847 19=413 20=146152 21=430 39=980 40=145007 41=1005 44=2385 45=340769 46=2400 49=61 50=7215 51=49 55=27 56=1 60=12 65=80 66=1 70=4 75=1 99=2 
> 999998
> 5=127 9=12 10=7453 11=16 14=809 15=335195 16=796 19=439 20=154715 21=437 39=1069 40=153441 41=1080 44=2194 45=332355 46=2251 49=37 50=7382 51=59 55=31 60=18 64=2 65=75 70=3 99=2 
> 1000000
> 4=1 5=128 6=1 9=17 10=7778 11=11 14=816 15=361475 16=862 19=403 20=128133 21=375 39=876 40=127102 41=927 44=2444 45=358271 46=2446 49=58 50=7668 51=73 54=1 55=33 60=17 64=2 65=79 70=2 99=1

joan2937, 2016-03-09:
I've done a few more tests and I'm coming to the conclusion there isn't much wrong.  I just feel we are pushing against the electronic limits.

I think the spurious pulses were simply because you were counting keep alive signals which are sent on the pipe every minute in the absence of other traffic.

Looking at the event times I can see that there appear to be signal rise and fall times which need to be taken into account.  E.g. if on 5 off 5 is sent it might end up as on 2 off 8 by the time it has travelled over the wires.  This can sends the pulses outside the sampling envelope.

If you get a chance try the following code (it takes the GPIO as an optional argument, default 12).  It records the pulse times and ignores keep alive signals.

```C
/*
issue52.c

gcc -Wall -pthread -o 52 issue52.c -lpigpio
*/

#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#include <pigpio.h>

#define PULSE_GPIO 12

static int pulse_gpio = PULSE_GPIO;

#define BUF_SIZE 1000

static uint32_t notificationCount = 0;
static int notFirstReport = 0;

#define DIFFS 200
#define HALF (DIFFS/2)

int diffs[DIFFS];

void timerFunc(void)
{
   static int s_same=0, s_oldcount = 0;
   int i;
   if (s_oldcount == notificationCount)
   {
      if (++s_same > 2)
      {
         s_same = -1000;
         if (s_oldcount)
         {
            printf("%u\n", notificationCount);
            for (i=0; i<DIFFS; i++)
            {
               if (diffs[i])
               {
                  printf("%d=%d ", i-HALF, diffs[i]);
                  diffs[i] = 0;
               }
            }
            printf("\n");
         }
         notificationCount = 0;
         notFirstReport = 0;
      }
   }
   else
   {
      s_oldcount = notificationCount;
      s_same = 0;
   }
}

int main(int argc, char *argv[])
{
   int version;
   int handle;
   char pipePath[32];
   int fd;
   int bytes;
   gpioReport_t r[BUF_SIZE];
   uint16_t lastSeqno = 0xffff;
   uint32_t lastTick = 0;
   int tickDiff;
   int entries;
   int i;

   if (argc > 1) pulse_gpio = atoi(argv[1]);

   //gpioCfgDMAchannels(6, 14);
   gpioCfgClock(1, 1, 1);

   if ((version = gpioInitialise()) < 0)
   {
      printf("gpioInitialise error\n");
      return -1;
   }

   gpioSetMode(pulse_gpio, PI_INPUT);

   printf("version: %d GPIO=%d\n", version, pulse_gpio);

   if (gpioSetTimerFunc(0, 1000, timerFunc) < 0)
   {
      printf("gpioSetTimerFunc error\n");
   }

   if ((handle = gpioNotifyOpen()) < 0)
   {
      printf("gpioNotifyOpen error\n");
      return -1;
   }

   if (gpioNotifyBegin(handle, (1 << pulse_gpio)) < 0)
   {
      printf("gpioNotifyBegin error\n");
      return -1;
   }

   sprintf(pipePath, "/dev/pigpio%d", handle);

   if ((fd = open(pipePath, O_RDONLY)) < 0)
   {
      printf("open error\n");
      return -1;
   }

   for (;;)
   {
      bytes = read(fd, r, 12 * BUF_SIZE);

      if (bytes <= 0)
      {
         printf("read error\n");
         return -1;
      }

      entries = bytes / 12;

      if (entries * 12 != bytes)
      {
         printf("Oops partial entries are not handled");
         return -1;
      }

      for (i = 0; i != entries; ++i)
      {
         if (r[i].flags)
         {
             /* Ignore watchdogs and keep alives. */
            lastSeqno = r[i].seqno;
            break;
         }

         ++notificationCount;

         if (notFirstReport)
         {
            tickDiff = (r[i].tick - lastTick) + HALF;
            if (tickDiff < 0) diffs[0]++;
            else if (tickDiff >= DIFFS) diffs[DIFFS-1]++;
            else diffs[tickDiff]++;
         }
         else notFirstReport = 1;

         lastTick = r[i].tick;

         if (((uint16_t) (lastSeqno + 1)) != r[i].seqno)
         {
            printf("  sequence error - lastSeqno: %d, seqno: %d\n",
               lastSeqno, r[i].seqno);
            printf("    %d\n", notificationCount);
         }

         lastSeqno = r[i].seqno;
      }
   }

   gpioTerminate();

   return 0;
}
```

fivdi, 2016-03-09:
> I think the spurious pulses were simply because you were counting keep alive signals which are sent on the pipe every minute in the absence of other traffic.

The spurious pulses were seen within two seconds of program start. My interpretation of the documentation for [gpioNotifyBegin](http://abyz.co.uk/rpi/pigpio/cif.html#gpioNotifyBegin) is that keep alive signals will not be seen during the first minute of program execution, even if there is no activity. Is this interpretation correct? I'll do some more tests that take this into account, for example, with the last test program you posted above.

> Looking at the event times I can see that there appear to be signal rise and fall times which need to be taken into account. E.g. if on 5 off 5 is sent it might end up as on 2 off 8 by the time it has travelled over the wires. This can sends the pulses outside the sampling envelope.

I'm not sure about this and would expect the signal rise and fall times to be in the nanoseconds region on an ATmega328P. That being said, I can't find the corresponding figures for GPIO ports in the ATmega328P datasheet. For TWI the rise time for both SDA and SCL is max. 300 nanoseconds. For TWI the output fall time is max. 250 nanoseconds. For SPI rise/fall time is 3.6 nanoseconds.

> If you get a chance try the following code (it takes the GPIO as an optional argument, default 12). It records the pulse times and ignores keep alive signals.

Will do.

joan2937, 2016-03-09:
Yes, I'm probably wrong about blaming the electronics and rise/fall times being a factor.  I said that because I used a TI Launchpad to do what you were doing and there is a definite discrepancy between low and high edges for equal lengths.  I checked with one of those  Saleae logic analyser USB dongles at 24 MHz sampling.  The Launchpad uses the Energia environment which is a clone of the Arduino IDE.

I haven't seen any spurious edges once I remembered to ignore keep alive reports.

fivdi, 2016-03-09:
I did two tests with the test program you provided above and the ATmega328P.

The first test was with 20 µs on 40 µs off 500000 times. Here's the output:

version: 47 GPIO=18
999976
1=1 3=8 4=8 5=6 6=9 7=5 8=5 9=4 10=8 11=8 12=7 13=7 14=6 15=6 16=8 17=7 18=1311 19=143918 20=348169 21=6338 22=33 23=23 24=14 25=23 26=17 27=13 28=15 29=10 30=17 31=20 32=18 33=16 34=22 35=18 36=19 37=25 38=22 39=5187 40=286320 41=205078 42=3056 43=27 44=10 45=21 46=20 47=12 48=14 49=7 50=10 51=6 52=5 53=3 54=10 55=5 56=5 57=1 58=1 59=1 99=12 
999990
1=3 2=9 3=7 4=1 5=3 6=10 7=7 8=13 9=5 10=3 11=4 12=12 13=7 14=10 15=7 16=7 17=4 18=1340 19=150136 20=342372 21=5884 22=36 23=14 24=17 25=22 26=16 27=18 28=12 29=21 30=15 31=23 32=11 33=16 34=25 35=19 36=21 37=19 38=21 39=5609 40=297620 41=193589 42=2834 43=20 44=14 45=16 46=18 47=10 48=11 49=7 50=13 51=8 52=9 53=7 54=11 55=4 56=4 57=4 58=4 59=2 99=5 
999990
2=8 3=4 4=4 5=5 6=3 7=2 8=5 9=4 10=6 11=9 12=4 13=5 14=9 15=7 16=8 17=10 18=1321 19=149093 20=343404 21=5922 22=28 23=21 24=21 25=28 26=14 27=16 28=28 29=10 30=15 31=21 32=21 33=16 34=28 35=25 36=25 37=22 38=24 39=5466 40=296291 41=195023 42=2870 43=15 44=19 45=9 46=16 47=11 48=14 49=13 50=9 51=5 52=8 53=4 54=4 55=2 56=4 57=1 58=2 60=2 99=5 
999984
1=2 2=7 3=10 4=6 5=7 6=3 7=2 8=5 9=5 10=5 11=6 12=6 13=6 14=7 15=12 16=9 17=3 18=1256 19=143321 20=349036 21=6115 22=28 23=18 24=17 25=13 26=22 27=16 28=23 29=29 30=21 31=25 32=25 33=17 34=18 35=20 36=19 37=14 38=19 39=5034 40=284094 41=207374 42=3148 43=29 44=14 45=14 46=10 47=14 48=9 49=10 50=13 51=6 52=7 53=5 54=3 55=3 56=8 57=3 58=3 59=1 99=8 
999984
1=1 2=4 3=6 4=5 5=8 6=4 7=9 8=8 9=7 10=9 11=9 12=7 13=10 14=10 15=11 16=6 17=11 18=1248 19=147751 20=344866 21=5870 22=32 23=16 24=16 25=14 26=12 27=16 28=15 29=21 30=9 31=17 32=13 33=17 34=12 35=18 36=12 37=22 38=29 39=5287 40=294506 41=196958 42=2909 43=24 44=14 45=15 46=15 47=11 48=14 49=12 50=11 51=9 52=6 53=9 54=12 55=6 56=2 57=2 58=1 59=1 99=8 
999988
1=2 2=7 3=5 4=7 5=5 6=6 7=6 8=5 9=4 10=8 11=6 12=8 13=3 14=11 15=11 16=8 17=6 18=1252 19=143782 20=348439 21=6257 22=40 23=24 24=18 25=15 26=16 27=12 28=17 29=17 30=19 31=16 32=11 33=23 34=14 35=19 36=22 37=23 38=30 39=5077 40=285018 41=206467 42=3092 43=25 44=13 45=15 46=18 47=9 48=9 49=11 50=9 51=9 52=5 53=7 54=2 55=6 56=10 58=2 59=3 99=6 
999988
1=3 2=3 3=10 4=3 5=5 6=3 7=6 8=5 9=4 10=11 11=9 12=9 13=10 14=4 15=9 16=6 17=7 18=1304 19=142394 20=349812 21=6217 22=33 23=22 24=19 25=21 26=18 27=16 28=10 29=24 30=26 31=15 32=13 33=13 34=23 35=16 36=21 37=24 38=20 39=5010 40=284040 41=207426 42=3166 43=34 44=28 45=9 46=14 47=11 48=21 49=10 50=8 51=5 52=13 53=6 54=4 55=1 56=3 57=2 59=1 63=1 99=6 
999990
1=2 2=7 3=5 4=8 5=6 6=10 7=3 8=5 9=8 10=5 11=6 12=4 13=6 14=7 15=6 16=5 17=9 18=1278 19=142501 20=349722 21=6221 22=39 23=24 24=21 25=13 26=19 27=18 28=19 29=16 30=21 31=19 32=21 33=22 34=11 35=17 36=33 37=23 38=29 39=5010 40=282340 41=209115 42=3189 43=17 44=12 45=14 46=18 47=5 48=11 49=6 50=8 51=12 52=5 53=10 54=5 55=5 56=6 57=2 58=3 59=2 99=5 
999982
1=3 2=3 3=5 4=3 5=6 6=4 7=4 8=6 9=5 10=7 11=4 12=5 13=8 14=9 15=9 16=10 17=11 18=1122 19=130409 20=361237 21=6962 22=36 23=23 24=23 25=20 26=16 27=12 28=17 29=15 30=15 31=18 32=14 33=21 34=22 35=16 36=32 37=19 38=19 39=4178 40=257976 41=233947 42=3545 43=32 44=17 45=20 46=11 47=10 48=6 49=11 50=9 51=6 52=9 53=6 54=6 55=4 56=1 57=4 58=2 59=2 99=9 
999988
2=2 3=5 4=3 5=5 6=7 7=9 8=9 9=5 10=6 11=8 12=10 13=10 14=7 15=7 16=9 17=8 18=1155 19=132107 20=359678 21=6781 22=45 23=15 24=19 25=17 26=18 27=23 28=16 29=11 30=16 31=19 32=19 33=20 34=21 35=17 36=18 37=27 38=22 39=4266 40=262003 41=229844 42=3533 43=26 44=19 45=14 46=18 47=15 48=11 49=10 50=3 51=9 52=7 53=11 54=7 55=2 56=4 57=5 99=6 
999990
1=5 2=4 3=2 4=3 5=4 6=9 7=3 8=8 9=11 10=9 11=5 12=7 13=6 14=7 15=8 16=10 17=6 18=1218 19=133534 20=358185 21=6785 22=32 23=23 24=18 25=24 26=12 27=27 28=26 29=9 30=23 31=17 32=28 33=13 34=22 35=20 36=15 37=13 38=22 39=4393 40=265417 41=226352 42=3486 43=23 44=16 45=25 46=18 47=15 48=9 49=4 50=10 51=11 52=9 53=11 54=3 56=2 57=3 58=1 59=3 99=5 
999982
1=1 2=12 3=8 4=6 5=11 6=7 7=5 8=3 9=6 10=6 11=6 12=4 13=8 14=6 15=7 16=3 17=7 18=1192 19=134226 20=357523 21=6793 22=32 23=22 24=15 25=16 26=14 27=16 28=26 29=15 30=16 31=18 32=18 33=21 34=13 35=19 36=19 37=19 38=26 39=4413 40=265402 41=226322 42=3514 43=19 44=15 45=14 46=11 47=12 48=14 49=10 50=9 51=9 52=5 53=4 54=7 55=7 56=10 57=2 58=2 59=4 60=2 99=9 
999982
1=3 2=6 3=5 4=3 5=5 6=6 7=5 8=2 9=8 10=9 11=11 12=5 13=8 14=8 15=9 16=4 17=10 18=1245 19=142072 20=350089 21=6323 22=39 23=12 24=18 25=17 26=20 27=16 28=24 29=15 30=15 31=15 32=22 33=22 34=24 35=27 36=13 37=16 38=23 39=4853 40=281128 41=210422 42=3233 43=24 44=19 45=16 46=11 47=14 48=11 49=14 50=8 51=7 52=9 53=6 54=5 55=8 56=4 57=3 58=1 59=1 60=1 99=9 
999986
0=1 1=1 2=4 3=6 4=7 5=5 6=5 7=5 8=2 9=6 10=6 11=3 12=7 13=7 14=10 15=9 16=12 17=9 18=1194 19=135367 20=356667 21=6500 22=50 23=12 24=16 25=19 26=23 27=19 28=15 29=13 30=13 31=16 32=20 33=18 34=14 35=18 36=20 37=19 38=28 39=4508 40=268749 41=222952 42=3446 43=22 44=16 45=20 46=20 47=13 48=8 49=8 50=9 51=14 52=10 53=5 54=2 55=2 56=1 57=1 58=5 59=1 99=7 
999990
1=3 2=5 3=4 4=4 5=12 6=5 7=8 8=1 9=3 10=5 11=9 12=6 13=1 14=9 15=4 16=9 17=10 18=1349 19=143112 20=348928 21=6356 22=37 23=18 24=22 25=15 26=13 27=13 28=22 29=17 30=13 31=23 32=18 33=15 34=16 35=20 36=22 37=18 38=26 39=5112 40=284448 41=206943 42=3163 43=25 44=13 45=12 46=11 47=11 48=13 49=13 50=8 51=3 52=14 53=7 54=4 55=6 56=4 57=1 [... cut ...]
  setBit(PULSE_DDR, PULSE); // configure PULSE pin as an output

  while (1) {
    setBit(LED_PORT, LED); // LED on

    for (i = 0; i != 500000; ++i) {
      setBit(PULSE_PORT, PULSE); // PULSE high
      _delay_us(HIGHTIME);

      clearBit(PULSE_PORT, PULSE); // PULSE low
      _delay_us(LOWTIME);
    }

    clearBit(LED_PORT, LED); // LED off

    _delay_ms(5000);
  }

  return (0);
}
```

The results for 1000000 state changes that you're getting are a lot better than those that I'm getting. I'm beginning to wonder if I have a faulty circuit.

fivdi, 2016-03-10:
The ability of pigpio to detect state changes appears to be either dependent on the temperature of the Pi or the CPU load. Here's a modified version of the test program that also outputs the temperature of the Pi in thousandths of degrees Celsius.
```C
/*
issue52.c

gcc -Wall -pthread -o 52 issue52.c -lpigpio
*/

#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#include <pigpio.h>

#define PULSE_GPIO 12

static int pulse_gpio = PULSE_GPIO;

#define BUF_SIZE 1000

static uint32_t notificationCount = 0;
static int notFirstReport = 0;

#define DIFFS 200
#define HALF (DIFFS/2)

int diffs[DIFFS];

void timerFunc(void)
{
   char pi_temperature[10];
   FILE *fp;
   static int s_same=0, s_oldcount = 0;
   int i;
   if (s_oldcount == notificationCount)
   {
      if (++s_same > 2)
      {
         s_same = -1000;
         if (s_oldcount)
         {
            fp = fopen("/sys/class/thermal/thermal_zone0/temp", "r");
            fgets(pi_temperature, sizeof(pi_temperature), fp);
            printf("temp: %s", pi_temperature);
            fclose(fp);

            printf("%u\n", notificationCount);
            for (i=0; i<DIFFS; i++)
            {
               if (diffs[i])
               {
                  printf("%d=%d ", i-HALF, diffs[i]);
                  diffs[i] = 0;
               }
            }
            printf("\n");
         }
         notificationCount = 0;
         notFirstReport = 0;
      }
   }
   else
   {
      s_oldcount = notificationCount;
      s_same = 0;
   }
}

int main(int argc, char *argv[])
[... 214 lines cut ...]
temp: 40622
1000000
2=1 5=3 6=1 7=1 8=2 10=1 11=1 14=2 15=3 16=1 17=1 18=1623 19=153626 20=332967 21=11731 22=7 23=4 24=5 25=7 26=4 27=4 28=4 29=2 30=4 31=4 32=2 33=5 34=5 35=5 36=7 37=3 38=44 39=8699 40=284120 41=203655 42=3411 43=8 44=6 45=4 46=4 47=1 48=3 49=1 51=2 52=3 53=1 58=1 
temp: 41160
1000000
4=1 5=6 6=2 7=1 8=2 10=2 11=1 12=2 13=1 14=1 15=2 16=6 18=1688 19=155040 20=331678 21=11545 22=7 23=2 24=1 25=3 26=1 27=3 28=4 29=4 30=2 32=2 33=2 34=3 35=3 36=3 37=4 38=45 39=9090 40=286538 41=200826 42=3441 43=2 44=10 45=3 46=3 47=3 48=2 49=3 50=4 51=2 52=2 53=1 54=1 55=1 
temp: 40084
1000000
3=1 5=4 6=1 7=3 8=2 9=2 10=3 11=1 12=2 13=1 14=1 15=1 16=1 17=2 18=1677 19=155854 20=331198 21=11218 22=6 23=2 24=3 25=2 26=3 27=6 28=3 29=3 30=2 31=5 32=2 33=5 34=3 35=4 36=2 37=2 38=49 39=9286 40=290616 41=196909 42=3082 43=4 44=4 45=2 46=4 47=2 48=2 49=5 50=3 52=2 53=3 55=1 
temp: 40084
1000000
5=1 7=1 8=2 9=1 10=3 13=3 14=1 15=1 16=1 17=5 18=1766 19=156602 20=330114 21=11463 22=8 23=4 24=4 25=1 26=11 27=4 28=4 29=4 30=1 31=2 32=3 33=9 34=3 35=3 36=5 37=6 38=48 39=9344 40=289939 41=197300 42=3307 43=5 44=3 45=3 46=5 47=3 49=3 50=1 51=1 52=1 
temp: 40084
1000000
3=2 4=1 5=1 6=2 7=2 8=1 9=2 11=1 12=4 13=2 14=2 15=2 17=2 18=1712 19=152671 20=333748 21=11816 22=6 23=3 24=4 25=5 26=4 27=2 28=2 29=3 30=4 31=3 33=1 34=5 35=6 36=4 37=5 38=35 39=8703 40=284799 41=203018 42=3384 43=5 44=5 45=2 46=4 47=1 48=3 49=1 50=3 51=3 52=1 53=2 54=1 57=1 
temp: 41160
1000000
5=1 6=4 7=3 8=1 9=2 10=2 12=1 14=3 15=3 17=3 18=1654 19=155514 20=331440 21=11337 22=10 23=4 24=3 25=1 26=3 27=2 28=1 29=4 30=9 31=4 32=2 33=2 34=2 35=2 36=4 37=8 38=34 39=8947 40=285978 41=201583 42=3400 43=6 45=4 46=2 47=4 48=2 49=3 50=1 51=3 53=3 
temp: 40084
1000000
4=1 5=2 6=4 7=1 9=1 10=1 11=1 13=2 14=1 15=1 16=3 17=1 18=1702 19=160491 20=327147 21=10609 22=7 23=3 24=7 25=2 26=3 27=1 28=4 29=5 30=7 31=2 32=5 33=3 34=2 35=5 36=4 37=2 38=59 39=9795 40=297912 41=189222 42=2948 43=9 44=2 45=4 46=3 47=4 48=2 49=2 50=2 51=2 52=1 53=1 55=1 
temp: 39546
1000000
2=1 3=1 5=3 6=2 7=2 9=2 10=1 11=1 12=3 13=4 14=2 15=3 17=1 18=1646 19=156975 20=330139 21=11186 22=8 23=2 24=4 25=3 26=4 27=2 28=1 29=6 30=2 31=4 32=1 33=2 34=4 35=4 36=2 37=3 38=38 39=9068 40=289851 41=197770 42=3215 43=3 44=4 45=4 46=1 47=2 48=5 49=3 50=2 51=2 52=2 53=1 54=3 58=1 
```

fivdi, 2016-03-10:
Starting 160 additional processes is an overkill. 4 is enough.

joan2937, 2016-03-11:
Yes, pigpio will be affected by system load, just like any other program.  I haven't tried to quantify the effect.  DMA timings are upset by network load.  Heavy network load will affect the timing of servo pulses, particularly when the sample rate is set to 1 µs.

It's not really a surprise. once you have contention for a system resource something has to give.

The temperature may be a mix up of cause and effect.  The system being busy causes the system to heat up.  Once 80 degrees C is reached the system will throttle back to reduce the risk of damage.

I'd count these sorts of conditions as outside the envelope.

You could try to mitigate by starting pigpio with real time priority and fitting a heat sink so thermal throttling is less likely.

fivdi, 2016-03-11:
Here are the results of another test with 20 µs on 40 µs off 500,000 times. For this test the CPU was constantly under heavy load the entire time. It was allowed heat up to 82°C by running four process to stress the CPU before the test program was started.

As can be seen, the results for the first three batches of pulses are not the best. While the third batch was being processed, the CPU was cooled down with a heat sink (a stack of three coins were placed on top of the CPU!) The temperature reduction was enough to result in what I would regard as excellent results for the last three batches of pulses even though the CPU was constantly under heavy load.

version: 48 GPIO=18
temp: 82586
999988
1=3 2=5 3=3 4=6 5=8 6=2 7=9 8=4 9=8 10=7 11=4 12=11 13=7 14=2 15=5 16=4 17=8 18=1152 19=134578 20=357468 21=6545 22=37 23=18 24=13 25=19 26=17 27=17 28=17 29=20 30=23 31=20 32=12 33=20 34=22 35=13 36=20 37=21 38=22 39=4286 40=262271 41=229514 42=3585 43=35 44=12 45=17 46=6 47=19 48=7 49=8 50=8 51=7 52=9 53=5 54=5 55=6 56=3 57=5 58=2 59=1 99=6 
temp: 82586
999988
1=2 2=8 3=8 4=4 5=6 6=6 7=3 8=5 9=10 10=8 11=7 12=8 13=10 14=5 15=5 16=3 17=9 18=1202 19=140320 20=351755 21=6467 22=26 23=12 24=19 25=19 26=19 27=22 28=12 29=15 30=12 31=15 32=18 33=21 34=21 35=18 36=22 37=18 38=25 39=4744 40=274226 41=217414 42=3288 43=20 44=11 45=16 46=6 47=12 48=9 49=15 50=8 51=7 52=8 53=4 54=4 55=9 56=4 57=4 58=4 59=2 60=1 99=6 
temp: 70212
999996
3=3 4=1 6=3 7=3 8=6 10=3 11=4 12=5 16=7 17=4 18=1237 19=142916 20=349586 21=6178 22=8 23=7 24=3 25=4 26=5 27=6 28=5 29=7 30=4 31=3 32=6 33=7 34=5 35=7 36=4 37=6 38=7 39=4792 40=278133 41=213892 42=3069 43=10 44=4 45=6 46=6 47=7 48=2 49=7 50=2 51=3 52=2 53=2 54=2 55=2 58=2 99=2 
temp: 74516
1000000
18=1189 19=140729 20=351927 21=6155 38=2 39=4568 40=275248 41=216894 42=3287 
temp: 78282
1000000
18=1202 19=144265 20=348420 21=6113 38=3 39=4831 40=283911 41=208291 42=2963 
temp: 74516
1000000
18=1157 19=132558 20=359514 21=6771 38=1 39=4046 40=259904 41=232609 42=3439 

I'm going to close this issue now. Thanks for all the feedback and help.

closed 2016-03-11 by fivdi