joan2937/pigpio #55: Raspberry version check

scryver, 2016-03-17:
I have recently updated my pigpio library and now it says I can't change certain pins. I tracked it down to a wrong value report (or comparison) of the hardware revision. The python function get_hardware_revision() returns 19 and from pigpio's standpoint that should be a raspberry pi B, but I got a B+.

scryver, 2016-03-17:
Some extra info...
cat /proc/cpuinfo reports:
`
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 2.00
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 0013
Serial		: 000000000ad70368
`

MrYsLab, 2016-03-17:
@scryver The hardware version of 19 actually is a B+ (PCB rev 1.2). You can check this [here.](http://elinux.org/RPi_HardwareHistory).

I am running V49 and get the same results as you for pigpio.VERSION.

joan2937, 2016-03-17:
Sorry about that.  I should have corrected my mistake in V50.

scryver, 2016-03-17:
Nice I'll try it later this evening and will report back.

scryver, 2016-03-17:
Couldn't wait to try, so yeah as expected it works perfectly. Closing this issue.

closed 2016-03-17 by scryver