joan2937/pigpio #59: Independent PWMs on two gpios do not works

nachoplus, 2016-04-25:
Hi,
I need to drive two steppers using independent PWM hardware with variable frequency.  I use gpio 12 and 13 that is supposed to have different channels but when I change the frequency of one of them the other is disturbed, piscope shows spikes. When I stop changing, each channels get its right frequency. The problem is when setting freq with  hardware_PWM function. This is the code:

```
#!/usr/bin/python
import pigpio
import time
if __name__ == '__main__':
	pi=pigpio.pi("cronostamper")
	pi.set_mode(12, pigpio.OUTPUT)
	pi.set_mode(13, pigpio.OUTPUT)
	pi.hardware_PWM(12,10,500000)
	for i in range(100):
		t1 = pi.get_current_tick()
		pi.hardware_PWM(13,i*1,500000)
		t2 = pi.get_current_tick()
		print pigpio.tickDiff(t1,t2)
		time.sleep(0.050)

```

Any advice are wellcome
Cheers,

joan2937, 2016-04-25:
At the moment changing one channel will cause a glitch on the other channel if it is also transmitting PWM.

I'll have a look at the datasheet to see if the glitch can be prevented.

joan2937, 2016-04-26:
I changed the title to more accurately reflect the actual behaviour.

I have been more intelligent in the way I set up the channels which means this should no longer happen.

The fix will be in V52 (when it is released).

nachoplus, 2016-04-26:
Ok. I'll be waiting for V52. I'll give you feedback when I try it.
Thank you very much! I really apreciate your fast support!

joan2937, 2016-04-26:
If you want to give it a try you could apply the following patch to V51 of pigpio.c.

Save as patch.txt and then

`patch pigpio.c patch.txt`

```
--- pigpio.c-V51	2016-04-26 17:42:48.539091253 +0100
+++ pigpio.c	2016-04-26 16:51:39.450230475 +0100
@@ -25,7 +25,7 @@
 For more information, please refer to <http://unlicense.org/>
 */
 
-/* pigpio version 51 */
+/* pigpio version 52 */
 
 /* include ------------------------------------------------------- */
 
@@ -1144,6 +1144,7 @@
 static uint32_t reportedLevel = 0;
 
 static int waveClockInited = 0;
+static int PWMClockInited = 0;
 
 static volatile gpioStats_t gpioStats;
 
@@ -7551,6 +7552,7 @@
    DBG(DBG_STARTUP, "");
 
    waveClockInited = 0;
+   PWMClockInited = 0;
 
    clock_gettime(CLOCK_REALTIME, &libStarted);
 
@@ -8958,6 +8960,7 @@
       stopHardwarePWM();
       initClock(0); /* initialise secondary clock */
       waveClockInited = 1;
+      PWMClockInited = 0;
    }
 
    p = rawWaveCBAdr(waveInfo[wave_id].topCB);
@@ -9194,6 +9197,7 @@
       stopHardwarePWM();
       initClock(0); /* initialise secondary clock */
       waveClockInited = 1;
+      PWMClockInited = 0;
    }
 
    dmaOut[DMA_CS] = DMA_CHANNEL_RESET;
@@ -11413,11 +11417,16 @@
       old_PWM_CTL = pwmReg[PWM_CTL] &
          (PWM_CTL_PWEN1 | PWM_CTL_MSEN1 | PWM_CTL_PWEN2 | PWM_CTL_MSEN2);
 
-      pwmReg[PWM_CTL] = 0;
+      if (!PWMClockInited)
+      {
+         pwmReg[PWM_CTL] = 0;
 
-      myGpioDelay(10);
+         myGpioDelay(10);
 
-      initHWClk(CLK_PWMCTL, CLK_PWMDIV, CLK_CTL_SRC_PLLD, 2, 0, 0);
+         initHWClk(CLK_PWMCTL, CLK_PWMDIV, CLK_CTL_SRC_PLLD, 2, 0, 0);
+
+         PWMClockInited = 1;
+      }
 
       if (pwm == 0)
       {
```

nachoplus, 2016-04-26:
Clean compilation and works perfectly! This was a blocking issue for me and now I can continue with my project!
Thank you very much!

joan2937, 2016-04-27:
Okay.  Unless I hear otherwise I'll close this as an issue when I release V52.

joan2937, 2016-05-01:
V52 released.

closed 2016-05-01 by joan2937