joan2937/pigpio #62: Q: SPI communication with BMP183

Mausy5043, 2016-05-06:
If you have a moment I would appreciate it very much if you could take a look at what I'm trying to do and explain to me why it is not working. 😃 

I have connected a [BMP183](https://www.adafruit.com/products/1900) to my Pi.
I've connected the instrument as follows:
```
BMP pin : RPi pin
========:===============
VIN     : 17  (3v3)
3vo     : N/C
GND     : 25  (GND)
SCLK    : 23  (GPIO11)
SDO     : 21  (GPIO09; MISO)
SDI     : 19  (GPIO10; MOSI)
CS      : 24  (GPIO08; CE0)
```
I've forked the BMP183 python class from PrezmoF and modified it for my implementation. The [measure.py](https://github.com/Mausy5043/bmp183) program works:
```
$ sudo python measure.py
[... calibration data in output removed...]
('Temperature: ', 22.899999999999999, 'deg C')
('Pressure: ', 1012.85, ' hPa')
```
I'm trying to achieve the same effect, but without `sudo` using the `pigpio` library in Python. 
I've created a small test program: [test.py](https://github.com/Mausy5043/bmp183/blob/master/test.py)
Sending `0xAA` to the instrument should yield a 16-bit word containing the decimal value 8600.
However, running this program returns 1 byte with the value 0.

I've also tried `pi.spi_write` followed by reading 3 bytes using `pi.spi_read`. The result is the same. Only 0-values are returned.

Any suggestions are very welcome

joan2937, 2016-05-08:
If the command 0xAA is meant to return two bytes send [0xAA, 0, 0] or three bytes.  The returned value should be in the second and third returned bytes.  SPI works by transferring bits, you send a bit and receive a bit.

```py
(count, rx_data) = pi.spi_xfer(hndl, [0xAA, 0,0])
if count == 3:
   print(rx_data[1], rx_data[2])
```

Mausy5043, 2016-05-08:
Briliant! That did the trick. Thanks!
👍

Mausy5043, 2016-05-10:
I'm still having some problems.
I can now read the contents of an address on the device but I can't seem to figure out how to send it a command by writing data to an address. Well, at least when reading back the result I get the wrong data.
I see that the python script that uses RPi.GPIO just does some bitbanging and that seems to work like a charm.

joan2937, 2016-05-10:
Do you have a script which demonstrates a fault?

Mausy5043, 2016-05-11:
I've prepared the [scripts `test1.py` and `test2.py`](https://github.com/Mausy5043/pigbmp183)  to demonstrate the problem. They use the modules `pigbmp183.py` and `bmp183.py` respectively (available in the same repo). 

```
$ ./test1.py ; sudo ./test2.py
TEST 1
Reading value stored at 0xAA
     Value stored at 0xAA : 8600
Writing value 0x2E into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 42211
Writing value 0x34 into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 42211

TEST 2
Reading value stored at 0xAA
     Value stored at 0xAA : 8600
Writing value 0x2E into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 26311
Writing value 0x34 into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 42212
```

Note, that TEST1 displays the value previously obtained from running TEST2. So, it seems that (at least sometimes) pigpiod is unable to write bytes into the BMP183 register.

Rebooting sometimes alleviates the problem for a while, but it is no guarantee for success.

Mausy5043, 2016-05-11:
In the spirit of full disclosure 😄 
I also have a 1-wire DS18B20 connected:
```
# Sensor pin       : R-Pi B+ pin
# =================:==============
# VIN   (red)      = 01  - 3v3
# Data  (yellow)   = 07  - GPIO04
# GND   (blue)     = 09  - GND
``` 
being interrogated (every 12s) by reading the kernel file somewhere in `/sys/bus/...`

And a DHT22 connected:
```
# Wiring (facing frontside of DHT22, left to right):
# Sensor pin       : R-Pi B+ pin
# =================:==============
# PWR              = 01  - 3v3
# data (digital)   = 12  - GPIO18 & R=4k7 > 3v3
# NC               = not connected
# GND              = 14  - GND
```
being read (every 12s) using your wonderful `DHTXXD` binary (as discussed in issue #60).

Both are performing near perfectly.

joan2937, 2016-05-11:
I've had a quick look at the BMP183 datasheet.  The first thing which leaps out is that the device requires SPI mode 3.  Your code appears to be using the default (SPI mode 0).  The first thing to try is opening in SPI mode 3.  e.g. pi.spi_open(0, 34000, 3) rather than pi.spi_open(0, 34000).

Mausy5043, 2016-05-12:
Well that completely escaped me.
But thanks to your suggestion I got to read the datasheet again and came across par. 5.2 which states:

> In the SPI protocol, the MSB of the register address remains is not used. For example, register 0xF7 is addressed by 0x77 in SPI mode. Bit 7 is replaced by an R/W bit (read: „1‟, write: „0‟).

This never made much sense to me ... until now. (Learn something every day 😉 )

So, I changed the code around a bit. Instead of instructing 0xF4, I instruct 0x74 (clearing the top bit) to write to 0xF4.
```
  print("Writing value 0x2E into address 0xF4")
  pi.spi_write(hndl, [0x74, 0x2E, 0])
```

Now I get this output:
```
$ ./test1.py ; sudo ./test2.py
TEST 1
Reading value stored at 0xAA
     Value stored at 0xAA : 8600
Writing value 0x2E into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 26426
Writing value 0x34 into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 42046

TEST 2
Reading value stored at 0xAA
     Value stored at 0xAA : 8600
Writing value 0x2E into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 26423
Writing value 0x34 into address 0xF4
Reading value stored at 0xF6
     Value stored at 0xF6 : 42049
```

Which is the expected output. 🎉  🎈 🍰 

Regarding the SPI mode: I tried `pi.spi_open(0, 34000, 3)` and `pi.spi_open(0, 34000)` which both yield the same results. But, `pi.spi_open(0, 34000, 1)` gives erroneous values. I'll force the mode 3 just to be on the safe side.

Again, thank you so very much. Your assistance is greatly appreciated.

Mausy5043, 2016-05-13:
Closing because the system has been running for about 24 hours without issues.

closed 2016-05-13 by Mausy5043