joan2937/pigpio #63: Multiple programs besides the daemon run

infusion, 2016-05-17:
Hi,

I have a question regarding a parallel usage of the daemon and other non-daemon programs at the same time. I use one program which writes via PWM to a pin - which unfortunately needs the daemon - and I have other programs which work on different pins at the same time. When the daemon runs, I get 
```
2016-05-17 02:54:22 initInitialise: Can't lock /var/run/pigpio.pid
```

Is it save to just write to a different pid file or are there any serious problems when doing this in parallel?

Thanks!

joan2937, 2016-05-17:
Only one copy of the pigpio library is allowed to run at a time.

The reason is that the library opens a rare resource, a DMA channel, and then starts sampling the GPIO at a high rate using that channel.

It is possible to start pigpio and tell it to use a different DMA channel, but this use is prevented by the lock on /var/run/pigpio.pid.  In extremis you could remove the lock code to allow multiple copies.

In fact there are very few DMA channels left.  Originally pigpio had about 13 channels (from 15) to select from, now its sometimes hard to find a free channel.

There is very little you can do when directly linked with the library that you can't do just as well by using the daemon.

If you do have such a need then I suggest you put all the time-critical code in one process and launch that instead of pigpiod.  Your process then in effect becomes the pigpiod and will automatically service daemon requests via its socket and pipe interface (unless you disable those interfaces with gpioCfg* options at process start).

infusion, 2016-05-17:
Thanks for your detailed explanation! In fact, it is what I did yesterday. I moved any critical code into one process. I just wanted to ask and it makes totally sense.

I must spread another thank-you for the great work you did on pigpio!

closed 2016-05-17 by infusion

SlySven, 2016-05-17:
BTW If you do run your program as a replacement for `pigpiod` you ought to use the same lock/runfile `/var/run/pigpio.pid` in the same way as `pigpiod`. I'm doing that for my UPiS daemon which uses the I<sup>2</sup>C bus and reserves a dedicated GPIO pin for a "_shut-down NOW_" "hardware" interrupt and it runs on startup as root - so it makes sense to run it as __the__ daemon so it can prevent other processes from touching/hogging resources that it wants.

joan2937, 2016-05-17:
That should be automatic as the locking is actually handled by the library during initialisation. The code is initGrabLockFile in pigpio.c.

SlySven, 2016-05-17:
Ah, I got that slightly wrong - my code is linked against `libpigpio` and using `pigpio.h` the header ___not__ `libpigpiod` and `pigpiod.h` - as I need to have different signal handling behaviour (my daemon needs to catch and handle SIGPIPE (and posibbly SIGPWR) and I do not want it to terminate if it gets SIGSTOP/SIGTSTP/SIGTTOUT/SIGTTIN/SIGCONT) - FWIIW I've clobbered the code base as per (my) PR #58...