joan2937/pigpio #64: Measure PWM timing

infusion, 2016-05-18:
Hi,

I read over the docs two times now and couldn't figure out how to measure the timing of an incoming pulse. In the arduino world, there is a function called `pulseIn()` for that purpose. Do you provide a similar functionality, which I missed somehow?

Thanks

fivdi, 2016-05-18:
One way to do it would be with [gpioSetAlertFunc](http://abyz.co.uk/rpi/pigpio/cif.html#gpioSetAlertFunc).

joan2937, 2016-05-18:
All pulseIn does is a busy spin waiting on pulse changes.  It then uses the number of cycles to calculate the pulse length.

You can do the same if you are happy with a busy spin, but use gpioTick() to record the microsecond tick at the start and end of the pulse (this rather assumes you are using C).

The above will be the most accurate but is wasteful.

If you are happy to get results within 10 µs then just use callbacks.  A callback gives you the GPIO, level, and the tick.  So pulseIn becomes record the tick, wait for the next callback, the current tick minus the last tick is the pulse duration.

There are plenty of examples in http://abyz.co.uk/rpi/pigpio/examples.html

Here is the Arduino pulseIn code.
```C
/*
  wiring_pulse.c - pulseIn() function
  Part of Arduino - http://www.arduino.cc/

  Copyright (c) 2005-2006 David A. Mellis

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA

  $Id: wiring.c 248 2007-02-03 15:36:30Z mellis $
*/

#include "wiring_private.h"
#include "pins_arduino.h"

/* Measures the length (in microseconds) of a pulse on the pin; state is HIGH
* or LOW, the type of pulse to measure.  Works on pulses from 2-3 microseconds
* to 3 minutes in length, but must be called at least a few dozen microseconds
* before the start of the pulse. */
unsigned long pulseIn(uint8_t pin, uint8_t state, unsigned long timeout)
{
	// cache the port and bit of the pin in order to speed up the
	// pulse width measuring loop and achieve finer resolution.  calling
	// digitalRead() instead yields much coarser resolution.
	uint8_t bit = digitalPinToBitMask(pin);
	uint8_t port = digitalPinToPort(pin);
	uint8_t stateMask = (state ? bit : 0);
	unsigned long width = 0; // keep initialization out of time critical area
	
	// convert the timeout from microseconds to a number of times through
	// the initial loop; it takes 16 clock cycles per iteration.
	unsigned long numloops = 0;
	unsigned long maxloops = microsecondsToClockCycles(timeout) / 16;
	
	// wait for any previous pulse to end
	while ((*portInputRegister(port) & bit) == stateMask)
		if (numloops++ == maxloops)
			return 0;
	
	// wait for the pulse to start
	while ((*portInputRegister(port) & bit) != stateMask)
		if (numloops++ == maxloops)
			return 0;
	
	// wait for the pulse to stop
	while ((*portInputRegister(port) & bit) == stateMask) {
		if (numloops++ == maxloops)
			return 0;
		width++;
	}

	// convert the reading to microseconds. The loop has been determined
	// to be 20 clock cycles long and have about 16 clocks between the edge
	// and the start of the loop. There will be some error introduced by
	// the interrupt handlers.
	return clockCyclesToMicroseconds(width * 21 + 16);
}
```

infusion, 2016-05-18:
Oh, I must have been blind. Sorry for the trivia question. It works like a charm!

closed 2016-05-18 by infusion