joan2937/pigpio #65: Funtion get_PWM_frequency() return 800Hz when hardware PWM freq is set to zero.

nachoplus, 2016-05-22:
Hello,

I found that get_PWM_frequency() reports 800Hz in stead of 0Hz when hardware_PWM freq is set to zero.  PWM really stop. Here I paste sample code showing the issue:

```
#!/usr/bin/python

import pigpio
if __name__ == '__main__':
	pi=pigpio.pi("cronostamper")
	pi.set_mode(13, pigpio.OUTPUT)
	for f in [142,2323,0,4320,1,0]:
		pi.hardware_PWM(13,f,500000)
		print "Hardware PWM/get_PWM_frequency()",f,"/",pi.get_PWM_frequency(13)
```

Executing this code shows:

Hardware PWM/get_PWM_frequency() 142 / 142
Hardware PWM/get_PWM_frequency() 2323 / 2323
Hardware PWM/get_PWM_frequency() 0 / 800
Hardware PWM/get_PWM_frequency() 4320 / 4320
Hardware PWM/get_PWM_frequency() 1 / 1
Hardware PWM/get_PWM_frequency() 0 / 800

Nacho

fivdi, 2016-05-22:
The above code always sets PWMduty to 500000. Does setting PWMduty to 0 when PWMfreq is 0 help? Does the call to hardware_PWM always return 0 or is an error code returned when PWMfreq is 0?

joan2937, 2016-05-22:
Strange as it may seem that is actually the intended behaviour.

See http://abyz.co.uk/rpi/pigpio/python.html#get_PWM_frequency

> Returns the frequency (in Hz) used for the GPIO.
> 
> For normal PWM the frequency will be that defined for the GPIO by set_PWM_frequency.
> 
> If a hardware clock is active on the GPIO the reported frequency will be that set by hardware_clock.
> 
> If hardware PWM is active on the GPIO the reported frequency will be that set by hardware_PWM.
> 
As hardware PWM is not active (0 switches it off) the default PWM frequency (or that set by set_PWM_frequency) is returned.

fivdi, 2016-05-22:
Is the intended behavior good?

joan2937, 2016-05-22:
Who's to say?  If the duty cycle is zero the frequency is irrelevant anyhow.

fivdi, 2016-05-22:
Yes, if the duty cycle is 0 the frequency is irrelevant. However, although I haven't tested it, I'd expect get_PWM_dutycycle to return 500000 in the above code. A frequency of 800Hz and a duty cycle of 50% implies 800 pulses per second, but in this case there are 0 pulses.

nachoplus, 2016-05-22:
I have not opinion about what is the good behavior. I just want a proper way to check when PWM pulse are stopped.

If I initialize soft PWM with pi.set_PWM_frequency(13,0) in the above test code then get_PWM_frequency()  return 10Hz (seen to be the soft PWM lower limit).

Alternative if I use dutycycle=0 to stop PWM and check it using  get_PWM_dutycycle(13) the code raised and pigpio.error: 'GPIO is not in use for PWM' if freq is also set to 0...

fivdi, 2016-05-22:
@nachoplus why not leave the frequency as is and set the duty cycle to 0 to turn PWM off?

The output of the following program:

```py
import pigpio

pi=pigpio.pi()
pi.set_mode(13, pigpio.OUTPUT)

# PWM on
pi.hardware_PWM(13,1000,500000)
print "get_PWM_frequency()/get_PWM_dutycycle()",pi.get_PWM_frequency(13),"/",pi.get_PWM_dutycycle(13)

# PWM off
pi.hardware_PWM(13,1000,0)
print "get_PWM_frequency()/get_PWM_dutycycle()",pi.get_PWM_frequency(13),"/",pi.get_PWM_dutycycle(13)
```
is:
```
get_PWM_frequency()/get_PWM_dutycycle() 1000 / 500000
get_PWM_frequency()/get_PWM_dutycycle() 1000 / 0
```
If `pi.get_PWM_dutycycle(13)` returns 0, PWM pulses are stopped.

joan2937, 2016-05-24:
I am reluctant to change a defined interface without a very good reason.  For all I know people may be depending on the currently defined interface in their code.

@fivdi gives a solution.  It seems good to me.

So I will not change the interface.

closed 2016-05-24 by joan2937

nachoplus, 2016-05-25:
OK. I have adopted the solotion propose by @fivdi . Thank you very much for supporting!