joan2937/pigpio #66: PI_NUM_I2C_BUS should be 4 when i2c-gpio enabled in config.txt

FishTest, 2016-05-28:
pi@raspberrypi:~ $ cat /boot/config.txt | grep i2c-gpio
dtoverlay=i2c-gpio

pi@raspberrypi:~ $ ls /dev/i2c*
/dev/i2c-3

pi@raspberrypi:~ $ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

when enabled dpi_lcd(vga666) in config.txt,gpio_i2c is the only way to enable i2c,it is supported by raspberry pi official kernel,so i sujjest that chang PI_NUM_I2C_BUS to 4 instead of 2 in pigpio.

joan2937, 2016-05-28:
I'm going to remove the bus number check completely.

At the moment its only real use is to add additional diagnostics to pin-point the failure in the debug log (/dev/pigerr) and avoid a speculative bus device open.

The error (-74, bad I2C bus) will still be reported when an attempt is made to open a non-existent bus.

It will be added to version 53 when released (few days to a few weeks).

joan2937, 2016-05-31:
Added to V53.

closed 2016-05-31 by joan2937