joan2937/pigpio #67: how to move a servo controller

converge, 2016-05-28:
I tried this to move a servo controller, but it wont move the servo.

Am I doing something wrong ?

`
#!/usr/bin/python

import pigpio
import time

pi = pigpio.pi()

pi.set_mode(17, pigpio.OUTPUT)

pi.set_PWM_frequency(17, 50)

time.sleep(1)

print (pi.get_PWM_frequency(17))
## output 50

##pi.set_PWM_frequency(17, 6) # 2.35%

pi.set_PWM_frequency(17, 32) # 2.35%

time.sleep(1)

pi.set_PWM_frequency(17, 0) # 2.35%

time.sleep(1)

pi.stop()
`

joan2937, 2016-05-28:
Not necessarily.  Can your servo can cope with a pulse width of 2500 µs?

I suggest you use the [set_servo_pulsewidth](http://abyz.co.uk/rpi/pigpio/python.html#set_servo_pulsewidth) which always uses 50 Hz and was designed for use with servos.

DBarney, 2016-06-01:
So I came here trying to get my servos working as well. I modified the script above following your suggestion with `set_servo_pulsewidth`.

```
#!/usr/bin/python

import pigpio
import time

pi = pigpio.pi()
pi.set_mode(18, pigpio.OUTPUT)

print ("mode: ", pi.get_mode(18))
print("setting to: ",pi.set_servo_pulsewidth(18, 1500))
print("set to: ",pi.get_servo_pulsewidth(18))

time.sleep(1)

print("setting to: ",pi.set_servo_pulsewidth(18, 1000))
print("set to: ",pi.get_servo_pulsewidth(18))

time.sleep(1)

print("setting to: ",pi.set_servo_pulsewidth(18, 2000))
print("set to: ",pi.get_servo_pulsewidth(18))

time.sleep(1)

pi.stop()
```

unfortunately the output shows that the pulsewidth was not set correctly, AND the servo does not move. Is there something I can do do debug this further? Or some setting I am missing in setting everything up?

joan2937, 2016-06-02:
The code looks fine.

Possibilities in increasing order of likeliness.

1. servo is bust
2. servo is underpowered
3. servo's control wire is not connected to GPIO 18
4. servo's ground is not connected to Pi ground.

If none of the above apply could you post a photo of your set-up and give details about your Pi model, pigpio version etc. etc.

DBarney, 2016-06-02:
Looks like I had forgotten to join the two grounds together. I appreciate your help and am very glad that it didn't turn out to be a software issue.

closed 2016-06-03 by joan2937