joan2937/pigpio #68: Bit Bang I2c examples

jrobinson-uk, 2016-05-29:
More examples for bb_i2c_zip would be really helpful, currently trying to read GPS data from a chip, trying to mimic existing C code in python.

executing:
print(gp.bb_i2c_zip(2,[4,0x42,2,7,1,0x85,3,0]))

and getting 
(-82, '') in return, but not sure what this means

joan2937, 2016-05-29:
[pigpio.error_text](http://abyz.co.uk/rpi/pigpio/python.html#pigpio.error_text)

-82 is I2C write failed

```
4,0x42      address 0x42
2           start
7,1,0x85    write 0x85
3           stop
0           end
```
Are you sure the device is connected properly?

jrobinson-uk, 2016-05-30:
Yeah connected and working, trying to RE implement this code (just the i2c
_getc function for now)

https://github.com/PiInTheSky/pits/blob/master/tracker/gps.c

which runs fine.

It uses clock stretching I believe, which I'm not sure about

joan2937, 2016-05-30:
I was wondering why you were using the bit bang routines rather than the standard Python smbus module or pigpio's SMBus/I2C calls.  The need for clock stretching is a valid reason.

Personally I'd try with the standard solutions before resorting to bit bang.

I'm not going to leave this open just as a documentation issue.  There is an awful lot I could document better if I had lots of time so I need to concentrate my efforts.

On the error, why are you writing if you are testing reading?

closed 2016-05-30 by joan2937

jrobinson-uk, 2016-05-30:
How do I make use of clock stretching using pigpio?

I can't try other solutions add the GPS is part of an add on board and
permanently connected to pin 2/3.

As to reading / writing. The original library seems to send a message to
request a byte. Not really sure, i2c is quite new to me.

joan2937, 2016-05-30:
Pins 2 and 3 are SDA and SCL which is the I2C bus.  I suggest you try `i2cdetect -y 1` to see if the device is visible.  If it is you should try standard calls first.  Perhaps experiment with the pigs i2c commands from the terminal.  E.g. http://abyz.co.uk/rpi/pigpio/pigs.html#I2CO