joan2937/pigpio #69: Bad Handle

biomed12, 2016-07-03:
#include<pigpio.h>
#include<stdio.h>

int main()
{

	if(gpioInitialise()<0)
	{	printf("gpio enabled succesfully!\n");
			}
		else
		{	printf("gpio couldn't have started!\n");
			}
	//serial port open
	int handle=0; //to test handle
	printf("prevalue of handle:%d\n",handle);
	if(handle=serOpen("/dev/ttyAMA0",9600,0)>=0)
	{printf("serial port opening:succesful!\n");}
		else{
		printf("serial port openin:failed!");}
	printf("postvalue of handle:%d\n",handle);

	//serial port close

	if(serClose(handle)==0)
	{	printf("serial port closed!\n");
			}
		else{
		printf("serial port closing failed!\n");
			}
	

gpioTerminate();
}

/*when i compile and run program serclose function gives Bad Handle(1) error. Thanks*/

biomed12, 2016-07-03:
When i compile the code everything is okey until serClose() function. After that gives me the bad handle(1) errror. like that:

http://hizliresim.com/qBpLaQ

joan2937, 2016-07-03:
I'm not sure what is going wrong with your code.  I suspect an error in the serOpen and test >=0 line or the serClose line.

I have simplified the logic as follows.

```C
#include"pigpio.h" //includes are not like that 
#include"stdio.h"

int main()
{
   int handle, status;

   if(gpioInitialise()>=0)
      printf("gpio enabled succesfully!\n");
   else
      printf("gpio couldn't have started!\n");

   handle=serOpen("/dev/ttyAMA0",9600,0);

   if (handle>=0)
      printf("serial port opening:succesful!\n");
   else
      printf("serial port openin:failed!");

   printf("handle:%d\n",handle);

   //serial port close

   status = serClose(handle);

   if (status == 0)
      printf("serial port closed!\n");
   else
      printf("serial port closing failed!\n");

   printf("status:%d\n",status);

   gpioTerminate();
}
```
See if that works.

joan2937, 2016-07-03:
There is an error in your serOpen line which should be.

if((handle=serOpen("/dev/ttyAMA0",9600,0))>=0)

closed 2016-07-03 by joan2937

biomed12, 2016-07-03:
Thanks dear joan2937. Code works fine!