joan2937/pigpio #70: Error while installing pigpio python library : This the Error : ->  "make: warning:  Clock skew detected.  Your build may be incomplete."

NilanjPatel, 2016-07-20:
pi@raspberrypi ~/PIGPIO $ sudo make
make: Warning: File 'Makefile' has modification time 1468166355 s in the future
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
pigpio.c: In function ‘fileApprove’:
pigpio.c:11717:15: warning: ‘mperm’ may be used uninitialized in this function [-Wuninitialized]
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -shared -o libpigpio.so pigpio.o command.o
strip --strip-unneeded libpigpio.so
size     libpigpio.so
   text	   data	    bss	    dec	    hex	filename
 223450	   5152	 593824	 822426	  c8c9a	libpigpio.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -shared -o libpigpiod_if.so pigpiod_if.o command.o
strip --strip-unneeded libpigpiod_if.so
size     libpigpiod_if.so
   text	   data	    bss	    dec	    hex	filename
  46834	   4156	  49244	 100234	  1878a	libpigpiod_if.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
gcc -shared -o libpigpiod_if2.so pigpiod_if2.o command.o
strip --strip-unneeded libpigpiod_if2.so
size     libpigpiod_if2.so
   text	   data	    bss	    dec	    hex	filename
  39576	   4144	   1844	  45564	   b1fc	libpigpiod_if2.so
gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
gcc -o pig2vcd pig2vcd.o
gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
gcc -o pigpiod pigpiod.o -L. -lpigpio -pthread -lrt
gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
gcc -o pigs pigs.o command.o
make: warning:  Clock skew detected.  Your build may be incomplete.

joan2937, 2016-07-20:
Have you set the date on your Pi?

NilanjPatel, 2016-07-21:
hey , thanks installation is done.
but during decoding 36 bit hid card (wiegand protocol) it gives following error : 

pi@raspberrypi ~ $ sudo python wiegand.py
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "wiegand.py", line 128, in <module>
    w = wiegand.decoder(pi, 14, 15, callback)
  File "/home/pi/wiegand.py", line 56, in __init__
    self.pi.set_mode(gpio_0, pigpio.INPUT)
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1119, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 914, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

closed 2016-07-21 by joan2937