joan2937/pigpio #71: Error while i try to run wiegand.py  to read 35 bit hid card on raspberry pi

NilanjPatel, 2016-07-21:
pi@raspberrypi ~/Downloads $ sudo python wiegand.py
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "wiegand.py", line 128, in <module>
    w = wiegand.decoder(pi, 29, 32, callback)
  File "/home/pi/Downloads/wiegand.py", line 56, in __init__
    self.pi.set_mode(gpio_0, pigpio.INPUT)
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1119, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 914, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

joan2937, 2016-07-21:
There is a message at the top of the listing.  You need to read it and take note.

If there is an actual error I need details of the error.

closed 2016-07-21 by joan2937

NilanjPatel, 2016-07-22:
pi@raspberrypi ~/Downloads $ sudo python wiegand.py 
Traceback (most recent call last):
  File "wiegand.py", line 128, in <module>
    w = wiegand.decoder(pi, 32, 29, callback)
  File "/home/pi/Downloads/wiegand.py", line 56, in __init__
    self.pi.set_mode(gpio_0, pigpio.INPUT)
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1119, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 901, in _u2i
    raise error(error_text(v))
pigpio.error: 'no permission to update GPIO'

joan2937, 2016-07-22:
Please read the documentation.