joan2937/pigpio #72: Reading BME 280 and Pigpio

Joey-1970, 2016-07-26:
Hello,

I'm searching for a example to read the I2C BME280. Has someone do something like this?
I get a handle but everything I want to read the result is always 0...
I work with the socket interface, but I wound thankful for any working example for doing it with pigpio...

Joachim

joan2937, 2016-07-27:
I haven't got any code for the BME280.  I tend to only write code for I2C devices I have to hand as it is hard to code for I2C devices using the specification sheets alone unless they are trivial.

If I were you I'd be looking to adapt an existing Python solution (assuming one exists).

You are probably better off asking this question on a site like raspberrypi.org/forums which will be viewed by many more people.

closed 2016-07-27 by joan2937

MrYsLab, 2016-07-27:
Have you tried looking at the [Adafruit Library](https://github.com/adafruit/Adafruit_Python_BME280) as a guide?

Using that code as a template, you should be able to port to pigpio fairly easily.

 I did something similar when developing code for the ADXL345 for my [Xideco](https://github.com/MrYsLab/xideco/wiki/ADXL345-Accelerometer) project

joan2937, 2016-08-05:
I borrowed a BME280 and have put together some Python code to read the compensated temperature, pressure, and relative humidity.  Both I2C and SPI are supported.

http://abyz.co.uk/rpi/pigpio/examples.html#Python_BME280_py