joan2937/pigpio #74: pigpiod 100% CPU

stheine, 2016-08-08:
Hi Joan,

I have a reproducible case (on my system, with my hardware environment, happening after some hours) of pigpiod going to 100% CPU consumption, and wonder if there is a way I can further troubleshoot this to find the reason.

this happens on Raspbian Jessie, pigpiod version 50 (the one that comes in Jessie).

I'm running an application that's pulling multiple sensors (SPI) about once per second, DHT22 once per minute, and is taking interrupt callbacks. the application is running for multiple hours ok, before I realize that pigpiod shows the high CPU usage (I haven't yet monitored if it's a slow increase, or a sudden jump).
when it's in that state, it still seems working ok, just the load is up.

is there anything I can do to troubleshoot this when it's in that state? any logging I could enable? any special version showing more troubleshooting data? I'd be willing to install special bits on the box.

Regards
  Stefan

joan2937, 2016-08-08:
The first thing I would suggest is changing to the latest version (V55) and see if the behaviour is still there.  I don't think anything between V50 and V55 would affect the behaviour but I'd like to be sure.

Is this a single core Pi?  If it's a Pi2 or Pi3 there may be some mileage in seeing which cores are busy in an attempt to identify the execution thread(s) which are using the CPU.

You could also try to see if anything unusual is happening in the program execution by switching on internal debugs.

On the Pi enter the following commands.
```
cat /dev/pigerr & # display errors and debug statements
pigs csi 5 mils 2000 csi 0 # level 5 debugs for two seconds
```

stheine, 2016-08-09:
Hi joan,

I've upgraded to v55. the same happens after ~5 hours.
It's a Pi model B, single core.

also running logging on my side, I realized that one of my processes got stuck and hours ago, another one running perfectly fine. but even after I kill the stuck process, pigpiod's CPU usage does not go down.

at that time, the debugging is showing data like, which looks like it working perfectly fine:

2016-08-09 12:30:58 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:30:58 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:30:59 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:30:59 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:31:00 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:31:00 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:31:00 gpioNotifyBegin: handle=1 bits=00040000
2016-08-09 12:31:00 gpioWrite: gpio=18 level=0
2016-08-09 12:31:00 gpioSetMode: gpio=18 mode=0
2016-08-09 12:31:01 gpioNotifyBegin: handle=1 bits=00000000
2016-08-09 12:31:01 gpioSetMode: gpio=18 mode=1
2016-08-09 12:31:01 gpioWrite: gpio=18 level=1
2016-08-09 12:31:02 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:31:02 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:31:03 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:31:03 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:31:04 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:31:04 spiXfer: handle=0 count=3 [06 40 00]

only the CPU usage is up:
  PID USER      PR  NI    VIRT    RES    SHR S **%CPU** %MEM     TIME+ COMMAND
14458 root      20   0   12692   1896   1604 S **93.6**  0.4  75:31.70 pigpiod

now, I've restarted everything to start from scratch, and leave the debugging on permanently (the pigs command in a loop). since both processes currently run, it's showing output like this, with handles 0 and 1, indicating both connections are working:

2016-08-09 12:41:49 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 12:41:49 spiXfer: handle=1 count=3 [06 40 00]
2016-08-09 12:41:49 spiClose: handle=1
2016-08-09 12:41:50 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:41:50 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:41:50 gpioSetMode: gpio=18 mode=0
2016-08-09 12:41:50 gpioNotifyBegin: handle=2 bits=0A440000
2016-08-09 12:41:50 gpioSetMode: gpio=18 mode=1
2016-08-09 12:41:50 gpioWrite: gpio=18 level=0
2016-08-09 12:41:50 gpioSetMode: gpio=18 mode=0
2016-08-09 12:41:50 gpioNotifyBegin: handle=2 bits=0A400000
2016-08-09 12:41:50 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 12:41:50 spiXfer: handle=1 count=3 [06 00 00]
2016-08-09 12:41:50 spiClose: handle=1
2016-08-09 12:41:50 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 12:41:50 spiXfer: handle=1 count=3 [06 40 00]
2016-08-09 12:41:50 spiClose: handle=1
2016-08-09 12:41:51 spiXfer: handle=0 count=3 [06 00 00]
2016-08-09 12:41:51 spiXfer: handle=0 count=3 [06 40 00]
2016-08-09 12:41:51 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 12:41:51 spiXfer: handle=1 count=3 [06 00 00]
2016-08-09 12:41:51 spiClose: handle=1
2016-08-09 12:41:51 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 12:41:51 spiXfer: handle=1 count=3 [06 40 00]

I'll see if I can find out anything from that log, when it starts looping the next time.

if you have any additional ideas for troubleshooting, let me know.

Regards
  Stefan

joan2937, 2016-08-09:
I can only guess at possibilities.

You could try the following to see if that recovers the situation.

```
pigs csi 0xf0 mils 2000 csi 0
```

I really need to be able to duplicate the behaviour myself to find the root cause.  Is there any chance of getting a version of your code which I can run to debug the problem?

stheine, 2016-08-09:
It happened again, this time after 9 hours (process started at 12:40 and got stuck at 21.43).

> pigs csi 0xf0 mils 2000 csi 0

that command does not make a difference, the pigpiod continues consuming high CPU, until I stop it.

my process stopped between 21:43:40 and 21:43:50. and here is the section of the log, that might indicate what's going wrong:

2016-08-09 21:43:40 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 21:43:40 spiXfer: handle=1 count=3 [06 00 00]
2016-08-09 21:43:40 spiClose: handle=1

2016-08-09 21:43:40 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 21:43:40 spiXfer: handle=1 count=3 [06 40 00]
2016-08-09 21:43:40 spiClose: handle=1

2016-08-09 21:43:41 spiXfer: handle=0 count=3 [06 00 00]

2016-08-09 21:43:41 spiXfer: handle=0 count=3 [06 40 00]

2016-08-09 21:43:41 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 21:43:41 spiXfer: handle=1 count=3 [06 00 00]
2016-08-09 21:43:41 spiClose: handle=1

2016-08-09 21:43:41 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 21:43:41 spiXfer: handle=1 count=3 [06 40 00]
2016-08-09 21:43:41 spiClose: handle=1

2016-08-09 21:43:42 spiXfer: handle=0 count=3 [06 00 00]

2016-08-09 21:43:43 spiOpen: spiChan=0 baud=500000 spiFlags=0x0
2016-08-09 21:43:43 spiXfer: handle=1 count=3 [06 00 00]

2016-08-09 21:43:43 spiXfer: handle=0 count=3 [06 40 00]

2016-08-09 21:43:44 spiXfer: handle=0 count=3 [06 00 00]

2016-08-09 21:43:44 spiXfer: handle=0 count=3 [06 40 00]

those sections of three (spiOpen, spiXfer, spiClose) are coming from the process, which gets stuck. the single ones (spiXfer) are coming from the other one that continues to run.

the problem one is coming from a new implementation in which I capsulated the spi-handling, meaning it does not keep a spi handle open, but always opens a new one and closes after the successful transfer. as I can see now, that might be a bad idea. I'll change that implementation in the coming days to see if that makes a difference.

from your side, can you imagine if there is some kind of an overrun happening after spi_open() and spi_close() getting called too often (9 hours, 2 calls per second -> 9 * 60 * 60 * 2 -> 64800 times), and resulting in some state that results in high CPU load?

joan2937, 2016-08-10:
I checked for errors with many open/close iterations.  There don't appear to be any around the 64k mark.

```py
#!/usr/bin/env python

import pigpio

pi = pigpio.pi()

if not pi.connected:
   exit()

count=0

while True:

   h = pi.spi_open(0, 50000)
   pi.spi_xfer(h, [1, 2, 3])
   pi.spi_close(h)

   count += 1

   print(count, h)
```

I killed the script after 150k iterations.

```
(150178, 6)
(150179, 6)
(150180, 6)
(150181, 6)
(150182, 6)
(150183, 6)
(150184, 6)
(150185, 6)
(150186, 6)
(150187, 6)
(150188, 6)
(150189, 6)
^C
```

You mention "are coming from the process, which gets stuck".  Is this your own process?

SlySven, 2016-08-11:
It may sound stupid but at 2 calls per second for around 9 hours - doesn't that sound like a 16-bit unsigned integer overflow...? :bomb:

stefanheine, 2016-08-11:
that's what I had also though initially, but it's not related as I realized now. since I change my code to get the handle once only, and use it for the ongoing spi reads, I now just realized that it still causes the daemon to start using high CPU. so my initial idea was wrong. :(

when I now stopped my stuck process, I realized it was returning -2001 from spi_xfer. so maybe something in my error handling is incorrect, that fails on such an intermittent failure.

I'll continue investigating... and let you know what I find.

joan2937, 2016-08-11:
@SlySven The 64k figure is suggestive.  It is hit on Linux when a file repeatedly opened without being close.  There are (by default I believe) 64k file descriptors per process.  pigpio SPI doesn't use the Linux driver though.

@stheine  pigif_bad_recv  = -2001 indicates that something has gone wrong on the socket.  It's not something I've come across in practice.

stheine, 2016-08-13:
I made progress. this time both processes are running fine since ~23 hours. what did I change? I changed the schedule for both processes, so that one of them is polling the spi sensors on the odd seconds, and the other one on the even seconds. so my new theory would be: two independed processes, with different pi handles, trying to pull the same spi data source in parallel.

do you think this could cause some trouble in pigpiod?

for the moment, I seem to be fine with my current implementation, so if you want to close this, I'd be ok (I could reopen this issue if I would find more details).

joan2937, 2016-08-13:
Yes, this is a fault within pigpio.

For various reasons pigpio does not use the Linux SPI driver.  Instead it directly drives the SPI hardware.  The current pigpio SPI implementation is not thread safe.

I may add a semaphore to protect the critical region.  I'll need to think a little about the consequences.

Your solution is a sort of time controlled semaphore.  Another solution would be to only use one thread to talk to the daemon.

When I decide what to do I will add a comment to this issue.

joan2937, 2016-09-22:
Should be fixed in V56.

closed 2016-09-22 by joan2937

stheine, 2016-09-28:
thanks! I've upgraded and will let you know in case I ever experience this issue again.