joan2937/pigpio #77: Pigpio doesn't run under Arch Linux after compilation

thamizh85, 2016-08-14:
When I tried to run pigpiod under Arch Linux, I got error regarding missing library (libpigpio.so). 

```
[root@alarmpi tmp]# ldd /usr/local/bin/pigpiod 
	linux-vdso.so.1 (0x7efbb000)
	libpigpio.so => not found
	librt.so.1 => /usr/lib/librt.so.1 (0x76f5e000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x76f35000)
	libc.so.6 => /usr/lib/libc.so.6 (0x76df9000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x76dcd000)
	/lib/ld-linux-armhf.so.3 (0x76f75000)

```
I didn't capture the exact error message, but I fixed it by adding '/usr/local/lib' to /etc/ld.so.conf. Is it possible to update the build command to automatically cater to such cases?

joan2937, 2016-08-14:
See #38 PIGPIO under Arch

closed 2016-08-14 by joan2937

SlySven, 2016-08-16:
I'm not sure which location should be used but the Linux File Hierarchy Standard on [/usr/local](http://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s09.html) may be helpful...