joan2937/pigpio #78: Wrong use of GPIO Set Mode

Joey-1970, 2016-08-21:
Helo Joan,

I make a mistake!
All work good, but I want to do it right and set the GPIO-Pin 2 and 3 to use in the "right" mode.
So I want to set to "ALT02 and set it in exactly this way. This is wrong, I had to set it to "0", but this mistake I can't get I²C-Bus to work...

I try different ways to take this back, but nothing I do help. I got always a IPS2GPIO I2C Read Word:  | Handle: 4 Register: 35 Value: PI_I2C_READ_FAILED...

Do you have a workaround for me?

Joachim

joan2937, 2016-08-21:
It's not clear what is wrong.  The Pi's Linux I2C driver expects GPIO 2/3 to be in mode ALT0 which is the default if I2C is enabled.

To force that mode use the following command from Python.

```py
pi.set_mode(2, pigpio.ALT0)
pi.set_mode(3, pigpio.ALT0)
```

From the command line use

```sh
pigs m 2 0
pigs m 3 0
```

Joey-1970, 2016-08-21:
Hello Joan,

thank you very much for your answer!
that don't work... :-( (I try it from command line)

If I start the Raspberry pi and I do the i2c-detect from commandline this is very fast and all three I2C-Devices will be detect.
If I start pigpiod, the same command is very very slow and he don't find the devices...

Joachim

joan2937, 2016-08-21:
Are you saying?

```sh
$ i2cdetect -y 1 # fast response
$ sudo pigpiod
$ i2cdetect -y 1 # very slow response
```

On my Pi the response is equally fast.

Joey-1970, 2016-08-22:
...yes.

1. I make a reboot
2. i2cdetect -y 1 takes < 1 second, all devices found
3. sudo pigpiod - okay
4. i2cdetect -y 1 takes < 1 second, all devices found
5. start my application, devices get handle
6. read devices "PI_I2C_READ_FAILED"
7. i2cdetect -y 1takes > 2 minutes, no devices found

Joachim

joan2937, 2016-08-22:
There is not enough information provided to say whether there is a bug or not.

You need to provide a (shortish) script which can be run and will demonstrate the problem.  Otherwise all I can do is guess at what you are doing.

Joey-1970, 2016-08-24:
Hello Joan,

afters hours of trying, I can say this:
If I send a pack("LLLL", 0, 2, "0", 0)/pack("LLLL", 0, 3, "0", 0) at the start of my application, the Raspberry Pi I2C-Application blocked all further communication. I get a handle but thats it. All write or read get a "failed".
I delete the SetMode, reboot several times, set up a new Jessie - but nothing help.

Yesterday I try to disable  I2C in raspi-config - this may take a long time - , reboot, enable I2C in raspi-config and it work again!

So try to  reproduce the failure:
- send the Pigpio-SetMode at the start -> I get read/write-failure
- delete the Pigpio-SetMode
- disable I2C in raspi-config
- reboot
- enable I2C in raspi-config
- work!

I think someone happens if I send the SetMode for ALT0. Somewhere in my application I send a SetMode with "R" and "W", this works without problems...

Hope it helps... ;-)

Joachim

joan2937, 2016-08-24:
What do you mean by 0, 2, "0", 0?  If you are trying to set mode ALT0 the correct value to send is 4 not "0".

See pigpio.h

```C
/* mode: 0-7 */

#define PI_INPUT  0
#define PI_OUTPUT 1
#define PI_ALT0   4
#define PI_ALT1   5
#define PI_ALT2   6
#define PI_ALT3   7
#define PI_ALT4   3
#define PI_ALT5   2
```

The equivalent pigs command is

```sh
pigs m 2 0
```

But in this case the character 0 (representing ALT0) is converted to 4 by the pigs software.

However there should be no need for you to set the mode of the I2C GPIO.  They will be set to mode ALT0 as long as you have enabled I2C in device tree (via an entry in /boot/config.txt).

Joey-1970, 2016-08-24:
Hello Joan,

thanks for your answer! :-)

I look in this table:
o set the mode use the code for the mode.

Mode	Input	Output	ALT0	ALT1	ALT2	ALT3	ALT4	ALT5
Code	R	W	0	1	2	3	4	5

Now it works! Thank yyou very much again!!!!

By the way: What is your favourite way to get PIGPIO start up at boot?

Joachim

joan2937, 2016-08-24:
Using the character '0' rather than the number 4 should have returned an error status,

The simplest way to start the daemon is probably to use raspi-config -> Advanced Options -> AA GPIO Server -> Yes.

Alternatively just add a @reboot entry into the root crontab.

closed 2016-09-22 by joan2937