joan2937/pigpio #80: Serial tests fail on RaspberryPi3B

RichardPotthoff, 2016-08-30:
I get errors when I run the x_... tests on my Raspberry Pi 3B. Here are some of the results:
```
pi@raspberrypi3:~/pigpio $ pigs PIGPV
55

pi@raspberrypi3:~/pigpio $ ./x_pipe|grep fail
SLR-f fail with (16 221 205 1 226 83 109 45 1 154 211 94 162 211 83 113 255)

pi@raspberrypi3:~/pigpio $ ./x_pigpio.py |grep FAILED
TEST  5.4  FAILED got 45 (callback: 50)
STRCMP FAILED: len(r)=566, len(s)=556
TEST  5.11 FAILED got 0 (wave tx busy, serial read: 1)
TEST  5.26 FAILED got 45 (callback: 50)
TEST 13.9  FAILED got 885 (glitch filter, wave chain, tally: 0)
TEST 13.10 FAILED got 1320 (noise filter, wave chain, tally: 1500)
TEST 13.11 FAILED got 660 (noise filter, wave chain, tally: 750)

pi@raspberrypi3:~/pigpio $ cat /proc/cpuinfo |grep Revision
Revision        : a22082

pi@raspberrypi3:~/pigpio $ uname -r
4.4.13-v7+
```
I have tried pigpio V50, but I got the same errors. Is there anything else I could try?

joan2937, 2016-08-30:
Do you have anything connected to the expansion header?

If you don't, and you get the same errors every time, your Pi and pigpio are incompatible.

RichardPotthoff, 2016-08-30:
I tried it with nothing connected to the header, and also with a LED+1.5k&ohm; between ground and gpio25. The results of the tests were the same.
What could the problem be? Is it clock speeds? Is it possible to adjust something to make it work?

RichardPotthoff, 2016-09-03:
It is working now! All tests pass.

The problem was that I was running the Raspberry Pi headless via ssh, without a monitor connected to the HDMI port.

I found out that a HDMI monitor has to be connected to the Raspberry Pi at boot time for pigpio to work properly.

If I connect the monitor when the Raspberry Pi is already running, the tests still fail -- even if the HDMI interface is configured to run at the same resolution and frequency. 

The following configuration in `/boot/config.txt` will **not** cause the pigpio tests to pass, even if I connect the monitor before I run the pigpio tests:
```
pi@raspberrypi3:~ $ cat /boot/config.txt|grep -e '^[^#]'
dtparam=audio=on
hdmi_force_hotplug=1
hdmi_group=2  
hdmi_mode=39  
start_x=1
gpu_mem=128
```
This configuration sets the HDMI resolution to 1360x768@60Hz DVI, which is the same mode that is automatically selected when the monitor is connected at boot time, and when HDMI is not configured in /boot/config.txt.

On the other hand: If I disconnect the monitor _after_ the Raspberry Pi is booted up, the pigpio tests will still pass, with nothing connected to the HDMI port.

It looks like the hardware is configured differently during boot, depending on the presence of a monitor at the HDMI port at boot time, and that somehow affects pigpio.

I did not see this problem on an older Raspberry Pi 2 B with an older Linux distribution. There I had no problems running the pigpio tests without a monitor. 

My conclusion is that it is not a hardware problem, but an issue with the Raspbian version I am using on my Raspberry Pi 3B.

RichardPotthoff, 2016-09-04:
I finally solved my problem by having the GPU read the `edid` data from a file. Here are the steps:

1. Copy the `edid` data from the monitor (with the monitor connected):

  ```
  sudo /opt/vc/bin/tvservice -d /boot/edid.dat
  ```
2. Configure `/boot/config.txt` to read the `edid` data from the file, and not from the monitor:

  ```
  hdmi_force_hotplug=1
  hdmi_edid_file=1
  ```

If I disconnect the monitor now, reboot, and run the pigpio tests again, I get no errors.

joan2937, 2016-09-05:
Okay, thanks for this information.

Actually the initial errors weren't that significant unless you got the same one repeatably or you cut out a lot of the error reports.

I always run pigpio headless without a monitor (I don't even have a HDMI device).

My best guess at the moment is that the HDMI device tree stuff (in /boot/config.txt) is screwing up the clocks.

I have seen similar reports for servoblaster and pi-blaster which also use DMA.

If you get the time it might be worth going back to your old /boot/config,txt entries but removing the start_x line.  There is a slight chance that X is behind the problem.

RichardPotthoff, 2016-09-05:
It was not the X, it was the audio. Here are the results with only audio enabled in `config.txt`, and with an empty `config.txt`:

```
pi@raspberrypi3:~/pigpio $ cat /boot/config.txt|grep -e '^[^#]'
dtparam=audio=on
pi@raspberrypi3:~/pigpio $ for i in `seq 1 10`; do echo $i `./x_pipe|grep SLR-f`; done
1 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
2 SLR-f fail with (16 221 205 1 226 91 109 45 1 146 211 94 162 211 83 113 255)
3 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
4 SLR-f fail with (16 221 205 1 226 83 109 37 1 146 211 94 162 211 83 113 255)
5 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
6 SLR-f fail with (16 221 205 1 226 91 109 45 1 146 211 94 162 211 83 113 255)
7 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
8 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
9 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
10 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
```
```
pi@raspberrypi3:~/pigpio $ cat /boot/config.txt|grep -e '^[^#]'
pi@raspberrypi3:~/pigpio $ for i in `seq 1 10`; do echo $i `./x_pipe|grep SLR-f`; done
1 SLR-f ok
2 SLR-f ok
3 SLR-f ok
4 SLR-f ok
5 SLR-f ok
6 SLR-f ok
7 SLR-f ok
8 SLR-f ok
9 SLR-f ok
10 SLR-f ok
```
I suspect that the audio defaults to PCM if no audio-capable monitor is detected during boot, and to audio via HDMI otherwise.
Could the PCM clock running at the wrong speed, e.g. 44.1kHz instead of 48kHz cause the problem I am seeing?

joan2937, 2016-09-05:
That seems conclusive.

It's an interesting idea about the clocks.  The following code will display the settings.

```C
/*
   gcc -o get_clocks get_clocks.c
   sudo ./get_clocks
*/

/*
get_clocks.c
2016-06-17
Public Domain
*/

#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <unistd.h>
#include <stdint.h>
#include <string.h>
#include <fcntl.h>
#include <errno.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <sys/types.h>

static volatile uint32_t piModel = 1;

static volatile uint32_t piPeriphBase = 0x20000000;
static volatile uint32_t piBusAddr = 0x40000000;

#define CLK_BASE   (piPeriphBase + 0x101000)

#define CLK_LEN   0xA8

#define CLK_PASSWD  (0x5A<<24)

#define CLK_CTL_MASH(x)((x)<<9)
#define CLK_CTL_BUSY    (1 <<7)
#define CLK_CTL_KILL    (1 <<5)
#define CLK_CTL_ENAB    (1 <<4)
#define CLK_CTL_SRC(x) ((x)<<0)

#define CLK_CTL_SRC_OSC  1  /* 19.2 MHz */
#define CLK_CTL_SRC_PLLC 5  /* 1000 MHz */
#define CLK_CTL_SRC_PLLD 6  /*  500 MHz */
#define CLK_CTL_SRC_HDMI 7  /*  216 MHz */

#define CLK_GP0_CTL 28
#define CLK_GP1_CTL 30
#define CLK_GP2_CTL 32
#define CLK_PCM_CTL 38
#define CLK_PWM_CTL 40

typedef struct
{
   char *name;
   int base;
} clock_info_t;

clock_info_t clock_info[]=
{
   {"GP0", CLK_GP0_CTL},
   {"GP1", CLK_GP1_CTL},
   {"GP2", CLK_GP2_CTL},
   {"PCM", CLK_PCM_CTL},
   {"PWM", CLK_PWM_CTL},
};

/*

Clock sources

0     0 Hz     Ground
1     19.2 MHz oscillator 
2     0 Hz     testdebug0
3     0 Hz     testdebug1
4     0 Hz     PLLA
5     1000 MHz PLLC (changes with overclock settings)
6     500 MHz  PLLD
7     216 MHz  HDMI auxiliary
8-15  0 Hz     Ground
*/

int src_freq[]=
{
   0, 19200000, 500000000, 0, 0, 1000000000, 500000000, 216000000,
   0,        0,         0, 0, 0,          0,         0,         0
};

static volatile uint32_t  *clkReg  = MAP_FAILED;

unsigned hardwareRevision(void)
{
   static unsigned rev = 0;

   FILE * filp;
   char buf[512];
   char term;
   int chars=4; /* number of chars in revision string */

   if (rev) return rev;

   piModel = 0;

   filp = fopen ("/proc/cpuinfo", "r");

   if (filp != NULL)
   {
      while (fgets(buf, sizeof(buf), filp) != NULL)
      {
         if (piModel == 0)
         {
            if (!strncasecmp("model name", buf, 10))
            {
               if (strstr (buf, "ARMv6") != NULL)
               {
                  piModel = 1;
                  chars = 4;
                  piPeriphBase = 0x20000000;
                  piBusAddr = 0x40000000;
               }
               else if (strstr (buf, "ARMv7") != NULL)
               {
                  piModel = 2;
                  chars = 6;
                  piPeriphBase = 0x3F000000;
                  piBusAddr = 0xC0000000;
               }
            }
         }

         if (!strncasecmp("revision", buf, 8))
         {
            if (sscanf(buf+strlen(buf)-(chars+1),
               "%x%c", &rev, &term) == 2)
            {
               if (term != '\n') rev = 0;
            }
         }
      }

      fclose(filp);
   }
   return rev;
}

static int showClock(int clock)
{
   int ctl, div, src, divi, divf;
   double d, freq;
   char * hz;

   ctl = clock_info[clock].base;
   div = clock_info[clock].base+1;

   src = clkReg[ctl]&15;
   divi = (clkReg[div]>>12)&4095;
   divf = clkReg[div]&4095;

   d = divi + (divf / 4096.0);

   if (divi) freq = src_freq[src]/d;
   else      freq = 0;

   if (freq < 1000)         {hz =  "Hz";                 }
   else if (freq < 1000000) {hz = "kHz"; freq /= 1000;   }
   else                     {hz = "MHz"; freq /= 1000000;}

   printf("%s: src=%d divi=%-4d divf=%-4d freq=%-7.3f %s\n",
      clock_info[clock].name, src, divi, divf, freq, hz);
}

/* Map in registers. */

static uint32_t * initMapMem(int fd, uint32_t addr, uint32_t len)
{
    return (uint32_t *) mmap(0, len,
       PROT_READ|PROT_WRITE|PROT_EXEC,
       MAP_SHARED|MAP_LOCKED,
       fd, addr);
}

int initialise(void)
{
   int fd;

   hardwareRevision(); /* sets piModel, needed for peripherals address */

   fd = open("/dev/mem", O_RDWR | O_SYNC) ;

   if (fd < 0)
   {
      fprintf(stderr,
         "This program needs root privileges.  Try using sudo\n");
      return -1;
   }

   clkReg   = initMapMem(fd, CLK_BASE,  CLK_LEN);

   close(fd);

   if (clkReg == MAP_FAILED)
   {
      fprintf(stderr,
         "Bad, mmap failed\n");
      return -1;
   }
   return 0;
}

int main(int argc, char *argv[])
{
   int i;

   if (initialise() < 0)
   {
      fprintf(stderr, "initialise failed\n");
      return 1;
   }

   for (i=0; i<(sizeof(clock_info)/sizeof(clock_info_t)); i++) showClock(i);
}
```

I get the following output.

```sh
gcc -o get_clocks get_clocks.c
sudo ./get_clocks  
GP0: src=0 divi=0    divf=0    freq=0.000   Hz
GP1: src=6 divi=20   divf=0    freq=25.000  MHz
GP2: src=0 divi=0    divf=0    freq=0.000   Hz
PCM: src=6 divi=250  divf=0    freq=2.000   MHz
PWM: src=0 divi=0    divf=0    freq=0.000   Hz
```

RichardPotthoff, 2016-09-05:
Here are my results, first the configuration that works, and then the one with the `hdmi_...` lines commented out:
```
pi@raspberrypi3:~/pigpio $ cat /boot/config.txt|grep -e '^[^#]'
dtparam=audio=on
disable_audio_dither=1
hdmi_force_hotplug=1
hdmi_edid_file=1
start_x=1
gpu_mem=128
pi@raspberrypi3:~/pigpio $ for i in `seq 1 3`; do echo $i `./x_pipe|grep SLR-f`; done
1 SLR-f ok
2 SLR-f ok
3 SLR-f ok
pi@raspberrypi3:~/pigpio $ sudo ./get_clocks  
GP0: src=0 divi=0    divf=0    freq=0.000   Hz
GP1: src=6 divi=20   divf=0    freq=25.000  MHz
GP2: src=1 divi=600  divf=0    freq=32.000  kHz
PCM: src=6 divi=250  divf=0    freq=2.000   MHz
PWM: src=6 divi=50   divf=0    freq=10.000  MHz
pi@raspberrypi3:~/pigpio $ 
```
```
pi@raspberrypi3:~/pigpio $ cat /boot/config.txt|grep -e '^[^#]'
dtparam=audio=on
disable_audio_dither=1
start_x=1
gpu_mem=128
pi@raspberrypi3:~/pigpio $ for i in `seq 1 3`; do echo $i `./x_pipe|grep SLR-f`; done
1 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
2 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
3 SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
pi@raspberrypi3:~/pigpio $ sudo ./get_clocks  
GP0: src=0 divi=0    divf=0    freq=0.000   Hz
GP1: src=6 divi=20   divf=0    freq=25.000  MHz
GP2: src=1 divi=600  divf=0    freq=32.000  kHz
PCM: src=6 divi=250  divf=0    freq=2.000   MHz
PWM: src=6 divi=50   divf=0    freq=10.000  MHz
pi@raspberrypi3:~/pigpio $
```

joan2937, 2016-09-05:
I can't duplicate this behaviour.  I don't have a HDMI device attached.

Are you playing audio at the time of the test?

Is your firmware up to date?  It's possible this is an issue which has been fixed.

gavin:~/PIGPIO $ uname -a
Linux gavin 4.4.19-v7+ #906 SMP Tue Aug 23 15:53:06 BST 2016 armv7l GNU/Linux

```sh
mercury:~$ ssh gavin

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Sep  5 16:39:54 2016 from mercury.lan
gavin:~ $ sudo ./get_clocks  
GP0: src=0 divi=0    divf=0    freq=0.000   Hz
GP1: src=6 divi=20   divf=0    freq=25.000  MHz
GP2: src=1 divi=600  divf=0    freq=32.000  kHz
PCM: src=0 divi=0    divf=0    freq=0.000   Hz
PWM: src=0 divi=0    divf=0    freq=0.000   Hz
gavin:~ $ sudo pigpiod
gavin:~ $ sudo ./get_clocks  
GP0: src=0 divi=0    divf=0    freq=0.000   Hz
GP1: src=6 divi=20   divf=0    freq=25.000  MHz
GP2: src=1 divi=600  divf=0    freq=32.000  kHz
PCM: src=6 divi=250  divf=0    freq=2.000   MHz
PWM: src=0 divi=0    divf=0    freq=0.000   Hz
gavin:~ $ cd PIGPIO/ 
gavin:~/PIGPIO $ for i in `seq 1 3`; do echo $i `./x_pipe|grep SLR-f`; done
1 SLR-f ok
2 SLR-f ok
3 SLR-f ok
gavin:~/PIGPIO $ sudo ~/get_clocks  
GP0: src=0 divi=0    divf=0    freq=0.000   Hz
GP1: src=6 divi=20   divf=0    freq=25.000  MHz
GP2: src=1 divi=600  divf=0    freq=32.000  kHz
PCM: src=6 divi=250  divf=0    freq=2.000   MHz
PWM: src=6 divi=50   divf=0    freq=10.000  MHz
gavin:~/PIGPIO $ grep Revision /proc/cpuinfo
Revision	: a02082
gavin:~/PIGPIO $ cat /boot/config.txt 
dtparam=audio=on
disable_audio_dither=1
start_x=1
gpu_mem=128
```

RichardPotthoff, 2016-09-05:
I am not playing audio, but I am sure there are audio drivers running in the background.
I could start killing processes, and see which one is interfering with pigpio.
Does Rasbian have a "Safe Mode" that loads only essential drivers?
```
pi@raspberrypi3:~ $ uname -a
Linux raspberrypi3 4.4.19-v7+ #906 SMP Tue Aug 23 15:53:06 BST 2016 armv7l GNU/Linux
pi@raspberrypi3:~ $ grep Revision /proc/cpuinfo
Revision        : a22082
```
P.S.: I ran `rpi-update` a few days ago (update from 4.4.13-v7+ to 4.4.19-v7+), but that did not change anything regarding the pigpio tests failing when no monitor was connected.

joan2937, 2016-09-05:
Here is a list of the processes running on my system generated  with the following command.

`ps -e|sort -k4`

```
  553 ?        00:00:00 avahi-daemon
  580 ?        00:00:00 avahi-daemon
  767 pts/0    00:00:00 bash
  896 tty1     00:00:00 bash
   28 ?        00:00:00 bioset
   46 ?        00:00:00 bioset
   47 ?        00:00:00 bioset
   48 ?        00:00:00 bioset
   49 ?        00:00:00 bioset
   50 ?        00:00:00 bioset
   51 ?        00:00:00 bioset
   52 ?        00:00:00 bioset
   53 ?        00:00:00 bioset
   54 ?        00:00:00 bioset
   55 ?        00:00:00 bioset
   56 ?        00:00:00 bioset
   57 ?        00:00:00 bioset
   58 ?        00:00:00 bioset
   59 ?        00:00:00 bioset
   60 ?        00:00:00 bioset
   61 ?        00:00:00 bioset
   62 ?        00:00:00 bioset
   63 ?        00:00:00 bioset
   64 ?        00:00:00 bioset
   65 ?        00:00:00 bioset
   66 ?        00:00:00 bioset
   67 ?        00:00:00 bioset
   68 ?        00:00:00 bioset
   69 ?        00:00:00 bioset
   83 ?        00:00:00 bioset
  843 ?        00:00:00 bluetoothd
  239 ?        00:00:00 brcmf_wdog/mmc1
  238 ?        00:00:00 brcmf_wq/mmc1:0
  233 ?        00:00:00 cfg80211
  PID TTY          TIME CMD
  545 ?        00:00:00 cron
   27 ?        00:00:00 crypto
  555 ?        00:00:00 dbus-daemon
  828 ?        00:00:00 dbus-daemon
  827 ?        00:00:00 dbus-launch
   81 ?        00:00:00 deferwq
  491 ?        00:00:00 dhclient
  562 ?        00:00:00 dhcpcd
   76 ?        00:00:00 DWC Notificatio
   74 ?        00:00:00 dwc_otg
   86 ?        00:00:00 ext4-rsv-conver
   34 ?        00:00:00 fsnotify_mark
  945 ?        00:00:00 gvfs-afc-volume
  850 ?        00:00:00 gvfsd
  867 ?        00:00:00 gvfsd-fuse
  959 ?        00:00:00 gvfsd-trash
  941 ?        00:00:00 gvfs-goa-volume
  950 ?        00:00:00 gvfs-gphoto2-vo
  937 ?        00:00:00 gvfs-mtp-volume
  921 ?        00:00:00 gvfs-udisks2-vo
  834 ?        00:00:00 hci0
  835 ?        00:00:00 hci0
  837 ?        00:00:00 hciattach
   89 ?        00:00:00 ipv6_addrconf
   78 ?        00:00:00 irq/92-mmc1
   73 ?        00:00:00 iscsi_eh
   85 ?        00:00:00 jbd2/mmcblk0p2-
   29 ?        00:00:00 kblockd
   22 ?        00:00:00 kdevtmpfs
   25 ?        00:00:00 khungtaskd
    3 ?        00:00:00 ksoftirqd/0
   11 ?        00:00:00 ksoftirqd/1
   15 ?        00:00:00 ksoftirqd/2
   19 ?        00:00:00 ksoftirqd/3
   32 ?        00:00:00 kswapd0
    2 ?        00:00:00 kthreadd
   44 ?        00:00:00 kthrotld
 1072 ?        00:00:00 kworker/0:0
    5 ?        00:00:00 kworker/0:0H
   30 ?        00:00:00 kworker/0:1
  407 ?        00:00:00 kworker/0:1H
 1127 ?        00:00:00 kworker/0:2
   13 ?        00:00:00 kworker/1:0H
 1018 ?        00:00:00 kworker/1:1H
  129 ?        00:00:00 kworker/1:2
  846 ?        00:00:01 kworker/1:3
   17 ?        00:00:00 kworker/2:0H
   87 ?        00:00:00 kworker/2:1
  778 ?        00:00:00 kworker/2:1H
  231 ?        00:00:00 kworker/2:2
 1128 ?        00:00:00 kworker/3:0
   21 ?        00:00:00 kworker/3:0H
   75 ?        00:00:00 kworker/3:1
 1017 ?        00:00:00 kworker/3:1H
   88 ?        00:00:01 kworker/3:2
 1125 ?        00:00:00 kworker/u8:0
   77 ?        00:00:01 kworker/u8:1
 1120 ?        00:00:00 kworker/u8:2
  833 ?        00:00:00 kworker/u9:0
  836 ?        00:00:00 kworker/u9:1
  672 ?        00:00:00 lightdm
  784 ?        00:00:00 lightdm
  693 tty1     00:00:00 login
  881 ?        00:00:13 lxpanel
  877 ?        00:00:00 lxpolkit
  800 ?        00:00:00 lxsession
  965 ?        00:00:00 menu-cached
    9 ?        00:00:00 migration/0
   10 ?        00:00:00 migration/1
   14 ?        00:00:00 migration/2
   18 ?        00:00:00 migration/3
   84 ?        00:00:00 mmcqd/0
   23 ?        00:00:00 netns
   35 ?        00:00:00 nfsiod
 1019 ?        00:00:00 nfsv4.0-svc
  700 ?        00:00:00 ntpd
  874 ?        00:00:00 openbox
  883 ?        00:00:00 pcmanfm
   24 ?        00:00:00 perf
  993 ?        00:09:31 pigpiod
  898 ?        00:00:00 polkitd
 1152 pts/0    00:00:00 ps
    8 ?        00:00:00 rcu_bh
    7 ?        00:00:00 rcu_sched
   31 ?        00:00:00 rpciod
  661 ?        00:00:00 rsyslogd
  762 ?        00:00:00 (sd-pam)
  797 ?        00:00:00 (sd-pam)
   80 ?        00:00:00 SMIO
  824 ?        00:00:00 ssh-agent
  889 ?        00:00:00 ssh-agent
  668 ?        00:00:00 sshd
  755 ?        00:00:00 sshd
  765 ?        00:00:00 sshd
    1 ?        00:00:03 systemd
  759 ?        00:00:00 systemd
  794 ?        00:00:00 systemd
  136 ?        00:00:00 systemd-journal
  547 ?        00:00:00 systemd-logind
  139 ?        00:00:00 systemd-udevd
  576 ?        00:00:00 thd
  924 ?        00:00:00 udisksd
   70 ?        00:00:00 VCHIQ-0
   79 ?        00:00:00 VCHIQka-0
   71 ?        00:00:00 VCHIQr-0
   72 ?        00:00:00 VCHIQs-0
   33 ?        00:00:00 vmstat
  393 ?        00:00:00 wpa_supplicant
   26 ?        00:00:00 writeback
  751 tty7     00:00:03 Xorg
```

RichardPotthoff, 2016-09-05:
Here are my results. 
I tried to kill all processes that were running on my Raspberry Pi, and not on yours (mpd, ntpd, Xtightvnc), but that did not fix the problem. 
To me it look like it is not a running program that is interfering, but a difference in configuration:
I have the same programs running when I add the `hdmi_edid_file` line in the config.txt file, and then everything is working.
```
pi@raspberrypi3:~/pigpio $ diff <(cat ps-joan.txt|awk '//{$1=$2=$3="";print}') <(ps -e|sort -k4|awk '//{$1=$2=$3="";print}')|grep -e '^<'
<    dhclient
<    jbd2/mmcblk0p2-
<    lightdm
<    lightdm
<    nfsv4.0-svc
<    Xorg
pi@raspberrypi3:~/pigpio $ diff <(cat ps-joan.txt|awk '//{$1=$2=$3="";print}') <(ps -e|sort -k4|awk '//{$1=$2=$3="";print}')|grep -e '^>'
>    awk
>    bash
>    bash <defunct>
>    diff
>    ext4-rsv-conver
>    grep
>    jbd2/mmcblk0p5-
>    jbd2/mmcblk0p7-
>    kworker/0:3
>    kworker/1:0
>    kworker/1:1
>    kworker/1:4
>    kworker/2:0
>    kworker/3:3
>    kworker/u8:3
>    kworker/u9:2
>    mpd
>    ntpd
>    sort
>    start-pulseaudi
>    xprop
>    xstartup
>    Xtightvnc
pi@raspberrypi3:~/pigpio $ sudo killall Xtightvnc mpd
pi@raspberrypi3:~/pigpio $ diff <(cat ps-joan.txt|awk '//{$1=$2=$3="";print}') <(ps -e|sort -k4|awk '//{$1=$2=$3="";print}')|grep -e '^>'
>    awk
>    bash
>    bash <defunct>
>    diff
>    ext4-rsv-conver
>    grep
>    jbd2/mmcblk0p5-
>    jbd2/mmcblk0p7-
>    kworker/0:3
>    kworker/1:0
>    kworker/1:1
>    kworker/1:4
>    kworker/2:0
>    kworker/3:3
>    kworker/u8:3
>    kworker/u9:2
>    ntpd
>    sort
pi@raspberrypi3:~/pigpio $ ./x_pipe|grep f
SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
pi@raspberrypi3:~/pigpio $ sudo killall ntpd
pi@raspberrypi3:~/pigpio $ ./x_pipe|grep f
SLR-f fail with (16 221 205 1 226 83 109 45 1 146 211 94 162 211 83 113 255)
pi@raspberrypi3:~/pigpio $
```

joan2937, 2017-02-22:
To date I have made no progress at tracking down the root cause.

Also see https://www.raspberrypi.org/forums/viewtopic.php?p=1042078#p1042078

evandavey, 2017-03-12:
I can confirm that the config.txt changes were required on a headless PI3 in order to get wave functionality to work.

config.txt
```
dtparam=audio=on
disable_audio_dither=1
start_x=1
gpu_mem=128
```

system
```
$ cat /proc/cpuinfo |grep Revision
Revision	: a22082

$ uname -r
4.4.26-v7+
```

OutsourcedGuru, 2017-05-22:
Getting consistent errors from the `./x_pigpio` test as well as `./x_pipe` similar to the error messages above for SLR-f (which brought me here when Google'ing same).  Interestingly, I'm running Raspbian **Jessie Lite** (so there's no Pixel and no startx for the GUI).  **Raspberry Pi 3 Model B**

```
$ sudo ./x_pigpio > log/somefile.txt
TEST  2.5  FAILED got 79 (set PWM dutycycle, callback: 40)
TEST  2.7  FAILED got 799 (callback: 400)
TEST  2.9  FAILED got 8000 (callback: 4000)
TEST  4.6  FAILED got 161 (number of notifications: 80)
TEST  5.4  FAILED got 100 (callback: 50)
TEST  5.11 FAILED got -215 (wave tx busy, serial read: 0)
$ cat /proc/cpuinfo|grep Revision
Revision	: a22082
$ uname -r
4.4.50-v7+
```

Tried booting with an HDMI monitor and tried adjusting the /boot/config.txt as suggested above but since I can't start the GUI on this, it's consistently stuck in failing the serial tests.

I note that I'm running the ssh daemon as well as the vncserver.

For what it's worth, I attempted to temporarily blacklist the snd-bcm2835 using modprobe.d but that didn't seem to change things.

Tried killing several background tasks and then intermittently re-running the test.

It could be that something's left over from another install.  I've recently been working with https://github.com/F5OEO/rpitx so that's possible.  On that theory, I'll try this on another rig.

* Starting over on a **Raspberry Pi Zero W running Jessie full version** (with PIXEL/gui)
* Confirm sound card is running with `lsmod`
`git clone --depth 1 https://github.com/i-am-digital/js-gpiozero.git`
`cd js-gpiozero/PIGPIO`
* Install per instructions (running `make clean` first in the PIGPIO folder before anything)
```
$ sudo ./x_pigpio > log/somefile.txt
TEST  5.4  FAILED got 44 (callback: 50)
TEST  5.11 FAILED got -8 (wave tx busy, serial read: 0)
$ cat /proc/cpuinfo|grep Revision
Revision	: 9000c1
$ uname -r
4.4.50+
```

joan2937, 2017-05-23:
That seems to be a different issue.   Could you raise a new issue?

OutsourcedGuru, 2017-05-23:
@joan2937 Fine.  Please ignore until I can circle back.

closed 2020-01-13 by guymcswain