joan2937/pigpio #81: Are there scenarios where it's necessary to call gpioTerminate?

fivdi, 2016-09-02:
I have a JavaScript module that wraps the pigpio C library. This wrapper never actually calls the gpioTerminate function. The wrapper relies on the pigpio C library to implicitly call gpioTerminate. For example the pigpio C library calls [atexit](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L7748) to register gpioTerminate as a function to call at normal process termination. Signals will also result in the pigpio C library calling gpioTerminate.

Is the assumption that it's not necessary to explicitly call gpioTerminate to stop the library correct as it will always be called implicitly?

joan2937, 2016-09-05:
I wish I could find a way to clear the socket(s) down properly.  The problem is you are relying on both ends, otherwise you get the long timeout.

joan2937, 2016-09-06:
The only reason you probably have to call gpioTerminate is if you want to change one of the configuration parameters  settable with the gpioCfg* functions.

E.g.

The following is valid (although I can't think of any reason for wanting to do it apart from testing.

```c
while (condition)
{
   gpioCfgFunc1(...);
   gpioCfgFunc2(...);
   gpioInitialise();
   ...
   gpioTerminate();
}
```

fivdi, 2016-09-06:
OK. Thanks for the clarification.

closed 2016-09-06 by fivdi