joan2937/pigpio #82: Getting automatic data from serial device

Joey-1970, 2016-09-06:
Hello Joan,

please excuse, but I have another question... ;-)

I try do communicate with a Nextion Touch Display on the serial port over the socket Interface.

I can send data to the display, but I don't get a (automatic) response from it.
I can use "SERDA" to see that there are data and then read it, but I must look every x seconds if there are data. I try to use notification on GPIO 15 to get information if there is communication, but is this the right way?

Can I do something, to get automatic data over the socket (like notification) when the Display send something (for example if someone touch the screen)?

Joachim

joan2937, 2016-09-06:
pigpio provides no automatic trigger on the arrival of serial data.

You could use GPIO callbacks to get a lower latency response, but I wouldn't try for more than 10 a second.

The following Python shows a method.

```py
#!/usr/bin/env python

import time

import pigpio

RXD=15

last_level = pigpio.TIMEOUT

def serial_cb(gpio, level, tick):
   global last_level
   if level == pigpio.TIMEOUT:
      if last_level != pigpio.TIMEOUT:
         print("data={}".format(pi.serial_read(h, 100)))
   last_level = level

pi = pigpio.pi()
if not pi.connected:
   exit(0)

h = pi.serial_open("/dev/serial0", 9600)

cb = pi.callback(RXD, pigpio.RISING_EDGE, serial_cb)

pi.set_watchdog(RXD, 100) # 100 ms timeout after last bit

time.sleep(60)

pi.set_watchdog(RXD, 0) # disable watchdog

cb.cancel()

pi.serial_close(h)

pi.stop()
```

Joey-1970, 2016-09-08:
Hello Joan,

may it be possible to do in the future as a new feature? Something like the notification, but if GPIO 15 used as RxD get only one notification on this? ;-)

Joachim

closed 2016-09-22 by joan2937