joan2937/pigpio #83: How to detect a button with pigpio and python

yves67, 2016-09-16:
Hello
I am a newbee and I don't see how to detect a button with the pigpio library using python ? 
I would like my python wait until the button has been pressed
thanks for a little example
best regards
Yves

MrYsLab, 2016-09-16:
@yves67 First [set the desired pin mode](http://abyz.co.uk/rpi/pigpio/python.html#set_mode) as input. Next, if your hardware requires either pullup/pulldown, [enable that](http://abyz.co.uk/rpi/pigpio/python.html#set_pull_up_down). Either poll the pin using a [read](http://abyz.co.uk/rpi/pigpio/python.html#read) or use a [callback](http://abyz.co.uk/rpi/pigpio/python.html#callback) for pin state change notification.

yves67, 2016-09-16:
THANKS for your answer, I just discover the pigpio library this afternoon....
If I use callback, does that mean my script will be in wait state until the button will be pressed ? 
in fact I am looking for a solution like an interrupt 
I will look at the callback function
thanks again for your help
Regards
Yves67

MrYsLab, 2016-09-16:
@yves67 You will need to create your own "wait loop" since the callback is not blocking. When a state change is detected by pigpio it simply calls your callback function. You can call sleep with a very short delay in a loop.

yves67, 2016-09-16:
OK, I have read some examples of the callback, I will test it tomorrow
Thanks for your help, I have understand the idea
Best Regards
Yves

yves67, 2016-09-17:
Thanks a lot, it is working....
Regards
Yves67

MrYsLab, 2016-09-18:
Glad to have helped. You might want to close this issue if it is resolved.

closed 2016-09-18 by joan2937