joan2937/pigpio #84: object has no attribute 'start

gert3d, 2016-09-22:
- freshly installed NOOBS on Raspberry Pi , updated distribution (sept 2016)
- next Í have installed pigpio using 'sudo make' and' sudo make install (sudo otherwise fail)
- I started daemon and verified using 'pigs t'
- start python 3
>>> import pigpio
>>> pigpio.start()
Problem: object has no attribute 'start'
I have searched and saw a couple of similar issues, however could not resolve my problem. What can I do?
thanks

joan2937, 2016-09-22:
pigpio.start() was deprecated several years ago.

The current style is shown in the example below.
```py
#!/usr/bin/env python

import pigpio

pi = pigpio.pi()

if not pi.connected:
   exit()

pi.write(4, 1) # set GPIO 4 high

print (pi.read(4))

pi.stop()
```

closed 2016-09-24 by joan2937

gert3d, 2016-09-24:
Thanks a lot!